lepton-eda/lepton-eda #44: General contact e-mail address for Lepton developers

peter-b, 2017-02-16:
When a Coverity Scan static analysis run completes, it currently e-mails the results to me, because there isn't a mailing list or group e-mail alias for Lepton developers in general.

jgriessen, 2017-02-21:
Are you ready for testers?  Should I switch from 	https://github.com/vzh/geda-gaf.git   unstable-1.9 yet?

peter-b, 2017-02-21:
@jgriessen Yes, we're open for business and making good progress!  @vzh's `unstable-1.9` branch is officially gone. Everything should work, so please let us know if you run into any problems with compiling or running Lepton 👍.

graahnul-grom, 2018-10-23:
Let's use my email here (in `.travis.yml`) and close the issue.
BTW, what does it mean:
> In order to trigger static analysis with Coverity Scan, push to
> the "coverity_scan" branch.  You should only usually scan the tip of
> "master".

Push what?

On [scan.coverity.com/projects/lepton-eda-lepton-eda](https://scan.coverity.com/projects/lepton-eda-lepton-eda) page I see "Last Analyzed Apr 14, 2017".

vzh, 2018-10-25:
> Let's use my email here (in `.travis.yml`) and close the issue.

How does it help?
We have at least to configure an email alias on some email service that supports its creation (e.g. lepton-devs@some-server.org) and to somehow provide the devs with the info on it.

> BTW, what does it mean:
> 
> > In order to trigger static analysis with Coverity Scan, push to
> > the "coverity_scan" branch.  You should only usually scan the tip of
> > "master".
> 
> Push what?

It means we have to merge new commits (e.g. the `master` branch) into `coverity_scan` to trigger the build and analysis.

> 
> On [scan.coverity.com/projects/lepton-eda-lepton-eda](https://scan.coverity.com/projects/lepton-eda-lepton-eda) page I see "Last Analyzed Apr 14, 2017".

Please re-check it. I've updated the `coverity_scan` branch.

graahnul-grom, 2018-10-25:
Thank you for the explanation!
According to that analysis, `lepton-attrib` is the most dangerous program in the suite. :)
I don't know, is it worth it, but couldn't we contact DJ and ask him to create email for us on his server? Otherwise, I have no idea where can we get one.

vzh, 2020-08-08:
I'll close the issue since there are not so many active devs here in Lepton and no one would want to maintain a new mailing list or ask someone to do it. Not a big deal, I believe.  We have many other ways to talk to each other these days: here on GH, on the `geda-user` ML, using private emails, etc.  Please don't re-open this issue if you are not a Lepton developer or if you don't want to deal with it yourself :-)

closed 2020-08-08 by vzh