lepton-eda/lepton-eda #46: Pull requests effectively stopped due to CI failures

vzh, 2017-02-17:
We cannot ever check now if code builds correctly after a pull
request. For three last PRs I see:

> cc1: error: -Werror=maybe-uninitialized: no option -Wmaybe-uninitialized

After googling I found some gcc versions are buggy in this
relation, people wrote so about 4.6 and 4.7. I wonder if this is
the case for CI builds. I could not find the info about compiler
version in the last report.

FWIW, [here](http://stackoverflow.com/questions/7067385/find-the-gcc-version)
I found how we could check the gcc version and set or drop some flags.

peter-b, 2017-02-17:
@vzh If you merge #40 then the problem will go away :+1:

vzh, 2017-02-17:
@peter-b Do you mean to merge it in the master branch?

peter-b, 2017-02-17:
@vzh Yes! It's ready to go. And then I can retry the CI build for each of the failed PRs

closed 2017-02-17 by vzh