lepton-eda/lepton-eda #54: libgedacairo needs dealing with somehow

peter-b, 2017-02-19:
Our `libgedacairo` actually uses the same _code_ as upstream geda-gaf at the moment, but when compiled it has a dependency on `liblepton` rather than `libgeda`.

The current split between `libgedacairo` and `liblepton` seems a little arbitrary.  Maybe it would make sense to merge `libgedacairo` into `liblepton` rather than rename the library.  I can't see any immediate downside to this.

This blocks v1.9.3 because of the soname conflict with geda-gaf.

ehennes775, 2017-02-19:
> Our libgedacairo actually uses the same code as upstream geda-gaf at the moment, but when compiled it has a dependency on liblepton rather than libgeda.
>
> The current split between libgedacairo and [...]
It would be nice to have a GUI widget library. Maybe move the contents of libgedacairo into liblepton, rename libgedacairo to libleptongui or something, then move widgets from gschem into libleptongui.

vzh, 2017-02-19:
@peter-b I would like to keep rendering library separate. Besides, it would be simpler to just rename it on this stage.
@ehennes775 I think having a gui widget library is a good idea. I wouldn't merge the cairo library with the gui library. Cairo funcs are used by non-gui tools, too.

PR #69 (2017-02-22): Rename libgedacairo to libleptonrenderer.
This branch is to close #54 and to address renaming of core libraries in #39.
The version number of the renamed library has been reset to 1.0.0.
closed 2017-02-22 by peter-b