lepton-eda/lepton-eda #61: Net renaming problems

noqsi, 2017-02-20:
There are a number of issues surrounding net renaming. Consider the following tiny project:

[netnames.tar.gz](https://github.com/lepton-eda/lepton-eda/files/788183/netnames.tar.gz)

"make" yields the following (with my commentary):

> Found duplicate net name, renaming [left] to [up]

Should we allow renaming on the same hierarchical level? If so, we should at least have some way to check that it's intentional, and not a short circuit.

> WARNING: Trying to rename something twice:
	+5V and +5V
are both a src and dest name
This warning is okay if you have multiple levels of hierarchy!

This actually the result of shorting +5V to GND. The message doesn't properly identify the problem. Furthermore, this message is so common (my big board designs generate these by the hundreds), that here the user is likely to ignore it.

> Found a pin [Short] on component [(null)] which does not have a label!

This one is just strange. What's the error?

vzh, 2017-07-04:
@noqsi Hi, I am just working on net renaming in
`lepton-netlist`. Some changes have been already made in #139, the
bug with *(null)* has been fixed, the warning about same src and
dest is hidden now. Other planned changes require some consensus
and some more refactoring work.

I think we should keep the possibility to rename any named net
using a power symbol such as GND.

I suggest the following approach:

- Disable *net-naming-priority* setting in `lepton-netlist`.
- Mapping by *net=* for non-power explicit pins must be
  disallowed. `lepton-netlist` must ignore *net=* values for such
  pins and report the issue.
- *net=* always has a priority over *netname=* but it is taken
  into consideration **ONLY** if it sets net name for a power
  pin. Power pins are explicit pins having *pintype=pwr*.
- Any unnamed net directly connected to an explicit power pin
  gets its name.
- Any named nets, directly connected to an explicit power pin, and
  nets connected to them by name (i.e. having the same *netname=*
  value) on the same hierarchical level are renamed to net name
  taken from that pin. `lepton-netlist` should report such
  renaming.
- We can also add an option to `lepton-netlist` to disallow such
  renamings, in which case `lepton-netlist` should report the
  conflict and exit with return code 1.
- If several power pins with different net names are connected
  directly or by any named or unnamed directly or indirectly
  connected nets on the same hierarchical level, `lepton-netlist`
  must report the conflict and exit with return code 1.
- Nothing should change for implicit pins. They may connect by
  *net=* to any net name.
- For directly connected nets having different *netname=*
  attributes attached we could add an option to tolerate this or
  consider this a fatal error.
- If we tolerate directly connected nets with different
  *netname=*, we could make `lepton-netlist` to rename them (and
  all other nets indirectly connected to them by *netname=* values
  on the same hierarchical level) to some special names in output,
  e.g. *shorted_netnames_value1_value2* for *netname=value1* and
  *netname=value2*. `lepton-netlist` should report the conflicts
  anyway.
- If we disallow directly connected nets with different
  *netname=*, `lepton-netlist` should report such conflicts and
  exit with 1.

WDYT?

noqsi, 2017-07-08:
Vladimir,

I’m not happy taking a DRC attribute, pintype, and loading extra functionality onto it beyond DRC. This is especially true because I think our DRC needs improvement, and entangling it with other functionality will make that more difficult.

I think that much simpler rules will be more comprehensible.

Names bound to nets by net= have precedence over names bound by netname=.
Other than this, net naming conflicts are, by default, errors, returning code 1. This would include short circuits through hierarchy, and netname= conflicts overridden by net= bindings.
Nothing should change for hierarchical renaming or implicit pins.
DRC should be responsible for any more nuanced detection of warnings or errors.

John Doty              Noqsi Aerospace, Ltd.
http://www.noqsi.com/
jpd@noqsi.com