lepton-eda/lepton-eda #70: ci: Coverity Scan builds are broken

peter-b, 2017-02-22:
The [latest `coverity_scan` branch builds are broken](https://travis-ci.org/lepton-eda/lepton-eda/builds/203950040) with error messages about the snarf macros.

<details>
<summary>Relevant extract from build log</summary>

```
  CC       liblepton_la-scheme_init.lo
In file included from /usr/include/guile/2.0/libguile.h:95:0,
                 from ./../include/libgeda_priv.h:4,
                 from scheme_init.c:26:
/usr/include/guile/2.0/libguile/snarf.h:343:3: error: initializer element is not constant
   {       \
   ^
/usr/include/guile/2.0/libguile/snarf.h:82:29: note: in definition of macro â€˜SCM_SNARF_HEREâ€™
 #  define SCM_SNARF_HERE(X) X
                             ^
/usr/include/guile/2.0/libguile/snarf.h:356:3: note: in expansion of macro â€˜SCM_IMMUTABLE_STRINGBUFâ€™
   SCM_IMMUTABLE_STRINGBUF (scm_i_paste (c_name, _stringbuf), contents); \
   ^
/usr/include/guile/2.0/libguile/snarf.h:202:3: note: in expansion of macro â€˜SCM_IMMUTABLE_STRINGâ€™
   SCM_IMMUTABLE_STRING (scm_i_paste (c_name, _string), scheme_name); \
   ^
scheme_init.c:32:1: note: in expansion of macro â€˜SCM_GLOBAL_SYMBOLâ€™
 SCM_GLOBAL_SYMBOL (edascm_object_state_sym, "object-state");
 ^
/usr/include/guile/2.0/libguile/snarf.h:343:3: error: (near initialization for â€˜edascm_object_state_sym_string_raw_cell[0].word_0â€™)
   {       \
   ^
/usr/include/guile/2.0/libguile/snarf.h:82:29: note: in definition of macro â€˜SCM_SNARF_HEREâ€™
 #  define SCM_SNARF_HERE(X) X
                             ^
/usr/include/guile/2.0/libguile/snarf.h:356:3: note: in expansion of macro â€˜SCM_IMMUTABLE_STRINGBUFâ€™
   SCM_IMMUTABLE_STRINGBUF (scm_i_paste (c_name, _stringbuf), contents); \
   ^
/usr/include/guile/2.0/libguile/snarf.h:202:3: note: in expansion of macro â€˜SCM_IMMUTABLE_STRINGâ€™
   SCM_IMMUTABLE_STRING (scm_i_paste (c_name, _string), scheme_name); \
   ^
scheme_init.c:32:1: note: in expansion of macro â€˜SCM_GLOBAL_SYMBOLâ€™
 SCM_GLOBAL_SYMBOL (edascm_object_state_sym, "object-state");
 ^
/usr/include/guile/2.0/libguile/snarf.h:343:3: error: initializer element is not constant
   {       \
   ^
/usr/include/guile/2.0/libguile/snarf.h:82:29: note: in definition of macro â€˜SCM_SNARF_HEREâ€™
 #  define SCM_SNARF_HERE(X) X
                             ^
/usr/include/guile/2.0/libguile/snarf.h:356:3: note: in expansion of macro â€˜SCM_IMMUTABLE_STRINGBUFâ€™
   SCM_IMMUTABLE_STRINGBUF (scm_i_paste (c_name, _stringbuf), contents); \
   ^
/usr/include/guile/2.0/libguile/snarf.h:202:3: note: in expansion of macro â€˜SCM_IMMUTABLE_STRINGâ€™
   SCM_IMMUTABLE_STRING (scm_i_paste (c_name, _string), scheme_name); \
   ^
scheme_init.c:32:1: note: in expansion of macro â€˜SCM_GLOBAL_SYMBOLâ€™
 SCM_GLOBAL_SYMBOL (edascm_object_state_sym, "object-state");
 ^
```

</details>

It's possible that this could be because we're defining `SCM_DEBUG_TYPING_STRICTNESS=0` in the `$CFLAGS` rather than in the `$CPPFLAGS`.

vzh, 2017-02-22:
@peter-b I have the same errors if I do `make clean && make 'CFLAGS=-DSCM_DEBUG_TYPING_STRICTNESS=2'` in my local repo, which IIUC should not be the case for CI builds. Could it be that the variable `COVERITY_SCAN_BRANCH` is not properly set? It would be probably of help to show in logs which flags for `configure` have been chosen.

gareth8118, 2017-03-03:
That link is showing a passing build now - can this be closed?

peter-b, 2017-03-04:
@gareth8118 Unfortunately not.  If you expand out the full log [you see that it bails early](https://travis-ci.org/lepton-eda/lepton-eda/builds/203950040#L1683) (while compiling `liblepton`) but that isn't being detected properly by the Travis CI plugin for Coverity.

gareth8118, 2017-03-05:
ah, OK.

PR #88 (2017-03-05): ci: Fix Coverity Scan builds on Travis
- It turns out that the `$COVERITY_SCAN_BRANCH` environment variable
  isn't set until _after_ the Coverity Scan plugin for Travis runs.

- Rename the Coverity Scan project

Closes #70
closed 2017-03-05 by gareth8118