lepton-eda/lepton-eda #76: Segfault on OS X

noqsi, 2017-02-28:
While lepton compiles on OS X (version 10.9.5), it does not function properly

When I first start `gschem`, it does not display any of the widgets (see the screen capture below).  If I mouse over them, the widgets slowly reveal one by one.  This is also true of the menu widgets.  The scroll for the drawing area is also out of place.

After a short amount of use, I get a segmentation fault with the following logged errors:

```
2017-02-27 20:11:02.111 gschem[96694:507] *** WARNING: Method userSpaceScaleFactor in class NSView is deprecated on 10.7 and later. It should not be used in new applications. Use convertRectToBacking: instead. 
Invalid path [/usr/local/share/gEDA/sym/local] passed to component-library
Feb 27 20:12:11 Kakuzen.local gschem[96694] <Error>: CGContextRestoreGState: invalid context 0x7f8c33c99110. This is a serious error. This application, or a library it uses, is using an invalid context  and is thereby contributing to an overall degradation of system stability and reliability. This notice is a courtesy: please fix this problem. It will become a fatal error in an upcoming update.
Feb 27 20:12:11 Kakuzen.local gschem[96694] <Error>: CGContextSetAllowsAntialiasing: invalid context 0x7f8c33c99110. This is a serious error. This application, or a library it uses, is using an invalid context  and is thereby contributing to an overall degradation of system stability and reliability. This notice is a courtesy: please fix this problem. It will become a fatal error in an upcoming update.
```

[image]

vzh, 2017-02-28:
@noqsi After some googling I found [this](https://github.com/trizen/youtube-viewer/issues/115). Two links there point to gtk3 code and to documentation for class NSScreen used in apple products (and in quartz among others). One of the gtk2 artifacts in the new gtk3 code is still the method `userSpaceScaleFactor` reported on your console. It is deprecated in OS X 10.7 and later, as the error says. As far as I can see on the first page, using link to differences between gtk 3.22 and 3.89, nothing has changed even in the latter version. We even don't support gtk3, so the situation with first warning seems to be hopeless :-( I don't know if the two errors after it are someway bound to it, though.

gareth8118, 2017-02-28:
@noqsi couple of questions:

- did gEDA/gaf run on OS X for you? 
- Is this under `homebrew`, ` macports` or `'fink` or similar? or is it a completely native XCode build?

I ask the former because I did some work on [getting gEDA to build under macports](http://wiki.geda-project.org/geda:gaf_building_git_version#macports) but Apple were making it harder and harder with each OS X release and I eventually realised it was much easier to run it under Parallels in a Linux guest.

noqsi, 2017-02-28:
gEDA/gaf 1.8 runs on OS X for me, although the graphics are glitchy.

This is under homebrew.

gareth8118, 2017-03-05:
Ah, I'd been avoiding homebrew on my current MBP because it used to conflict badly with csound in its use of `/usr/local` but it looks like hb has moved into `/usr/local/Homebrew` now? If I get time, I may install hb to try to reproduce this but I can make no promises!

graahnul-grom, 2018-10-23:
Do we intend to support Mac? If we don't, we should state that in `README` (and list supported systems/architectures there - Linux, BSD, Windows/Cygwin).
I have no access to any mac hardware, so I cannot help here.

FWIW, I found a similar bug on launchpad:
[gschem rendering issues on Mac OSX](https://bugs.launchpad.net/geda/+bug/1551985)

vzh, 2020-03-20:
@noqsi, have you tried to build lepton on Mac since then?

noqsi, 2020-03-20:
I have not tried it. To use Lepton on my Mac, I have an Ubuntu VM. I could try building and running 1.9.10 natively.

vzh, 2020-03-20:
It would be much appreciated. Please try.

noqsi, 2020-03-20:
`configure` gives me:

> configure: error: Your awk (/usr/local/bin/gawk) is missing the `gsub' and/or `toupper' functions.

while `homebrew upgrade gawk` fails, complaining that my OS (10.12) is no longer supported. I suppose I could try upgrading the OS, but this is an old Mac, and it's too important in my present situation for me to feel comfortable experimenting with changes that drastic right now.

vzh, 2020-03-20:
Just out of curiosity, your OS should have some other awk (nawk, mawk, ...)? Maybe it supports those functions.
Just found that on my system there are 2 awk binaries: `gawk` and `mawk`. On Debian (and its derivatives) the one working is chosen via the `alternatives` facility.  Maybe your Mac supports something like this?

noqsi, 2020-03-20:
OK, installing `texinfo` and doing `export PATH="/usr/local/opt/texinfo/bin:$PATH"` got it to build. Still problems. Here's lepton-schematic:
[image]
No menus. There appears to be no text in the schematic, but it's just very tiny. Mousing over the blank upper part gets menus and widgets to appear. Creating a new page works, but adding components to it soon generates an error:

> 2020-03-20 17:32:12.751 lepton-schematic[91290:1168325] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: 'Unlocking Focus on wrong view (<GdkQuartzView: 0x7f8f74ebcea0>), expected <GdkQuartzView: 0x7f8f72482670>'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff81d157eb __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fff96b5648d objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff81d9419d +[NSException raise:format:] + 205
	3   AppKit                              0x00007fff7f8cb306 -[NSView unlockFocus] + 243
	4   libgdk-quartz-2.0.0.dylib           0x000000010de683e5 gdk_quartz_cairo_surface_destroy + 29
	5   libcairo.2.dylib                    0x000000010def684b _cairo_user_data_array_fini + 64
	6   libcairo.2.dylib                    0x000000010df43bb2 cairo_surface_destroy + 132
	7   libcairo.2.dylib                    0x000000010def684b _cairo_user_data_array_fini + 64
	8   libcairo.2.dylib                    0x000000010df03e88 _cairo_default_context_destroy + 14
	9   libleptonrenderer.2.dylib           0x000000010de2d36a eda_renderer_finalize + 42
	10  libgobject-2.0.0.dylib              0x000000010e225be5 g_object_unref + 346
	11  lepton-schematic                    0x000000010d84e54e gschem_toplevel_free + 270
	12  lepton-schematic                    0x000000010d846986 preview_dispose + 102
	13  libgobject-2.0.0.dylib              0x000000010e2259c1 g_object_run_dispose + 46
	14  libgtk-quartz-2.0.0.dylib           0x000000010d92dfe1 gtk_container_destroy + 76
	15  libgobject-2.0.0.dylib              0x000000010e220f91 g_closure_invoke + 255
	16  libgobject-2.0.0.dylib              0x000000010e23458c signal_emit_unlocked_R + 2743
	17  libgobject-2.0.0.dylib              0x000000010e234d6b g_signal_emit_valist + 1816
	18  libgobject-2.0.0.dylib              0x000000010e235454 g_signal_emit + 120
	19  libgtk-quartz-2.0.0.dylib           0x000000010d9c05c6 gtk_object_dispose + 40
	20  libgobject-2.0.0.dylib              0x000000010e2259c1 g_object_run_dispose + 46
	21  libgtk-quartz-2.0.0.dylib           0x000000010d96643f gtk_frame_forall + 34
	22  libgtk-quartz-2.0.0.dylib           0x000000010d92dfe1 gtk_container_destroy + 76
	23  libgobject-2.0.0.dylib              0x000000010e220f91 g_closure_invoke + 255
	24  libgobject-2.0.0.dylib              0x000000010e23458c signal_emit_unlocked_R + 2743
	25  libgobject-2.0.0.dylib              0x000000010e234d6b g_signal_emit_valist + 1816
	26  libgobject-2.0.0.dylib              0x000000010e235454 g_signal_emit + 120
	27  libgtk-quartz-2.0.0.dylib           0x000000010d9c05c6 gtk_object_dispose + 40
	28  libgobject-2.0.0.dylib              0x000000010e2259c1 g_object_run_dispose + 46
	29  libgtk-quartz-2.0.0.dylib           0x000000010d9c3b90 gtk_paned_forall + 39
	30  libgtk-quartz-2.0.0.dylib           0x000000010d92dfe1 gtk_container_destroy + 76
	31  libgobject-2.0.0.dylib              0x000000010e220f91 g_closure_invoke + 255
	32  libgobject-2.0.0.dylib              0x000000010e23458c signal_emit_unlocked_R + 2743
	33  libgobject-2.0.0.dylib              0x000000010e234d6b g_signal_emit_valist + 1816
	34  libgobject-2.0.0.dylib              0x000000010e235454 g_signal_emit + 120
	35  libgtk-quartz-2.0.0.dylib           0x000000010d9c05c6 gtk_object_dispose + 40
	36  libgobject-2.0.0.dylib              0x000000010e2259c1 g_object_run_dispose + 46
	37  libgtk-quartz-2.0.0.dylib           0x000000010d92dfe1 gtk_container_destroy + 76
	38  libgobject-2.0.0.dylib              0x000000010e220f91 g_closure_invoke + 255
	39  libgobject-2.0.0.dylib              0x000000010e23458c signal_emit_unlocked_R + 2743
	40  libgobject-2.0.0.dylib              0x000000010e234d6b g_signal_emit_valist + 1816
	41  libgobject-2.0.0.dylib              0x000000010e235454 g_signal_emit + 120
	42  libgtk-quartz-2.0.0.dylib           0x000000010d9c05c6 gtk_object_dispose + 40
	43  libgobject-2.0.0.dylib              0x000000010e2259c1 g_object_run_dispose + 46
	44  libgtk-quartz-2.0.0.dylib           0x000000010d900da2 gtk_box_forall + 51
	45  libgtk-quartz-2.0.0.dylib           0x000000010d92dfe1 gtk_container_destroy + 76
	46  libgobject-2.0.0.dylib              0x000000010e220f91 g_closure_invoke + 255
	47  libgobject-2.0.0.dylib              0x000000010e23458c signal_emit_unlocked_R + 2743
	48  libgobject-2.0.0.dylib              0x000000010e234d6b g_signal_emit_valist + 1816
	49  libgobject-2.0.0.dylib              0x000000010e235454 g_signal_emit + 120
	50  libgtk-quartz-2.0.0.dylib           0x000000010d9c05c6 gtk_object_dispose + 40
	51  libgobject-2.0.0.dylib              0x000000010e2259c1 g_object_run_dispose + 46
	52  libgtk-quartz-2.0.0.dylib           0x000000010d92dfe1 gtk_container_destroy + 76
	53  libgobject-2.0.0.dylib              0x000000010e220f58 g_closure_invoke + 198
	54  libgobject-2.0.0.dylib              0x000000010e23458c signal_emit_unlocked_R + 2743
	55  libgobject-2.0.0.dylib              0x000000010e234d6b g_signal_emit_valist + 1816
[... cut ...]
	86  libgc.1.dylib                       0x000000010e19abce GC_call_with_stack_base + 26
	87  libguile-2.2.1.dylib                0x000000010e0ee66b scm_with_guile + 45
	88  libguile-2.2.1.dylib                0x000000010e0a1234 scm_boot_guile + 68
	89  lepton-schematic                    0x000000010d832c21 main + 81
	90  libdyld.dylib                       0x00007fff97490235 start + 1
	91  ???                                 0x0000000000000002 0x0 + 2
)
libc++abi.dylib: terminating with uncaught exception of type NSException

vzh, 2020-03-20:
So there are issues with text rendering, preview widget (`eda_renderer_finalize`), and menus.  The two former issues may relate to our cairo rendering. The latter one is questionable...

graahnul-grom, 2020-07-05:
[geda-gaf 1.10](https://github.com/macports/macports-ports/tree/master/science/geda-gaf) has been added to [macports](https://www.macports.org) recently.
If it's there, chances are lepton could be built under macports, too.
macports system is similar to FreeBSD ports, so I think it's worth a try.
The problem is that I have no mac machine to do it.
I've found an [article](https://www.howtogeek.com/289594/how-to-install-macos-sierra-in-virtualbox-on-windows-10/) describing the process of installing MacOS 10.13 ("High Sierra") in VirtuaBox, but you need a mac to download the installer (~8Gb) from the Mac App Store and create an ISO image.
I'll ask my friends if anybody has a mac, good internet connection and
enthusiasm to help us with that. :-)