lepton-eda/lepton-eda #77: An attempt in a Vagrant Ubuntu 14.04 VM

noqsi, 2017-02-28:
When I issue the following at the command line:

```
# ./configure --prefix=$HOME/lepton --disable-update-xdg-database
# make
# make install
# ~/lepton/bin/gschem &
```

I get in the log window of gschem:

```
gEDA/gschem version 1.9.3.20170226

gEDA/gschem comes with ABSOLUTELY NO WARRANTY; see COPYING for more details.

This is free software, and you are welcome to redistribute it under certain

conditions; please see the COPYING file for more details.

Read init scm file [/home/vagrant/lepton/share/gEDA/scheme/gschem.scm]

Backtrace:
In ice-9/boot-9.scm:
 157: 6 [catch #t #<catch-closure 2244400> ...]
In unknown file:
   ?: 5 [apply-smob/1 #<catch-closure 2244400>]
In ice-9/boot-9.scm:
 157: 4 [catch #t #<catch-closure 273d660> ...]
In unknown file:
   ?: 3 [apply-smob/1 #<catch-closure 273d660>]
   ?: 2 [primitive-load "/usr/share/gEDA/print-colormap-lightbg"]
In ice-9/boot-9.scm:
 102: 1 [#<procedure 273a5c0 at ice-9/boot-9.scm:97:6 (thrown-k . args)> system-error ...]
In unknown file:
   ?: 0 [apply-smob/1 #<catch-closure 273d620> system-error ...]

ERROR: In procedure apply-smob/1:
ERROR: In procedure open-file: No such file or directory: "/usr/share/gEDA/print-colormap-lightbg"

ERROR: Failed to load RC file [/home/vagrant/lepton/share/gEDA/system-gafrc]: ERROR: No such file or directory: "/usr/share/gEDA/print-colormap-lightbg"

Backtrace:
In ice-9/boot-9.scm:
 157: 6 [catch #t #<catch-closure 2244400> ...]
In unknown file:
   ?: 5 [apply-smob/1 #<catch-closure 2244400>]
In ice-9/boot-9.scm:
 157: 4 [catch #t #<catch-closure 2a9b560> ...]
In unknown file:
   ?: 3 [apply-smob/1 #<catch-closure 2a9b560>]
   ?: 2 [primitive-load "/usr/share/gEDA/gschem-colormap-darkbg"]
In ice-9/boot-9.scm:
 102: 1 [#<procedure 2a433c0 at ice-9/boot-9.scm:97:6 (thrown-k . args)> system-error ...]
In unknown file:
   ?: 0 [apply-smob/1 #<catch-closure 2a9b520> system-error ...]

ERROR: In procedure apply-smob/1:
ERROR: In procedure open-file: No such file or directory: "/usr/share/gEDA/gschem-colormap-darkbg"

ERROR: Failed to load RC file [/home/vagrant/lepton/share/gEDA/system-gschemrc]: ERROR: No such file or directory: "/usr/share/gEDA/gschem-colormap-darkbg"

New file [/home/vagrant/lepton-eda-1.9.3/untitled_1.sch]
Tried to set the sensitivity on non-existent menu item '_Edit/Cu_t'
Tried to set the sensitivity on non-existent menu item '_Edit/_Copy'
Tried to set the sensitivity on non-existent menu item '_Edit/_Delete'
Tried to set the sensitivity on non-existent menu item '_Edit/Copy Mode'
Tried to set the sensitivity on non-existent menu item '_Edit/Multiple Copy Mode'
Tried to set the sensitivity on non-existent menu item '_Edit/Move Mode'
Tried to set the sensitivity on non-existent menu item '_Edit/Rotate 90 Mode'
Tried to set the sensitivity on non-existent menu item '_Edit/Mirror Mode'
Tried to set the sensitivity on non-existent menu item '_Edit/Edit...'
Tried to set the sensitivity on non-existent menu item '_Edit/Slot...'
Tried to set the sensitivity on non-existent menu item '_Edit/Lock'
Tried to set the sensitivity on non-existent menu item '_Edit/Unlock'
Tried to set the sensitivity on non-existent menu item '_Edit/Embed Component/Picture'
Tried to set the sensitivity on non-existent menu item '_Edit/Unembed Component/Picture'
Tried to set the sensitivity on non-existent menu item '_Edit/Update Component'
Tried to set the sensitivity on non-existent menu item 'Hie_rarchy/_Down Schematic'
Tried to set the sensitivity on non-existent menu item 'Hie_rarchy/Down _Symbol'
Tried to set the sensitivity on non-existent menu item 'Hie_rarchy/D_ocumentation...'
Tried to set the sensitivity on non-existent menu item 'A_ttributes/_Attach'
Tried to set the sensitivity on non-existent menu item 'A_ttributes/_Detach'
Tried to set the sensitivity on non-existent menu item 'A_ttributes/Show _Value'
Tried to set the sensitivity on non-existent menu item 'A_ttributes/Show _Name'
Tried to set the sensitivity on non-existent menu item 'A_ttributes/Show _Both'
Tried to set the sensitivity on non-existent menu item 'A_ttributes/_Toggle Visibility'
Tried to set the sensitivity on non-existent menu item '_Edit/_Paste'
```

noqsi, 2017-02-28:
The line

```
ERROR: In procedure open-file: No such file or directory: "/usr/share/gEDA/print-colormap-lightbg"
```

Makes me think that some paths are hard-coded, which breaks custom installations.

vzh, 2017-02-28:
@noqsi They should not.

Could you try the following things:
- `~/lepton/bin/gaf shell -c '(display geda-rc-path)` - what does it output?
- do `./configure` with another prefix, say `/tmp` to see if `$HOME` settings do not interfere in your installation.
Thank you for report!

noqsi, 2017-02-28:
`~/lepton/bin/gaf shell -c '(display geda-rc-path)'` crashes and burns just like gschem.

`ERROR: No such file or directory: "/usr/share/gEDA/print-colormap-lightbg"`

This file is actually installed at ~/lepton/share/gEDA/print-colormap-lightbg.

noqsi, 2017-02-28:
Same problem installing in `/tmp`. Looking in `/usr/share` rather than `/tmp/lepton/share`.

vzh, 2017-02-28:
John, try to add `--enable-relocatable` flag to `./configure`.

vzh, 2017-02-28:
@noqsi There are also variables *GEDADATA*, *GEDADATADIR*, *GEDARCDIR*. Could you show the output of the command `grep GEDA config.log` on your system?

vzh, 2017-02-28:
@noqsi BTW, the variables I've mentioned can be set in you environment, or the path could have been set in one of your `gafrc` files, say, in `~/.gEDA`.

vzh, 2017-02-28:
@noqsi I found the definition of `geda-rc-path` which is used to find colormap scheme files in `geda.scm`:
> (define geda-rc-path (or (getenv "GEDADATARC") (getenv "GEDADATA")
>                         ((@ (srfi srfi-1) last) (sys-config-dirs))))

`env` should show if one of the two first variables is set.

vzh, 2017-02-28:
@noqsi The last option `(sys-config-dirs)` uses [g_get_system_config_dirs ()](https://developer.gnome.org/glib/stable/glib-Miscellaneous-Utility-Functions.html#g-get-system-config-dirs) to search the directory where `share/gEDA` may live. Please also look if the env var `XDG_CONFIG_DIRS` is set (though normally your `--prefix` should define the last entry in `(sys-config-dirs)`).

peter-b, 2017-03-03:
@noqsi What happens if you run this?

    gaf --no-rcfiles shell -c '(display geda-rc-path)'

noqsi, 2017-03-04:
@peter-b
`Kakuzen:~ jpd$ lepton/bin/gaf --no-rcfiles shell -c '(display geda-rc-path)'`
`Backtrace:`
yadda yadda ...
`ERROR: Unbound variable: geda-rc-path`

vzh, 2017-03-04:
@peter-b Neither `gaf --no-rcfiles ...` works for me with the same error as @noqsi showed, although anything other works OK.

peter-b, 2017-03-04:
@vzh Sorry, you're completely correct; the `geda-rc-path` variable is set in `geda.scm`, which is loaded by `system-gafrc`.  So if you disable RC files, `geda-rc-path` isn't available; the Scheme environment is left pretty empty because most of the Scheme stuff has now been moved into loadable modules.  It seems like it would be better to run something like:

    lepton/bin/gaf --no-rcfiles shell -c '(display (%search-load-path "geda")) (newline)'

@noqsi My suspicion is that Lepton is finding its Scheme input files in the wrong place due to the new path lookup features.  It would be very useful to check the output of the following commands:

    $HOME/lepton/bin/gaf --version
    $HOME/lepton/bin/gaf shell -c '(display geda-rc-path)(newline)'
    XDG_DATA_DIRS=$HOME/lepton $HOME/lepton/bin/gaf shell -c '(display geda-rc-path)(newline)'
    strace -e trace=open $HOME/lepton/bin/gaf shell -c '(display geda-rc-path)(newline)' 2>&1 | grep -v '^open.*\(\.[ms]o\|/sym/\)'

noqsi, 2017-03-04:
```
vagrant@vagrant-ubuntu-trusty-64:~$ lepton/bin/gaf --no-rcfiles shell -c '(display (%search-load-path "geda")) (newline)'
/home/vagrant/lepton/share/gEDA/scheme/geda.scm
```

noqsi, 2017-03-04:
@vzh 
@peter-b 

`XDG_DATA_DIRS=$HOME/lepton` seems to fix a lot. With that, gschem at least superficially functions (although some icons are broken).

Could the problem be:

`--disable-update-xdg-database` in my configure?

vzh, 2017-03-04:
@noqsi Could you add just in installed */home/vagrant/lepton/share/gEDA/scheme/geda.scm* the following code just after definition of `geda-rc-path`:
```C
(format (current-error-port)
        "GEDADATARC: ~A\nGEDADATA: ~A\nsys-config-dirs: ~A\n"
        (getenv "GEDADATARC")
        (getenv "GEDADATA")
        (sys-config-dirs))
```

We would see then, if the file is loaded in your configuration, and what settings it uses.

noqsi, 2017-06-12:
And now I cannot reproduce the problem.

vzh, 2017-06-14:
@noqsi OK, thank you for the report!
I close the issue. We can always re-open it if somebody hits the same problem.

closed 2017-06-14 by vzh