lepton-eda/lepton-eda #81: Build with --enable-relocatable on FreeBSD

graahnul-grom, 2017-03-02:
On FreeBSD (11.0-STABLE, amd64, r313735) build failed with --enable-relocatable:

> make
[skip]
../liblepton/src/.libs/liblepton.so: undefined reference to `canonicalize_file_name'
make[4]: stopped in /home/dmn/lepton/lepton.git/gaf

  Then I found guess_install_data_dir() function in
liblepton/src/edapaths.c and there:
1) canonicalize_file_name() in a non-portable GNU extension
2) there is no /proc/self/exe on FreeBSD (there is /proc/curproc/file
symlink, however). In fact, the system and all programs are functionig
perfectly well even when /proc is unmounted.
  The following discussion on stackoverflow.com may be of help:
http://stackoverflow.com/questions/1023306/finding-current-executables-path-without-proc-self-exe

Thank you for your work.
Regards,
Dmitry.

vzh, 2017-03-02:
@graahnul-grom I wonder why it worked at all.
- The function `g_file_test("/proc/self/exe", G_FILE_TEST_IS_SYMLINK)` should return _false_ if there is no such file on your system, so the `canonicalize_file_name` should not work at all. Please check the existence of _/proc/self/exe_ once more. Even the dangling symlink will make the test pass as it is stated [here](https://developer.gnome.org/glib/stable/glib-File-Utilities.html#g-file-test).
- If the function has even been called, [here](https://www.gnu.org/software/gnulib/manual/html_node/canonicalize_005ffile_005fname.html) I see that since (for?) FreeBSD 6.0 it goes as "Portability problems fixed by Gnulib:". Do you have _glibc_ on your machine?
- Will it work if you replace _/proc/self/exe_ with _/proc/curproc/file_?

Thank you for the report!

peter-b, 2017-03-03:
@graahnul-grom If you are building and installing on your own machine, I highly recommend `--disable-relocatable`.  I added `--enable-relocatable` to assist with Win32 support and I don't think it will give good results on Win32 systems yet.

graahnul-grom, 2017-03-06:
@peter-b Thank you. Have no problems building without --enable-relocatable.

graahnul-grom, 2017-03-06:
@vzh
> The function g_file_test("/proc/self/exe", G_FILE_TEST_IS_SYMLINK) should return false if
> there is no such file on your system, so the canonicalize_file_name should not work at all.

This is a link-time dependency, not a run-time one.

> If the function has even been called, here I see that since (for?)
> FreeBSD 6.0 it goes as "Portability problems fixed by Gnulib:"

gnulib is just a bunch of source code files, is does not fix anything.

> Do you have glibc on your machine?

No. GNU C library (**g**libc) is primarily a linux thing.

> Will it work if you replace /proc/self/exe with /proc/curproc/file?

Yes, using readlink() instead of canonicalize_file_name().

vzh, 2017-03-07:
@graahnul-grom Ah, sorry for my ignorance.
Then, since we don't support relocatable builds on FreeBSD, as @peter-b said, we should probably close this issue. WDYT?

graahnul-grom, 2017-03-07:
@vzh I think, yes (and I have more issues for you :) ). But I believe that statement about lack of support of relocatable builds on FreeBSD should be included somewhere in installation instructions.

vzh, 2017-03-07:
@peter-b Could you please add info on limited use of `--enable-relocatable` some place in *INSTALL*? We'll close the issue then.

graahnul-grom, 2018-04-16:
I think it's enough to describe it in NEWS.md and 'configure --help' output.
#187

PR #187 (2018-04-22): Indicate that --enable-relocatable is not supported on BSD systems
Should close #81
closed 2018-04-22 by vzh