lepton-eda/lepton-eda #100: gnetlist ignores 'graphical' attribute

graahnul-grom, 2017-03-15:
It does not depend on the type of backend being used,
nor on the value the 'graphical' attribute is set to.
All components end up in the generated netlist.

Tested revision: latest git (2d8cc86).
Tested on: FreeBSD 11.0-STABLE, Debian 8.7 Stable.

Commands used:
gnetlist test_gnetlist_ignores_graphical.sch -g spice-sdb -v -o spice-sdb.net > sdpice-sdb.out
gnetlist test_gnetlist_ignores_graphical.sch -g pcbpins -v -o pcbpins.net > pcbpins.out

In the attachment there are simple test sch file and
gnetlist's output for the above commands:

https://drive.google.com/open?id=0B8Dmx_DW9FydLTZoQ2ZzWXBNVGc

vzh, 2017-03-15:
@graahnul-grom Could you please put your archives just here? Personally I hate third party services that require such clumsy and fat browsers as firefox or chrome (which slow down on my old machines), and would prefer to always have needed data at hand. Thanks.

graahnul-grom, 2017-03-16:
[test_gnetlist_ignores_graphical.zip](https://github.com/lepton-eda/lepton-eda/files/848001/test_gnetlist_ignores_graphical.zip)

vzh, 2017-03-17:
@graahnul-grom  A package won't be in netlist if it has no `refdes=` attribute attached. Inherited ones aren't taken into consideration. You can either remove attached refdeses or rename them, say, to `_refdes=` to get what you need.
Do you think components having `graphical=1` should be ignored anyway?

peter-b, 2017-03-17:
Most of the gnetlist netlist backends can't cope with components that have no `refdes=` attribute.  There are basically two options:

- gnetlist should treat components with no `refdes=` attribute as if they had `graphical=1`
- gnetlist should generate a suitable unique `refdes=` attribute for components that have no `refdes=` and no `graphical=1`

We should probably pick one of these!

Also, two related questions that spring to mind:

- why aren't `refdes=` attributes inherited?  Does anything break if they are?
- if there are multiple components with the same `refdes=` attribute (i.e. they should be merged) and one of the components has `graphical=1`, does that apply to the whole merged component?

pcjc2, 2017-03-18:
peter-b's last point raises a thorny question...

Notionally:

Are attributes attached to symbol instances _always_ to be attributes of the netlist entity of that refdes?

OR.. are they attributes of the the symbol instance, that just happen to get propagated down (in most cases), to the netlist entity created when the symbol instance has a refdes=.

The graphical=1 case (which I'm unsure is a feature, or anti-feature) tends to make me think the later is true?

Would _other_ attributes attached to the instance with graphical=1 propagate to the merged component, or do we consider graphical=1 basically equivalent to "please ignore anything to do with this symbol instance".

FWIW, I think no refdes, no connections is a good use of graphical=1. I would suggest it is a BAD usage if the graphical=1 is applied to a component which otherwise _appears_ connected in the schematic. The main use case is (I think), avoiding warnings for missing refdes= for things like title-blocks etc.. (Or am I wrong?)

Peter C

vzh, 2017-03-21:
Hi guys!
I've investigated the issue a bit and found that:
- `gnetlist` ignores packages without `refdes=` only if they have the attribute `graphical=1`
- I believe the above behaviour  (no inheritance of `refdes=` @peter-b has asked about) is written in C and will be fixed when I have some time for that and good tests (help with tests is very appreciated)
- if a package has no attached `refdes=` and no `graphical=1` attributes, its name is formed as prefix plus *U?*, e.g. *RU?* for a resistor with inherited `refdes=R?`

I think `gnetlist` should warn the user about non-graphical things having no attached or inherited `refdes=`.

I have to investigate other issues you mentioned a bit more.
Thanks.

vzh, 2017-03-21:
I think that the fact that `gnetlist` ignores packages with `graphical=1` and no `refdes=` is "the right thing" IF there is no inherited "refdes=" attribute. Otherwise we have to warn the user, anyway.

vzh, 2017-03-21:
If a component has no `refdes` and no `graphical` we have to warn the user. It might be just a mistake, though not critical one.

vzh, 2017-03-21:
@peter-b About multiple components (your last question).
As of now, processing of multiple components with the same refdes is buggy. The patch proposed doesn't solve the issue because it is a different topic.
You can edit the attached schematic a little, e.g. copy *R3* and delete its `graphical` attribute and do (with my patch applied):
> gnetlist -i test_gnetlist_ignores_graphical.sch
> scheme@(guile-user)> (schematic-graphicals toplevel-schematic)
> $1 = (#<geda-package 58>)
> scheme@(guile-user)> (schematic-packages toplevel-schematic)
> $2 = ("R1" "R2" "R3")
> (map package-refdes (schematic-graphicals toplevel-schematic))
> ("R3")

The way I would wish it to work is to cross-check the attribute user attaches in the GUI for symbols with same `refdes` and leave only one where it makes sense (e.g., it doesn't make sense for `slot`). So `package` in my view is a super-entity for `component`s and it should have its own set of attributes (as well as functions to check them, and package consistensy as a whole).

graahnul-grom, 2017-03-25:
Just tried Vladimir's patch: now gnetlist handles `graphical=1` correctly.
Also, I found that setting this attribute to any other value effectively
negates its meaning. I think this is great, because adding/removing it
every time is very inconvenient.
However, not sure how about backward compatibility here: the former behaviour was to check just for `graphical` attribute's _presence_.

vzh, 2017-03-25:
@graahnul-grom If `gnetlist` even supported `graphical` with other values, it should be considered a bug. Please look at [this](http://wiki.geda-project.org/start?do=search&id=%22graphical%3D%22). And imagine how confusing it would be if some user set `graphical=0` or `graphical=no` to prevent treating a component as a graphical one if we processed it another way. So I think it was a considered decision. I remember we applied patches to avoid other usage a few years ago.

PR #107 (2017-03-28): Fix `graphical` packages processing.
Fixes the issue that packages with attributes 'graphical=' appear in
backend netlists (github issue #100).

Closes #100
closed 2017-03-28 by peter-b