lepton-eda/lepton-eda #99: Github identifies .sch files as KiCAD

gareth8118, 2017-03-15:
reported by pcjc2:

https://github.com/lepton-eda/lepton-eda/search?l=kicad suggests that Lepton is 10.2% KiCAD files as it IDs schematic files as KiCAD files.

GH Language defs are at

https://github.com/github/linguist/blob/master/lib/linguist/languages.yml

this file fixed and PRed

vzh, 2017-03-15:
@gareth8118 Could not find your PR. Could you provide us with direct link to it?

gareth8118, 2017-03-15:
@vzh I haven't created it yet. I'll clarify the ticket.

vzh, 2019-01-11:
Two thoughts:
- we could use another (and unique) file extension to distinguish our files from others. Say, `.lsch`.
- OTOH, what's bad if people looking for KiCAD find our EDA suite?

BTW, the first link doesn't work, something has changed since the report time.

I suggest to close the issue.
Opinions?

graahnul-grom, 2019-01-11:
> BTW, the first link doesn't work, something has changed since the report time.

It has changed from "KiCad Schematic" to "KiCad Layout"

graahnul-grom, 2019-01-11:
> we could use another (and unique) file extension to distinguish our files from others. Say, .lsch.

I think we should change it when we depart enough from the `geda-gaf` (e.g. when we change the file format).

graahnul-grom, 2019-01-11:
> OTOH, what's bad if people looking for KiCAD find our EDA suite?

Indeed. IMHO, nothing to worry about. There's a color bar at the top of the repo page,
and "KiCad Layout" isn't there, at least.

vzh, 2019-01-13:
Found it: `https://github.com/lepton-eda/lepton-eda/search?&q=language:"KiCad Schematic"`

OK, I close the issue.

closed 2019-01-13 by vzh