lepton-eda/lepton-eda #104: Empty attribute value should be valid

noqsi, 2017-03-19:
Right now, attributes with empty values are not allowed. But an empty string is a perfectly valid string, and can be useful for back ends driving downstream flows for which blank input is valid.

vzh, 2017-03-21:
I agree with this. Besides, such attributes could work as templates, e.g. in `spice` backends.

graahnul-grom, 2018-10-23:
@noqsi @vzh Although I agree with this, too, doing so right now will cause interoperability problems with `geda-gaf`. Schematics created in lepton
cannot be processed by that suite (`gschem` attributes with empty values
will be treated just like text, `gnetlist` will fail).
What about postponing the discussion and reopening the issue after
some major release?

vzh, 2018-10-25:
Nothing will prevent users to create attributes in backward compatible manner.

graahnul-grom, 2018-10-25:
Yes, but one have to always keep it in mind while working on schematics.
Allowing an empty values for attributes seems to me as something akin to file format change.

PR #845 (2021-11-15): Allow attributes with empty values
The patch proposed by Karl Hammar on the **geda-user** mailing list.
Closes #104
closed 2021-11-15 by vzh