lepton-eda/lepton-eda #106: Enumerated attributes

noqsi, 2017-03-19:
It would be better for enumerated attributes to have the form `name:number=value` than `name=number:value`. Cleaner to display and easier to promote and override.

vzh, 2017-03-21:
@noqsi You've written the same forms. Please fix this.
And how to support backward compatibility in this case, do you think? Using some config settings or probably toplevel attribs?
Thanks.

peter-b, 2017-03-21:
@noqsi You say that `name:number=value` would be "cleaner to display and easier to promote and override."  Could you provide some detailed examples of what you're talking about, please?

noqsi, 2017-06-12:
Imagine a symbol with "net=VCC:14" and "net=GND:7". How do you unambiguously promote and override? Which one does net=+5CL3:8 override, or does it add another hidden connection? But if it was "net:14=" (empty value) and "net:8=+5CL3" that would indicate that you're moving the connection to pin 8 and changing its netname.

vzh, 2017-06-14:
@noqsi Got it, thanks.
So you're suggesting using of empty values here to explicitly disable connections. What do you think of using multiline `net=` for such cases, e.g.:
> net=
> 14=
> 8=+5CL3

I think using of one attribute instead of multiple ones would be easier for promoting and editing as well as for parsing.