lepton-eda/lepton-eda #114: Incorrect behavior when undo-panzoom option is set to "disabled"

graahnul-grom, 2017-04-02:
Steps to reproduce:
1) add `( undo-panzoom "disabled" )` to `gschemrc` config file
2) start `gschem`
3) open some schematic
4) change the zoom level
5) make changes to schematic (e.g. move or delete something)
6) `Edit -> Undo`

- What happen:
When all changes are undone, the initial zoom level is restored.

- What should happen:
Undo/redo operations should not change the zoom level at all.

This is really annoying.
The bug was [reported](https://bugs.launchpad.net/geda/+bug/1367462) in the past.
Please review a possible [fix](https://github.com/graahnul-grom/lepton-eda/commit/a2751bf9697ef19a7e9d0e5d664fea8e8ccff826).

vzh, 2017-04-02:
No, `undo-panzoom` is for adding additional undo steps to undo zooming changes. Please read its documentation. If you want to add a config setting to disable zooming on undo at all, please add another option using a new config system (as of now, *geda.conf* and similar files).

graahnul-grom, 2017-04-03:
@vzh `system-gschemrc`:
> undo-panzoom string
>
> Controls if pan or zoom commands are saved in the undo list.  If this
> is enabled then a pan or zoom command will be considered a command and
> can be undone.  If this is false, then panning and zooming is not saved
> in the undo list and cannot be undone.  Note, the current viewport
> information is saved for every command, so the display will change to the
> viewport before a command is executed.

Is this correct?
> If this is false, then panning and zooming is not saved
> in the undo list and **cannot be undone**.

Now they *are* undone, and in a strange way: the zoom level is restored
when the undo stack becomes empty.
Am I missing something here? Maybe `undo-panzoom` is documented elsewhere?
Please let me repeat: the current behavior is indeed very irritating. If it's necessary, of course I will add config variable to fix it, as you have said.

vzh, 2017-04-03:
It is related to *ONLY* zooming and panning steps that are done *WITHOUT* schematic modification. You can *git grep* for `undo_panzoom` through the code and see it. Viewport information is saved nevertheless for schematic modification steps. For example, if you add a component, then do VE, then zoom in, then move the component, these four steps should be saved in undo history with `undo-panzoom=enabled`, otherwise only two events (with all viewport scaling and centering info) are saved, that is, adding and moving the component.
I think the documentation looks a bit misleading, especially for users whose first language is not English. Though I cannot make up something better and short just now. I think the `undo-panzoom` rc setting should have been named something like `additional-panzoom-undo-steps` or something like this. (And this can be fixed in new config system when we'll find an appropriate name.)
The feature you're requesting is not implemented yet, and has to have another setting in the configuration system.

graahnul-grom, 2017-04-05:
Would it be appropriate place and name for [the new option](https://github.com/graahnul-grom/lepton-eda/commit/d7c95e6b1fdae66686e720b0e246c7cd837aa0e9)?
```ini
[gschem]
undo-can-modify-panzoom=false
```

vzh, 2017-04-05:
I have asked [here](https://gitter.im/lepton-eda). Join us there to discuss anything you wish :-)

vzh, 2017-04-06:
I think we could use:
```ini
[gschem.undo]
preserve-viewport=false
```
by default.

BTW, please don't modify HTML docs. They have been autogenerated from geda wiki. A better way until we come up with a new documentation system (not very soon, I think) would be translating of parts of wiki into *Texinfo* files in `docs/`.

graahnul-grom, 2017-04-06:
> I have asked here. Join us there to discuss anything you wish :-)

Thanks! :) OT: what irc client would you recommend?

> BTW, please don't modify HTML docs. They have been autogenerated from geda wiki.

Didn't know about that.

Honestly, for me `preserve-viewport` name is ambiguous:
Will viewport be preserved and then restored by `undo`?
Or will it be preserved in the state set by the user, so that `undo` could not change it?
I cannot tell what does it mean.
So far, [the group name is changed and wiki files are reverted](https://github.com/graahnul-grom/lepton-eda/commit/cd0ec0cb3ebe70ae04cb5f8660c2a89fc0eae3d6).

vzh, 2017-04-06:
> Thanks! :) OT: what irc client would you recommend?

I used web interface so far. Though you [can use](irc.gitter.im) anything supporting *ssl*, e.g. *erc-ssl* in Emacs.

> Honestly, for me preserve-viewport name is ambiguous:

Will `restore-viewport` do?

graahnul-grom, 2017-04-06:
Isn't it much of the same as `preserve-viewport` - with just the opposite meaning
(and equally ambiguous)?
It seems that we have to be a bit more verbose here to avoid any confusion.
How about leaving the name as it is in the code now (`undo-can-modify-panzoom`) :-)

vzh, 2017-04-06:
I don't think so. The group name already suggests the option is related to undo. I think *viewport* is a good term for what changes. If you don't like the word *restore*, probably just *modify* or *change* (e.g. `modify-viewport`) would be less ambiguous, what do you think?

graahnul-grom, 2017-04-06:
Yes! Precise.
```ini
[gschem.undo]
modify-viewport=false
```
[Done](https://github.com/graahnul-grom/lepton-eda/commit/1cbdf28ade65128a735a9bde8253b229da0929ef)

vzh, 2017-04-06:
OK, please squash your commits and make a PR.

graahnul-grom, 2017-04-06:
Moreover, I believe that `undo-panzoom` option in `system-gschemrc` should be  disabled by default. Most new users, I think, will not like it. [For example](http://www.eevblog.com/forum/geda/who-besides-me-uses-geda/msg915232/?PHPSESSID=jap455ua872le6s1k9vb8n0gf8#msg915232).

PR #117 (2017-04-11): gschem: Add "modify-viewport" configuration key.
Closes #114
closed 2017-04-11 by vzh