lepton-eda/lepton-eda #127: 'make check' on FreeBSD

graahnul-grom, 2017-05-16:
It's not possible to do a successful `make check`
on FreeBSD. Although I've managed to find where it fails
and make a [fix](https://github.com/graahnul-grom/lepton-eda/commit/e6de9ea12c1b2e8fee62bc0ff2463ee9b11c3fcc), I'm not sure, should it be committed as is.
The way 'make check' fails looks very strange.

1. Unportable `sed` command in `gnetlist` tests is fixed; not a problem.

2. `liblepton` unit test `t0401-os-expand-env-variables.scm`:
It turned out that `(setenv "var" "val")` does not change
environment variable, unless `(putenv "var")` is called first to remove it.

3. `liblepton` unit test `t0402-config.scm`:
Here the real magic begins. In test `'config-real` a call to
`config-load!` results in `SIGBUS` crash. And it crashes only
when `config-load!` is called odd number of times (once, 3, 5, etc.).
Add a second call (third and fourth, etc.) - and crash disappear.

Any ideas on what's going on?
FreeBSD 11.0-STABLE, guile 2.0.14

vzh, 2017-05-17:
@graahnul-grom Do you really think `config-load!` is the culprit? Will it work if you remove all the test body but the first `config-load!` (I mean all `assert-equal` and such)? If it works, then you could add tests one by one.

It looks like you've triggered the second case (*unaligned access*) described [here](https://en.wikipedia.org/wiki/Bus_error).

Don't know what's wrong with `setenv`, it should work the same way as `putenv` (only syntaxes differ). Would `(setenv "var" #f)` instead of `putenv` work? Or would just `(putenv "var=val")` instead of `setenv` work?

Thank you for the report!

graahnul-grom, 2017-05-18:
It seems that only `putenv()` works:
scheme@(guile-user)> `(getenv "USER")`
$1 = "bot"
scheme@(guile-user)> `(setenv "USER" "123")`
scheme@(guile-user)> `(getenv "USER")`
$2 = "bot"
scheme@(guile-user)> `(putenv "USER=123")`
scheme@(guile-user)> `(getenv "USER")`
$3 = "bot"
scheme@(guile-user)> `(setenv "USER" #f)`
scheme@(guile-user)> `(getenv "USER")`
$4 = "bot"
scheme@(guile-user)> **`(putenv "USER")`**
scheme@(guile-user)> `(getenv "USER")`
$5 = "bot"
scheme@(guile-user)> **`(setenv "USER" "123")`**
scheme@(guile-user)> `(getenv "USER")`
$6 = "123"

graahnul-grom, 2017-05-18:
> Will it work if you remove all the test body but the first  config-load!  (I mean all   assert-equal  and such)? If it works, then you could add tests one by one.

Commenting/uncommenting test doesn't make a difference,
only number of times `config-load!()` is called.

vzh, 2017-05-20:
First, let's split these issues. Please make different commits for different topics.

I read [here](http://www.greenend.org.uk/rjk/tech/putenv.html) that FreeBSD violates specifications wrt `setenv` & `putenv`. So I don't mind to add `putenv`, it shouldn't break anything. The only question I have: since those tests are in *geda-gaf* for years, why did nobody  report the same issue so far?

The issue with `config-load!` is most disgusting :-) I haven't understood something, I think. Do other tests (and the function `config-load!` in them) work if you remove the test `config-real` entirely?

Can you try the same code (the test body) in schematic's or gnetlist's REPL?

Just a shot in the dark: might it be issues with your Guile version? Could you check the repo with some other Guile version (say, 2.0.11)? Or maybe with any Guile 2.0 on your architecture (what is it, BTW?).

graahnul-grom, 2017-05-26:
1. No errors in REPL
2. The same failure with guile 2.0.11
3. `make check` may be successful if:
    - `config-real` is commented out
    - previous test (`config-int`) is commented out
    - `config-real` is moved to another place in the file, but it looks completely random: at some places it works, at other it doesn't.

The architecture is amd64.
Apparently, nobody ever bother to run `make check` until recently :)

vzh, 2017-05-27:
@graahnul-grom Could you try to debug this using `gdb` in order to get the backtrace?
It seems to be a lowlevel issue (and I'm still thinking it's an unaligned access to memory as it is related to `config-real`), so we need to have some more information to decide what to do further. I would suggest using of the same command that Makefile uses for tests. E.g. on my system after `make` I do:
> cd ~/path/to/lepton-eda/liblepton/scheme/
> ~/path/to/lepton-eda/cli/lepton-cli --no-rcfiles shell  -L ~/path/to/lepton-eda/liblepton/scheme/ -c '(use-modules (unit-test)) (load (list-ref (command-line) 1)) (exit (if (tests-passed?) 0 1))' unit-tests/t0402-config.scm

and the command works.
You could run the command under `gdb --args` to get the backtrace.

Other options to check:
  - will `make check` work if another optimisation level is used when compiling, you could try `-O0`, `-O1`, etc
  - will it work after compilation in C++ mode.

In order to compile in C++ mode I do as follows:
  - replace `-std=gnu99` with `-std=gnu++03` in `AX_GCC_FLAGS` in *configure.ac*
  - `./autogen.sh && ./configure CC=g++ && make CC=g++`

graahnul-grom, 2017-05-28:
Just got gdb backtrace.
That was not `config-real`. It's `config-string`. This line:
```scheme
(assert-equal a (set-config! a "foo" "bar" '("wib;ble" "wobble")))
```

graahnul-grom, 2017-05-28:
Starting program: /home/dmn/lepton/bin.tb/bin/lepton-cli --no-rcfiles shell -L /home/dmn/lepton/lepton.git/liblepton/scheme -c \(use-modules\ \(unit-test\)\)\ \(load\ \(list-ref\ \(command-line\)\ 1\)\)\ \(exit\ \(if\ \(tests-passed\?\)\ 0\ 1\)\) unit-tests/t0402-config.scm
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/local/lib/libguile-2.0.so.22.8.1-gdb.scm
;;; compiled /home/dmn/.cache/guile/ccache/2.0-LE-8-2.0/usr/local/lib/libguile-2.0.so.22.8.1-gdb.scm.go
PASS: default-config-context
PASS: system-config-context
PASS: user-config-context
[New LWP 101267 of process 57033]
PASS: path-config-context
PASS: config-load
PASS: config-save

** (process:57033): CRITICAL **: eda_config_set_property: assertion '!eda_config_is_descendent (parent, config)' failed
PASS: config-parent
PASS: config-trust
PASS: config-changed
PASS: config-groups
PASS: config-source
PASS: config-keys
PASS: config-boolean
PASS: config-int
PASS: config-real

Thread 1 received signal SIGBUS, Bus error.
__je_arena_mapbits_get (chunk=0x6500656c62600000, pageind=<optimized out>)
    at /usr/src/contrib/jemalloc/include/jemalloc/internal/arena.h:806
806		return (arena_mapbitsp_read(arena_mapbitsp_get_const(chunk, pageind)));
(gdb) bt
#0  __je_arena_mapbits_get (chunk=0x6500656c62600000, pageind=<optimized out>)
    at /usr/src/contrib/jemalloc/include/jemalloc/internal/arena.h:806
#1  __je_arena_mapbits_binind_get (chunk=0x6500656c62600000, pageind=<optimized out>)
    at /usr/src/contrib/jemalloc/include/jemalloc/internal/arena.h:863
#2  __je_arena_salloc (demote=<error reading variable: access outside bounds of object referenced via synthetic pointer>, 
    tsdn=<optimized out>, ptr=<optimized out>) at /usr/src/contrib/jemalloc/include/jemalloc/internal/arena.h:1384
#3  __je_isalloc (demote=<error reading variable: access outside bounds of object referenced via synthetic pointer>, 
    tsdn=<optimized out>, ptr=<optimized out>) at /usr/src/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal.h:951
#4  ifree (tsd=0x6500656c62600000, ptr=<optimized out>, tcache=<optimized out>, slow_path=<optimized out>)
    at jemalloc_jemalloc.c:1810
#5  0x000000080492bd55 in __free (ptr=0x6500656c62626f77) at jemalloc_jemalloc.c:1937
#6  0x0000000803f5a980 in g_free (mem=0x6500656c62626f77) at gmem.c:190
#7  0x0000000803f7e982 in g_strfreev (str_array=0x80bcad3a0) at gstrfuncs.c:2487
#8  0x00000008034d92d8 in scm_dynwind_end () from /usr/local/lib/libguile-2.0.so.22
#9  0x000000080322ea1d in set_config_x (
    cfg_s=<error reading variable: ERROR: Cannot access memory at address 0x4cfffff000e78149>0x80be52ed0, group_s="foo", 
    key_s="bar", value_s=("wib;ble" "wobble")) at scheme_config.c:1009
#10 0x00000008035669cb in ?? () from /usr/local/lib/libguile-2.0.so.22
#11 0x000000080356375b in ?? () from /usr/local/lib/libguile-2.0.so.22
#12 0x00000008034dbb25 in scm_eval () from /usr/local/lib/libguile-2.0.so.22
#13 0x0000000803566a0c in ?? () from /usr/local/lib/libguile-2.0.so.22
#14 0x000000080356375b in ?? () from /usr/local/lib/libguile-2.0.so.22
#15 0x00000008034db276 in scm_call_1 () from /usr/local/lib/libguile-2.0.so.22
#16 0x0000000803566a0c in ?? () from /usr/local/lib/libguile-2.0.so.22
#17 0x000000080356375b in ?? () from /usr/local/lib/libguile-2.0.so.22
#18 0x00000008034dbb25 in scm_eval () from /usr/local/lib/libguile-2.0.so.22
#19 0x000000080353985c in scm_shell () from /usr/local/lib/libguile-2.0.so.22
#20 0x00000000004065c4 in cmd_shell_impl (data=0x0, argc=6, argv=0x7fffffffe648) at shell.c:136
#21 0x00000008034fb902 in ?? () from /usr/local/lib/libguile-2.0.so.22
#22 0x00000008034d323f in ?? () from /usr/local/lib/libguile-2.0.so.22
#23 0x0000000803566a0c in ?? () from /usr/local/lib/libguile-2.0.so.22
#24 0x000000080356375b in ?? () from /usr/local/lib/libguile-2.0.so.22
---Type <return> to continue, or q <return> to quit---
#25 0x00000008034db335 in scm_call_4 () from /usr/local/lib/libguile-2.0.so.22
#26 0x00000008034d3206 in scm_c_with_continuation_barrier () from /usr/local/lib/libguile-2.0.so.22
#27 0x000000080355f135 in ?? () from /usr/local/lib/libguile-2.0.so.22
#28 0x0000000803828c3b in GC_call_with_stack_base () from /usr/local/lib/libgc-threaded.so.1
#29 0x000000080355c1eb in scm_with_guile () from /usr/local/lib/libguile-2.0.so.22
#30 0x00000008034fb8c5 in scm_boot_guile () from /usr/local/lib/libguile-2.0.so.22
#31 0x00000000004065f8 in cmd_shell (argc=6, argv=0x7fffffffe648) at shell.c:144
#32 0x0000000000403831 in main (argc=8, argv=0x7fffffffe638) at lepton-cli.c:179

graahnul-grom, 2017-05-28:
lepton was compiled with `CFLAGS=-O0 -ggdb`

graahnul-grom, 2017-05-28:
Executing that code in REPL gives no error.

graahnul-grom, 2017-05-28:
Nothing changes if lepton is built using g++.

vzh, 2017-05-29:
@graahnul-grom Could you please check out [this branch](https://github.com/vzh/lepton-eda/tree/fix-freebsd-make-check) and check if it solves the issue with config test?

graahnul-grom, 2017-05-29:
Unfortunately, it doesn't help.

graahnul-grom, 2017-05-29:
It seems that I have found the error. `g_strfreev()` expects to receive a NULL-terminated array of strings.
```
diff --git a/liblepton/src/scheme_config.c b/liblepton/src/scheme_config.c
index c42af6c41..38ca8e03c 100644
--- a/liblepton/src/scheme_config.c
+++ b/liblepton/src/scheme_config.c
@@ -961,7 +961,7 @@ SCM_DEFINE (set_config_x, "%set-config!", 4, 0, 0,
     int i = 0;
 
     if (scm_is_string (first_s)) {
-      gchar **value = g_new0 (gchar *, len);
+      gchar **value = g_new0 (gchar *, len + 1);
       scm_dynwind_unwind_handler ((void (*)(void *)) g_strfreev, value,
                                   SCM_F_WIND_EXPLICITLY);
       for (curr_s = value_s; !scm_is_null (curr_s); curr_s = scm_cdr (curr_s)) {
@@ -969,6 +969,10 @@ SCM_DEFINE (set_config_x, "%set-config!", 4, 0, 0,
         value [i++] = g_strdup (tmp);
         free (tmp);
       }
+
+      /* NULL-trminate the list of strings to be passed to g_strfreev(): */
+      value [i] = NULL;
+
       eda_config_set_string_list (cfg, group, key,
                                   (const gchar * const *) value, len);
 ```

vzh, 2017-05-29:
OK, great! :+1: 
Please fix a typo in *trminate* and make a PR.

graahnul-grom, 2017-05-29:
Thank you!
It was really a strange issue.
Why does this bug *never* manifest itself on Linux? I don't know...
It'll be interesting if somebody could try this on Mac.

closed 2017-05-30 by vzh