lepton-eda/lepton-eda #143: Sometimes inherited net= inside symbol overrides net= attached to it.

vzh, 2017-07-11:
Sometimes inherited net= inside symbol overrides net= attached to
it.  See, e.g., the net connected to *U500:10* from
*gnd-custom.sym* inside `gnetlist/tests/common/netattrib.sch`,
having inherited *net=DIGITAL_GND:1* and attached *net=DGND:1*.

How to reproduce
---
> $ gnetlist -g geda -o - common/netattrib.sch |grep "U500 10"
DIGITAL_GND : U500 10

graahnul-grom, 2019-04-16:
Launching `netlist -g geda -o - common/netattrib.sch` gives the following
errors, prefixed by
`"** (process:< PID here>): CRITICAL **: < system time here >: "`

```
Found duplicate net name, renaming two to one
Rename shorted nets (net= has priority): out3 -> signal3
Rename shorted nets (net= has priority): out8 -> signal8
Found duplicate net name, renaming GND to sig9
```

graahnul-grom, 2019-04-16:
@vzh Why "sometimes" in the title?

vzh, 2019-04-26:
@graahnul-grom, well, if the word "sometimes" is confusing here, probably "in some cases" would be better?

graahnul-grom, 2019-04-26:
@vzh It's not about wording, I'm asking because whenever I run this command (and on various systems), it always produces the output shown above. In what cases it gives a different result?

vzh, 2019-05-05:
OK, the reason I'd written this was using of different approaches in **`gnetlist`** and its backends for working with attributes in many *various* places.  Since then, several things have changed.  My intention, when writing of this issue, was to standardize what attribs should be used for netlisting. You could remember or look through the commit history that I added new tests to show what issues we have. Some of them have been discussed on the `geda-user` list though have never been fixed.  Some backends continue today parsing of attributes' text representation, IIRC.  Here, I'm suggesting to fix such (sometimes wrong) usage of attribs and write TRT down explicitly somewhere.  Please feel free to add your suggestion, changes to the topic description, etc.  I have to emphasize here, it's all about standardization of attributes' usage (which, however, can cause some unexpected backward incompatible changes...).