lepton-eda/lepton-eda #147: ./autogen.sh fails

szg0000, 2017-07-18:
**./autogen.sh fails with:**

```
user@debian:~/lepton-eda$ ./autogen.sh 
checking for ./configure.ac ... yes
checking for ./build-tools/desktop-i18n ... yes
checking for autoconf ... /usr/bin/autoconf
checking for autoheader ... /usr/bin/autoheader
checking for automake ... /usr/bin/automake
checking for aclocal ... /usr/bin/aclocal
checking for libtoolize ... /usr/bin/libtoolize
checking for autopoint ... no

autogen.sh: You must have GNU gettext installed.

If your operating system distribution doesn't provide a package, you
can get download it from <http://www.gnu.org/software/gettext>.

checking for automake >= 1.11.0 ... 1.15.1
***Error*** autogen.sh: Some required tools could not be found.
```

**automake is installed:**

```
user@debian:~/lepton-eda$ automake --version
automake (GNU automake) 1.15.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Tom Tromey <tromey@redhat.com>
       and Alexandre Duret-Lutz <adl@gnu.org>.
```
**And gettext also installed:**
```
user@debian:~/lepton-eda$ gettext --version
gettext (GNU gettext-runtime) 0.19.8.1

```

vzh, 2017-07-19:
Hi,
AFAICS, you have to install `autopoint` to make it work.

In `configure.ac` there is macro *AM_GNU_GETTEXT_VERSION* which is processed by `autopoint` to extract the `gettext` version and copy appropriate files to the build infrastructure. `apt-cache show autopoint` will show the info about it for you.

Please report if this helps.
Thanks.

szg0000, 2017-07-19:
Thanks for your help.
Now I have different problem, when running make:

```
Makefile:1108: recipe for target 'lepton-cli.html' failed
make[4]: *** [lepton-cli.html] Error 1
make[4]: Leaving directory '/home/valeo/lepton-eda/cli'
Makefile:791: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/home/valeo/lepton-eda/cli'
Makefile:539: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/valeo/lepton-eda/cli'
Makefile:550: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/valeo/lepton-eda'
Makefile:482: recipe for target 'all' failed
make: *** [all] Error 2

```

vzh, 2017-07-19:
Have you `groff` installed? You can check as follows:
> grep -i groff config.log -A1

BTW, some more info before the hunk you provided would be of some help.

szg0000, 2017-07-19:
**More info:**
```
user@debian:~/lepton-eda$ make
version.h is unchanged
make  all-recursive
make[1]: Entering directory '/home/user/lepton-eda'
Making all in liblepton
make[2]: Entering directory '/home/user/lepton-eda/liblepton'
Making all in po
make[3]: Entering directory '/home/user/lepton-eda/liblepton/po'
make[3]: Leaving directory '/home/user/lepton-eda/liblepton/po'
Making all in data
make[3]: Entering directory '/home/user/lepton-eda/liblepton/data'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/user/lepton-eda/liblepton/data'
Making all in docs
make[3]: Entering directory '/home/user/lepton-eda/liblepton/docs'
Making all in images
make[4]: Entering directory '/home/user/lepton-eda/liblepton/docs/images'
Type: make images  to create all the png/pdf images
make[4]: Leaving directory '/home/user/lepton-eda/liblepton/docs/images'
make[4]: Entering directory '/home/user/lepton-eda/liblepton/docs'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/home/user/lepton-eda/liblepton/docs'
Type: make doxygen  to create doxygen documentation for libgeda
make[3]: Leaving directory '/home/user/lepton-eda/liblepton/docs'
Making all in include
make[3]: Entering directory '/home/user/lepton-eda/liblepton/include'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/user/lepton-eda/liblepton/include'
Making all in lib
make[3]: Entering directory '/home/user/lepton-eda/liblepton/lib'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/user/lepton-eda/liblepton/lib'
Making all in src
make[3]: Entering directory '/home/user/lepton-eda/liblepton/src'
make  all-am
make[4]: Entering directory '/home/user/lepton-eda/liblepton/src'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/home/user/lepton-eda/liblepton/src'
make[3]: Leaving directory '/home/user/lepton-eda/liblepton/src'
Making all in scheme
make[3]: Entering directory '/home/user/lepton-eda/liblepton/scheme'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/user/lepton-eda/liblepton/scheme'
Making all in tests
make[3]: Entering directory '/home/user/lepton-eda/liblepton/tests'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/user/lepton-eda/liblepton/tests'
make[3]: Entering directory '/home/user/lepton-eda/liblepton'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/home/user/lepton-eda/liblepton'
make[2]: Leaving directory '/home/user/lepton-eda/liblepton'
Making all in libleptonrenderer
make[2]: Entering directory '/home/user/lepton-eda/libleptonrenderer'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/user/lepton-eda/libleptonrenderer'
Making all in cli
make[2]: Entering directory '/home/user/lepton-eda/cli'
make  all-recursive
make[3]: Entering directory '/home/user/lepton-eda/cli'
Making all in po
make[4]: Entering directory '/home/user/lepton-eda/cli/po'
make[4]: Leaving directory '/home/user/lepton-eda/cli/po'
make[4]: Entering directory '/home/user/lepton-eda/cli'
No working groff available: cannot create lepton-cli.html
Makefile:1108: recipe for target 'lepton-cli.html' failed
make[4]: *** [lepton-cli.html] Error 1
make[4]: Leaving directory '/home/user/lepton-eda/cli'
Makefile:791: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/home/user/lepton-eda/cli'
Makefile:539: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/user/lepton-eda/cli'
Makefile:550: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/user/lepton-eda'
Makefile:482: recipe for target 'all' failed
make: *** [all] Error 2
user@debian:~/lepton-eda$ 

```
**grep -i groff config.log -A1**

```
user@debian:~/lepton-eda$ grep -i groff config.log -A1
configure:17973: checking for groff
configure:17989: found /usr/bin/groff
configure:18001: result: groff
configure:18014: checking whether groff has an html driver
groff: can't find `DESC' file
groff:fatal error: invalid device `html' (try installing the `groff' package?)
configure:18023: result: no
--
ac_cv_env_GROFF_set=
ac_cv_env_GROFF_value=
ac_cv_env_GTK_CFLAGS_set=
--
ac_cv_prog_GROFF=groff
ac_cv_prog_GUILE=guile
--
ENABLE_GROFF_HTML_FALSE=''
ENABLE_GROFF_HTML_TRUE='#'
ENABLE_KDE_DESKTOP_DATA_FALSE=''
--
GROFF='groff'
GTK_CFLAGS='-pthread -I/usr/include/gtk-2.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/libpng16 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/freetype2'
user@debian:~/lepton-eda$ 
```

vzh, 2017-07-19:
Well, the last command says something is wrong with your `groff` installation. I mean this:
> groff: can't find `DESC' file

> groff:fatal error: invalid device `html' (try installing the `groff' package?)

E.g., on my system:
> $ locate DESC|grep roff|grep html
> /usr/share/groff/1.22.2/font/devhtml/DESC
> $ dlocate /usr/share/groff/1.22.2/font/devhtml/DESC
> groff: /usr/share/groff/1.22.2/font/devhtml/DESC

Please check this. You may have to reinstall the package.

szg0000, 2017-07-19:
`locate DESC|grep roff|grep html`
gives back nothing.

**without html grep:**
```
user@debian:~$ locate DESC | grep roff
/usr/share/groff/1.22.3/font/devascii/DESC
/usr/share/groff/1.22.3/font/devlatin1/DESC
/usr/share/groff/1.22.3/font/devps/DESC
/usr/share/groff/1.22.3/font/devutf8/DESC

```

**groff version:**
```
user@debian:~$ groff --version
GNU groff version 1.22.3
Copyright (C) 2014 Free Software Foundation, Inc.
GNU groff comes with ABSOLUTELY NO WARRANTY.
You may redistribute copies of groff and its subprograms
under the terms of the GNU General Public License.
For more information about these matters, see the file
named COPYING.

called subprograms:

GNU troff (groff) version 1.22.3
GNU grops (groff) version 1.22.3

```

**ls /usr/share/groff/1.22.3/font/devhtml/**
```
user@debian:~$ ls /usr/share/groff/1.22.3/font/devhtml/
B  BI  CB  CBI  CI  CR  DESC  I  R  S
```

vzh, 2017-07-19:
OK, once *devhtml* exists, `locate` just cannot show it because its database is outdated. `sudo updatedb` should fix this (`update-dlocatedb` does the same for the database of Debian packages info `dlocate` works with). Though it is not the issue, I think. The files you found first are from the package `groff-base`, while the directory *devhtml* is in `groff`.

Have you tried to reinstall `groff` (and maybe `groff-base`):
> $ sudo aptitude reinstall groff groff-base

szg0000, 2017-07-20:
After `updatedb`

```
root@debian:~# locate DESC|grep roff|grep html
/usr/share/groff/1.22.3/font/devhtml/DESC

```

vzh, 2017-07-21:
OK, I've looked at the test of `groff` more gazely, and found its test in `geda-groff.m4`. Basically, it consists of two commands:
> echo '.TH conftest 1 "January 1st, 1970" Version 1' > conftest.1

and
> LC_NUMERIC=C $GROFF -man -T html > /dev/null

which, AFAIUI, should take its input from stdin. So you can try, e.g.:
> echo '.TH conftest 1 "January 1st, 1970" Version 1' | LC_NUMERIC=C $(which groff) -man -T html

It should output correct html and exit with 0 on your system, that is
> echo $?

after it should yield 0.

Please check this.

vzh, 2017-07-26:
Any progress with this?

szg0000, 2017-07-26:
Thanks for your reminder. I will check tomorrow.

szg0000, 2017-07-27:
autogen runs successful without errors. Now I have problems with running lepton-schematic:

```
user@debian:~/lepton-eda$ lepton-schematic 
lepton-schematic: error while loading shared libraries: libleptonrenderer.so.1: cannot open shared object file: No such file or directory
```

vzh, 2017-07-27:
Please try this:
> sudo ldconfig

szg0000, 2017-07-30:
Thanks. It works.

closed 2017-07-30 by szg0000