lepton-eda/lepton-eda #148: netlister error

drforbin, 2017-08-03:
I'm getting this error.

Backtrace:
In ice-9/boot-9.scm:
 160: 7 [catch #t #<catch-closure 2071660> ...]
In unknown file:
   ?: 6 [apply-smob/1 #<catch-closure 2071660>]
In ice-9/boot-9.scm:
  66: 5 [call-with-prompt prompt0 ...]
In ice-9/eval.scm:
 432: 4 [eval # #]
In ice-9/boot-9.scm:
2404: 3 [save-module-excursion #<procedure 2091980 at ice-9/boot-9.scm:4051:3 ()>]
4058: 2 [#<procedure 2091980 at ice-9/boot-9.scm:4051:3 ()>]
In unknown file:
   ?: 1 [%with-toplevel #<geda-toplevel 0x258c270> #<procedure main ()>]
   ?: 0 [scm-error misc-error #f ...]

vzh, 2017-08-03:
What exact command do you use?

drforbin, 2017-08-03:
~/lepton/bin/lepton-netlist -g spice-sdb -o out..net Q2.sch

Lepton is installed in $HOME directory under 'lepton'

vzh, 2017-08-04:
OK, AFAICS, if the procedure `scm-error` has a long list of arguments, it usually shown as:
>    ?: 0 [scm-error misc-error #f ...]
>
> ERROR: In procedure scm-error:
> ERROR: some error string

Do you have any lines beginning with *ERROR* after the backtrace you've shown?

vzh, 2017-08-04:
AFAICS from guile sources, your error relates to loading of some source code. You may try to insert the following line into `lepton-netlist` itself to check to what value guile's load-path is set:
> (display %load-path) (newline)

The best place is just before the `(with-toplevel ...)` line.
BTW, what is your version of guile?

drforbin, 2017-08-04:
Thank you for your response.
I'm using guile-2.0.13-r1.
I was also getting an error related to not finding geda.conf.
I created empty files for these and the error went away.
Also recompiled the entire suite (lepton-eda) and it seems to be working now.
I tried some test runs with spice-sdb and verilog net listers and and they seems to work ok!
Seem to. I will do further tests.

I'll get back with you tonight when I try some more tests.

drforbin, 2017-08-05:
Hi...I checked everything out last night and found another problem.
I think it may be related.

the first file (test_lep) is a verilog netlist with your gnetlister.
the 2nd test.txt is via gnetlist. Your version seems to drop the first argument  from the module definition. Let me know if I can help. 

[test_lep.txt](https://github.com/lepton-eda/lepton-eda/files/1202527/test_lep.txt)
[test.txt](https://github.com/lepton-eda/lepton-eda/files/1202528/test.txt)

mario-giovinazzo, 2017-08-05:
Hi,
I'm porting lepton-eda-master on windows10-64bit under CygWin.
Gschem runs fine but netlist has a very similar behaviour.
Just asking for the version (guile 2.0.14) :
```
$ lepton-netlist --version
Backtrace:
In ice-9/boot-9.scm:
 160: 7 [catch #t #<catch-closure 6002be260> ...]
In unknown file:
   ?: 6 [apply-smob/1 #<catch-closure 6002be260>]
In ice-9/boot-9.scm:
  66: 5 [call-with-prompt prompt0 ...]
In ice-9/eval.scm:
 432: 4 [eval # #]
In ice-9/boot-9.scm:
2412: 3 [save-module-excursion #<procedure 6002e0980 at ice-9/boot-9.scm:4084:3 ()>]
4089: 2 [#<procedure 6002e0980 at ice-9/boot-9.scm:4084:3 ()>]
In /cygdrive/c/ZBU/TOOLS/GEDA/bin/lepton-netlist:
  22: 1 [#<procedure 6002e1e80 ()>]
In unknown file:
   ?: 0 [load-extension "/home/mario/geda/lib/liblepton" "libgeda_init"]

ERROR: In procedure load-extension:
ERROR: In procedure dynamic-link: file: "/home/mario/geda/lib/liblepton", message: "The specified module could not be found."

$ dir /home/mario/geda/lib/
liblepton.a  liblepton.la  libleptonrenderer.a  libleptonrenderer.la  pkgconfig
```
What can I do?
Thanks

vzh, 2017-08-06:
@drforbin The issue with verilog is another bug, and it's my fault, sorry. It's trivial to fix and I'll do it soon, I hope. Anyway, we need more tests for each and every backend. For examble, I have not found any special test for verilog, there is no output file vere any verilog module name is defined. If you have some schematics to test verilog and you can share them, this would be of great help, I'll add them to the repo.
Thank you for the report!

vzh, 2017-08-06:
@mario-giovinazzo Your issue is very different. Please open a new issue.

FWIW, I was working on mingw port a couple of years ago ([you can find it here](https://github.com/vzh/minipack)) and am still interested
in a package of Lepton for Windows.

If you point me at your repo, I could probably try it myself.
Thanks.

vzh, 2017-08-06:
@drforbin Please check if the suggested patches fix your verilog output. Thanks.

drforbin, 2017-08-06:
viz, 

Thank you. I will look at the patches.
I would like to help you if I can.
Firstly, I have some schematics and syms etc for verilog testing I can provide.
Please advise on if you would like them.
2nd, I have a modified version of gmk_sym which allows the easy creation of negedge triggered symbols. let me know if you would like to have a look at it.

drforbin, 2017-08-07:
ok..I got it merged...I compiled the patch and it works.

drforbin, 2017-08-07:
Let me know what you wish me to do with my additions

vzh, 2017-08-07:
@drforbin Probably the best option is to integrate your schematics, syms and whatelse with the current netlister test suite and make a PR. Another option is just make a new issue and put all the files there in hope someone other will do the work for you.

vzh, 2017-08-07:
@drforbin As for `gmk_sym` and other utils, well, I would prefer to have them rewritten in Scheme as extension modules which could be then used inside `lepton-schematic`. I don't think it is a good idea to have the tools dependent on the file format we have now. My long term goal is either to replace contrib scripts and utils with Scheme modules, or move them into another repository, since we don't have enough man power to support all them these days. Yet you can still make a PR with your changes and we can include them into the next release. Though the tool should be renamed to not conflict with its sibling in `geda-gaf`.

closed 2017-08-07 by vzh

drforbin, 2017-08-07:
ok...will place them in repo and do a PR so you can have a look.
The verilog examples really helped me. It took me sometime to figure out how to get it all working.
That is a problem with geda is DOC. as for writing in scheme. I know C C++ java, python. have not written anything in scheme. I will try and learn. For now please just have a look.

PR #149 (2017-08-07): Fix verilog module output
Fixes one of the issues reported in #148.
Closes #148