lepton-eda/lepton-eda #150: Netlist error in windows cygwin porting with most recent lepton-eda

mario-giovinazzo, 2017-08-06:
I have successfully compiled lepton-eda-1.9.3 under cigwin targeting windows 64bit.

Starting from $HOME my fold scheme was:
lepton-eda-1.9.3
lepton-eda-1.9.3/build
lepton-eda-1.9.3/source

My procedure was:
1) untar lepton-eda-1.9.3 in source dir
2) run autogen.sh in source/lepton-eda-1.9.3 dir
3) run build.sh in lepton-eda-1.9.3 dir

build.sh script was basically:

```
cd build
mkdir lepton-eda-1.9.3
(
   cd lepton-eda-1.9.3
   #--------------
   export LD_LIBRARY_PATH=$HOME/geda/lib:$LD_LIBRARY_PATH
   export PATH=$HOME/geda/bin:$PATH
   export PKG_CONFIG_PATH=/usr/lib/pkgconfig:$HOME/geda/lib/pkgconfig:$PKG_CONFIG_PATH
   #--------------
   echo "Configuring lepton-eda-1.9.3 (for details see build/lepton-eda-1.9.3-configure.log)"
   ../../source/lepton-eda-1.9.3/configure \
   --prefix=${dstdir} \
   --build=x86_64-pc-cygwin \
   --host=x86_64-pc-cygwin \
   --target=x86_64-pc-cygwin \
   --enable-relocatable \
   --disable-nls \
   --disable-shared \
   > ../lepton-eda-1.9.3-configure.log 2>&1
   #--------------
   echo "Building lepton-eda-1.9.3 (for details see build/lepton-eda-1.9.3-make.log)"
   make LDFLAGS="-no-undefined -lgio-2.0 -lgobject-2.0" > ../lepton-eda-master-make.log 2>&1
   #--------------
   echo "Installing lepton-eda-1.9.3 (for details see build/lepton-eda-1.9.3-install.log)"
   make install > ../lepton-eda-1.9.3-install.log 2>&1
   #--------------
   echo "done"
)
```

Just few adjustments was necessary:
1) add /usr/lib/pkgconfig to PKG_CONFIG_PATH (otherwise it cannot find guile)
2) add GIO lib to link command LDFLAGS="-no-undefined -lgio-2.0 -lgobject-2.0"
(-no-undefined is just to remove a warning)

I can create NETLIST BOM and DRC report directly from windows using a makefile,
and I can run gschem from cygwin bash and an X-server like Xming (to be fix later).

I would like to do the same with the most recent version (lepton-eda-master).
Using the same procedure I can configue-make-install without error.
As result gschem runs fine but NETLIST show this errors
Just asking for the netlist version (guile 2.0.14) :

```
$ lepton-netlist --version
Backtrace:
In ice-9/boot-9.scm:
 160: 7 [catch #t #<catch-closure 6002be260> ...]
In unknown file:
   ?: 6 [apply-smob/1 #<catch-closure 6002be260>]
In ice-9/boot-9.scm:
  66: 5 [call-with-prompt prompt0 ...]
In ice-9/eval.scm:
 432: 4 [eval # #]
In ice-9/boot-9.scm:
2412: 3 [save-module-excursion #<procedure 6002e0980 at ice-9/boot-9.scm:4084:3 ()>]
4089: 2 [#<procedure 6002e0980 at ice-9/boot-9.scm:4084:3 ()>]
In /cygdrive/c/ZBU/TOOLS/GEDA/bin/lepton-netlist:
  22: 1 [#<procedure 6002e1e80 ()>]
In unknown file:
   ?: 0 [load-extension "/home/mario/geda/lib/liblepton" "libgeda_init"]

ERROR: In procedure load-extension:
ERROR: In procedure dynamic-link: file: "/home/mario/geda/lib/liblepton", message: "The specified module could not be found."

$ dir /home/mario/geda/lib/
liblepton.a  liblepton.la  libleptonrenderer.a  libleptonrenderer.la  pkgconfig

```
lepton-netlist.exe doesn't exist, is it correct?
I need help to continue
Thanks

P.S.
I will tray minipack in the next time

mario-giovinazzo, 2017-08-07:
I have forgotten to mention that after run autogen.sh  I manually copied the missing file "version.h" from lepton-eda-1.9.3 to lepton-eda-master source dir.

vzh, 2017-08-07:
@mario-giovinazzo Yes, it is correct that there is no `lepton-netlist.exe`. `lepton-netlist` is not a binary file but a *guile* script.

vzh, 2017-08-07:
Dynamic linking depends on shared libraries, AFAIK. IIUC, those are *libname.dll*, *libname.dll.a* and such. I don't see any such named libs in the `dir` output shown above. However I see `--disable-shared` flag for `configure`. Could this be the reason of the issue?

mario-giovinazzo, 2017-08-08:
Shared libraries have some limitations and complications in CygWin.
So I compiled lepton-eda-1.9.3 with 
   --disable-shared
   --enable-relocatable
The tool is fully working and relocatable, included guile scripts like schdiff.
Following your suggestion I will try to compile the git repository version with shared library, this when I will be back from holiday in September.
Thanks

mario-giovinazzo, 2017-09-25:
Hi vzh, 
yes, you are right.
I have successfully compiled (under cigwin) the latest version of lepton-eda from the git repository Using shared memory and all I tested works fine.
`make LDFLAGS="-no-undefined -llepton -lleptonrenderer -lgio-2.0 -lgobject-2.0"`
Just a note, the link to documentation from help menu seems to be not updated, I renamed the doc dir from lepton-eda to geda-gaf.
Thanks.

vzh, 2017-10-01:
@mario-giovinazzo Thank you for your work!
It would be very nice to have this info written down somewhere. WDYT?
WRT the docs, I've opened #157. Please look if the patch proposed works for you.

closed 2017-10-01 by vzh

mario-giovinazzo, 2017-10-02:
I will post here (next week) information on how to build and run lepton-eda under cygwin.

mario-giovinazzo, 2017-10-05:
How to build lepton-eda on windows 64 bit under CYGWIN starting from zero.
1) Download "setup-x86_64.exe" from cygwin and install cygwin base packages.
2) Run "setup-x86_64.exe" again and add the following packages:
```
Package         Category
-------------------------
gcc-g++         devel
make            devel
pkg-config      devel
Guile           interpreters (libguile2, devel)
GTK+            lib          (gtk, devel, runtime)
FLEX            devel
GAWK            interpreters
Python2.7       interpreter  (interpreter, devel)
Git             devel
Autoconf        devel
Automake        devel
Libtool         devel        (generic)
Gettext         text
Gettext-devel   devel
Gmp             lib          (gmp, libgmp-devel)
Image-magick    graphics
xorg-server     X11
xinit           X11
```
3) Start cygwin
mkdir lepton-eda
cd lepton-eda
make in a file with the following script, let say "build.sh", and run it.
It takes a while
```
#START
#----------------------------------
# Set up Your destination directory
# (it can be outside cygwin)
#----------------------------------
DSTDIR=$HOME/geda

#----------------------------------
#* Cleanup any existing SOURCE directory
#* Create a new one
#* Clone GIT
#----------------------------------
if [ -d source ]; then
   echo "Removing old source directory"
   rm -Rf source
   echo "done"
fi
#-----------
mkdir source
cd source
git clone https://github.com/lepton-eda/lepton-eda.git
cd lepton-eda
./autogen.sh
cd ..
cd ..
echo "autogen done"

#----------------------------------
# Cleanup any existing BUILD directory
# Create a new one
#----------------------------------
echo "Building Lepton Eda and installing in $DSTDIR"
echo "Create a build directory"
if [ -d build ]; then
   echo "Removing old build directory"
   rm -Rf build
   echo "done"
fi
#-----------
mkdir build
cd build

#----------------------------------
# Build and install LEPTON-EDA
#----------------------------------
mkdir lepton-eda
(
   cd lepton-eda
   #-----------------
   # SET ENV..
   #-----------------
   PATH=$DSTDIR/bin:$PATH
   LD_LIBRARY_PATH=$DSTDIR/lib:$LD_LIBRARY_PATH
   PKG_CONFIG_PATH=/usr/lib/pkgconfig:$DSTDIR/lib/pkgconfig:$PKG_CONFIG_PATH
   #-----------------
   # CONFIGURE
   #-----------------
   echo "Configuring lepton-eda (for details see build/lepton-eda-configure.log)"
   ../../source/lepton-eda/configure \
   --prefix=$DSTDIR \
   --build=x86_64-pc-cygwin \
   --host=x86_64-pc-cygwin \
   --target=x86_64-pc-cygwin \
   --disable-update-xdg-database \
   --enable-relocatable \
   --disable-nls \
   > ../lepton-eda-configure.log 2>&1
   #-----------------
   # MAKE step 1 -> This will exit with error on undefined reference to liblepton but create
   #                $DSTDIR/bin/cyglepton-1.dll $DSTDIR/lib/liblepton.dll.a & liblepton.la
   #-----------------
   echo "STEP 1/2 Building lepton-eda (for details see build/lepton-eda-make-s1.log)"
   make LDFLAGS="-no-undefined -lgio-2.0 -lgobject-2.0" > ../lepton-eda-make-s1.log 2>&1
   echo "Installing lepton-eda (for details see build/lepton-eda-install-s1.log)"
   make LDFLAGS="-no-undefined" install > ../lepton-eda-install-s1.log 2>&1
   echo "done"
   #-----------------
   # MAKE step 2 -> let add -llepton -lleptonrenderer and
   #                run make again to full build lepton-eda package
   #-----------------
   echo "STEP 2/2 Building lepton-eda (for details see build/lepton-eda-make-s2.log)"
   make LDFLAGS="-no-undefined -llepton -lleptonrenderer -lgio-2.0 -lgobject-2.0" > ../lepton-eda-make-s2.log 2>&1
   echo "Installing lepton-eda (for details see build/lepton-eda-install-s2.log)"
   make LDFLAGS="-no-undefined" install > ../lepton-eda-install-s2.log 2>&1
   echo "done"
   #-----------------
)

# Make /.local/share dir just because it requires privileges
mkdir $HOME/.local
mkdir $HOME/.local/share
#END
```
5) Add the following environmental variables to windows.
Change /home/mario/ with your cygwin home ($HOME) and adjust for your $DSTDIR
(mine is /cygdrive/c/cygwin64/home/mario/geda)
```
GEDABIN         =/cygdrive/c/cygwin64/home/mario/geda/bin
GEDADATA        =/cygdrive/c/cygwin64/home/mario/geda/share/geda
GEDADATARC      =/cygdrive/c/cygwin64/home/mario/geda/share/geda
LIBLEPTON       =/cygdrive/c/cygwin64/home/mario/geda/lib/liblepton
GUILE_LOAD_PATH =/cygdrive/c/cygwin64/usr/share/guile/2.0
DISPLAY         =:0
```
(you need a restart to make them active)

6) END

RUN SCHEMATIC
Before run lepton-schematic you need to start an X-server like cygwin one "XWin Server", there is an icon in start menu (I added in start-up).
After you can run "lepton-schematic" directly from windows or "gschem" from a bash shell "bash gschem"

NETLIST
Lepton-netlist is a guile script so you must run it from a bash shell
as example:
bash lepton-netlist -g tango -o schema.net schema.sch

P.S.
I'm basically an hardware designer so probably this task can be made better.

vzh, 2018-02-21:
@mario-giovinazzo This description (thanks to @graahnul-grom) is now [in our wiki](https://github.com/lepton-eda/lepton-eda/wiki/Lepton-EDA-and-Cygwin). Thank you!