lepton-eda/lepton-eda #153: v1.9.3 configuration howto is wanted for efficiency.

jgriessen, 2017-08-13:
By efficiency, I mean, "Not ready for results-required user making a OSHW board on a low budget."  Also with a low time budget, not just low money.

I saw a comment about using command line to set config files.
Then I put a question to the email list geda-user, then to irc #geda,
"Where do I find a howto on config files for letpon-eda?"
then I did other things, checked elists
and no answer, so I try to refind the  comment about using command line to set config files and cannot find it in 30 minutes.

So now I am working around this by using /symbols dir in each project, and no access to my other libraries without copying from my filesystem.

I did find and  start reading the info formatted doc on geda-scheme, but have not come across how to use config files -- more about inheritance and contexts with more reading required to understand...

I just remembered that the word migration was in that comment I was trying to refind...  Going to look for that.

vzh, 2017-08-14:
@jgriessen Probably nobody but me can answer this question here these days ;-) Though I'm also having a busy life, sorry. Probably asking at [gitter](https://gitter.im/Lepton-EDA) would give you a more quick answer.

As for the lepton configuration, not much things have changed since 1.9.2 (@dmn has added some new config setting though they are mostly for `lepton-schematic` (`gschem`)). If you have used 1.9.2, almost nothing should change. The main documents are `lepton-cli(1)` (`gaf(1)`) and the [gaf utility document](http://wiki.geda-project.org/geda:gaf_utility) I added a few years ago. Replace `gaf` with `lepton-cli` and voila ;-) Let me know if something doesn't work. Resource libraries have not been implemented yet, so we still use `gafrc` to define any libraries.

However, yes, we should definitely add a wiki document concerning the lepton configuration.

graahnul-grom, 2018-03-21:
@jgriessen @vzh Please take a look at this [wiki page](https://github.com/graahnul-grom/lepton-eda/wiki/Configuration-Settings) - an attempt to compile information about new configuration system. It would be great to hear your opinion and suggestions.

noqsi, 2018-03-21:
Very nice.

One thing that doesn’t quite fit all cases, though, is:

lepton-cli config option to access context: —project

My projects often have multiple gnetlist configurations. For example https://github.com/noqsi/gnet-spice-noqsi/wiki/Broadband-amplifier-board <https://github.com/noqsi/gnet-spice-noqsi/wiki/Broadband-amplifier-board> flattens hierarchy for layout, but keeps it for simulation.

> @jgriessen <https://github.com/jgriessen> @vzh <https://github.com/vzh> Please take a look at this wiki page <https://github.com/graahnul-grom/lepton-eda/wiki/Configuration-Settings> - an attempt to [...]

John Doty              Noqsi Aerospace, Ltd.
http://www.noqsi.com/
jpd@noqsi.com

graahnul-grom, 2018-03-27:
@noqsi Good point, thank you!
Just edited the page to indicate that `--project` option can take (optional) `PATH` argument.

graahnul-grom, 2018-04-15:
@jgriessen @vzh @noqsi 
I've uploaded the updated document about configuration system to our [wiki page](https://github.com/lepton-eda/lepton-eda/wiki/Configuration-Settings).
Please take a look. If everything is fine (any suggestions are welcome), I think we may close this issue.

graahnul-grom, 2018-05-23:
@vzh 
Vladimir, as I see no questions nor any suggestions, maybe we can close the issue?
Or you just have not enough time to look into this?

vzh, 2018-05-23:
@graahnul-grom There are some minor issues in your wiki document (that will be fixed in course of time, I believe), though I think it's good enough as a whole. Thank you very much!

P.S. Please update it with info about your recently added settings related to status bar ;-)

closed 2018-05-23 by vzh

graahnul-grom, 2018-05-24:
@vzh 
> update it with info about your recently added settings related to status bar

Done.