lepton-eda/lepton-eda #158: Height of One Pixel TextView

ehennes775, 2017-10-03:
Inside the multiattrib dialog box, the value textview has a height of one pixel. This issue was fixed in a commit and reintroduced in a later commit.

Original Fix:
42182f0ae6b16171329f00f32ff1857d38c1c33e

Reintroduction:
79d6da9e2dec05716570ae9b99ba65f5ddec5bbc

vzh, 2017-10-03:
@ehennes775 Does resurrecting of the line with "height-request" fixes the dialog on your system? (I cannot just reproduce the issue on my computer...)

ehennes775, 2017-10-04:
It should, and if you fix it, I will test it.

The height of the textview is defined by the scrollbar buttons. Since my system has popup scroll bar widgets, the height is one pixel.

I've experimented with letting the labels on the left define the height of the table row and setting the textview cell to GTK_FILL. So far, it does not match the size of the other widgets.

vzh, 2017-10-04:
Edward, please review #159.

closed 2017-11-05 by vzh

vzh, 2017-11-05:
@ehennes775 OK, the fix is in the master branch now. Please report if it works for you.

ehennes775, 2017-11-06:
The fix works.

vzh, 2017-11-06:
Great, thanks!

PR #159 (2017-11-05): Fix regression in multiattrib dialog
Closes #158