lepton-eda/lepton-eda #172: Single-Pin-Net in netlist for unconnected pins

mario-giovinazzo, 2018-02-14:
I have an unconnected pin generating DRC warning.
Connecting this pins to NC symbol solves but generate a single-Pin-net in netlist (tango)
that I must manually remove.
Is there a way to avoid this?
Thanks

graahnul-grom, 2018-02-15:
@mario-giovinazzo Could you please provide us with some test schematics and `lepton-netlist` output?

mario-giovinazzo, 2018-02-15:
This is a simple no-sense circuit,  pins U1-10 and U1-12 are unconnected.
[image]
Running “lepton-netlist -g drc2” will produce errors:
```
Checking non-numbered parts...
Checking duplicated references...
Checking NoConnection nets for connections...
Checking nets with only one connection...
Checking pins without the 'pintype' attribute...
Checking type of pins connected to a net...
Checking unconnected pins...
ERROR: Unconnected pin U1:10
ERROR: Unconnected pin U1:12
Checking slots...
Checking duplicated slots...
Checking unused slots...
No warnings found.
Found 2 errors.
```
And running “lepton-netlist -g geda” will produce a correct netlist:

```
START header
gEDA's netlist format
Created specifically for testing of gnetlist
END header

START components
CONN1 device=CONNECTOR_4
D1 device=LED
D2 device=LED
D3 device=LED
D4 device=LED
R1 device=RESISTOR
R2 device=RESISTOR
R3 device=RESISTOR
R4 device=RESISTOR
U1 device=7414
END components

START renamed-nets
END renamed-nets

START nets
GND : U1 7, U1 11, U1 13
NN_net1 : CONN1 3, U1 5
NN_net2 : CONN1 2, U1 3
NN_net3 : CONN1 4, U1 9
NN_net4 : CONN1 1, U1 1
NN_net5 : D2 2, U1 4
NN_net6 : D3 2, U1 6
NN_net7 : D4 2, U1 8
NN_net8 : D1 2, U1 2
NN_net9 : D1 1, R1 1
NN_net10 : D2 1, R2 1
NN_net11 : D3 1, R3 1
NN_net12 : D4 1, R4 1
Vcc : R1 2, R2 2, R3 2, R4 2, U1 14
END nets
```
Now connecting  unconnected pins U1-10 and U1-12 to nc-right-1.sym 
[image]

“lepton-netlist -g drc2” runs without errors:
```
Checking non-numbered parts...
Checking duplicated references...
Checking NoConnection nets for connections...
Checking nets with only one connection...
Checking pins without the 'pintype' attribute...
Checking type of pins connected to a net...
Checking unconnected pins...
Checking slots...
Checking duplicated slots...
Checking unused slots...
No warnings found.
No errors found. 
```
But “lepton-netlist -g geda” will produce a netlist with Single-Pin-Netlist (net-8, net-9) that I must manually remove:
```
START header
gEDA's netlist format
Created specifically for testing of gnetlist
END header

START components
CONN1 device=CONNECTOR_4
D1 device=LED
D2 device=LED
D3 device=LED
D4 device=LED
R1 device=RESISTOR
R2 device=RESISTOR
R3 device=RESISTOR
R4 device=RESISTOR
U1 device=7414
END components

START renamed-nets
END renamed-nets

START nets
GND : U1 7, U1 11, U1 13
NN_net1 : CONN1 3, U1 5
NN_net2 : CONN1 2, U1 3
NN_net3 : CONN1 4, U1 9
NN_net4 : CONN1 1, U1 1
NN_net5 : D2 2, U1 4
NN_net6 : D3 2, U1 6
NN_net7 : D4 2, U1 8
NN_net8 : U1 10
NN_net9 : U1 12
NN_net10 : D1 2, U1 2
NN_net11 : D1 1, R1 1
NN_net12 : D2 1, R2 1
NN_net13 : D3 1, R3 1
NN_net14 : D4 1, R4 1
Vcc : R1 2, R2 2, R3 2, R4 2, U1 14
END nets

```

graahnul-grom, 2018-02-16:
@mario-giovinazzo Thank you!
It looks like `lepton-netlist` does not completely ignore
"graphical" symbols (maybe issue #100 is related).
Just tested it with a simple schematic:
[image]
`lepton-netlist -g geda -o - issue172.sch`:
```
<skip>
START nets
GND : R1 1, R2 1
END nets
```
[image]
`lepton-netlist -g geda -o - issue172_with_nc.sch`:
```
<skip>
START nets
GND : R1 1, R2 1
unnamed_net1 : R1 2
unnamed_net2 : R2 2
END nets
```
[issue172_test_schematics.zip](https://github.com/lepton-eda/lepton-eda/files/1731336/issue172_test_schematics.zip)

graahnul-grom, 2018-02-16:
It seems that we have to delete nets with *NC* symbols (maybe
conditionally, depending on backend).
Will test and post the patch for `gnetlist.scm` tomorrow.
```scheme
; function: nets_delete_not_conected()
; delete nets if they have "no connect" symbol attached
;
; [netnames]: list of net names
; {return}:   updated list of net names
;
( define ( nets_delete_not_conected netnames )
( let
    (
    ( nc_dirs  #f ) ; "NoConnection" DRC directives list
    ( verbose  #f )
    ( result  '() ) ; empty list
    )

    ( if ( = (gnetlist:get-verbosity) 1 )
        ( set! verbose #t )
    )

    ( for-each
    ( lambda( netname )

        ( if verbose
            ( format #t " >> nets_delete_not_conected(): netname: [~a]~%" netname )
        )

        ( set! nc_dirs
            ( gnetlist:graphical-objs-in-net-with-attrib-get-attrib
                netname
                "device=DRC_Directive"
                "value"
            )
        ) ; => list of "value" attributes: e.g. "NoConnection"

        ( if verbose
            ( format #t "      nets_delete_not_conected(): nc_dirs: ~a~%" nc_dirs )
        )

        ( if ( member "NoConnection" nc_dirs )
            ; if:
            ( format #t "      nets_delete_not_conected(): deleting net: [~a]~%" netname )
            ; else:
            ( set! result ( cons netname result ) )
        )

    ) ; lambda()
    netnames
    ) ; for-each

    ; return:
    ( reverse result )

) ; let
) ; nets_delete_not_conected()
```

graahnul-grom, 2018-02-17:
@mario-giovinazzo Please try [this patch](https://github.com/graahnul-grom/lepton-eda/commit/e4eaa87ff6fda29a33e8984fe4e06a2a80226f56), or
simply replace `$INSTALLDIR/share/gEDA/scheme/gnetlist.scm` file with the one
from the archive attached.
[issue172_patch_0.zip](https://github.com/lepton-eda/lepton-eda/files/1733778/issue172_patch_0.zip)

mario-giovinazzo, 2018-02-17:
Hi graahnul-grom, 
thanks You for your great and rapidly work.
I tested the patch with a multi-sheet hierarchy design and it works fine.
Tomorrow I will make more test with more backend.

P.S.
The only aspect that may be evaluated is when the NC symbol is connected to a net with more than 1 pin (a real net), condition that of course is an error.
In this case DCR2 recognizes the error and the PATCH removes the intere net.
For me this is a good behavior,
but may be considered the opportunity of remove only the nets containing just 1 pin connected to NC.

mario-giovinazzo, 2018-02-19:
I have done various tests with various backend and the patch works fine.

graahnul-grom, 2018-02-21:
@mario-giovinazzo Thank you! I've filed a [PR with that changes](https://github.com/lepton-eda/lepton-eda/pull/176).
Also, we'd like to thank you very much for your Cygwin instructions. They are published  in [Lepton EDA wiki](https://github.com/lepton-eda/lepton-eda/wiki/Lepton-EDA-and-Cygwin).

graahnul-grom, 2018-10-12:
Testing PR #254: add sample schematics and `lepton-netlist` test script.

`issue172.sch`:

[image]

`issue172_with_nc_newsym.sch`

[image]

`test_backends_pr254.sh`:

```sh
#!/bin/sh

# test lepton-netlist:
# lepton-eda PR #254 (which should close issue #172 and pr #237)

sch=$1
exe="lepton-netlist"
tmp="/tmp/lepton-home-test-pr254"

if [ -z "${sch}" ]; then
    echo "Usage: `basename $0` sch_file"
    exit 1
fi

mkdir -p $tmp
rm -fr $tmp/.cache $tmp/.config
export HOME=$tmp

echo ">> Backends with 'unnamed_net' in output:"
for be in `$exe --list-backends 2>/dev/null`
do
    if $exe -g $be -o - $sch 2>/dev/null | grep -q "unnamed_net"
    then
        echo $be
    fi
done

echo ">> Backends with 'unconnected' in output:"
for be in `$exe --list-backends 2>/dev/null`
do
    if $exe -g $be -o - $sch 2>/dev/null | grep -i "unconnected"
    then
        echo $be
    fi
done
```

[issue172.sch.txt](https://github.com/lepton-eda/lepton-eda/files/2472630/issue172.sch.txt)
[issue172_with_nc_newsym.sch.txt](https://github.com/lepton-eda/lepton-eda/files/2472631/issue172_with_nc_newsym.sch.txt)
[test_backends_pr254.sh.txt](https://github.com/lepton-eda/lepton-eda/files/2472704/test_backends_pr254.sh.txt)

graahnul-grom, 2018-10-12:
`> test_backends_pr254.sh issue172.sch`

```
>> Backends with 'unnamed_net' in output:
>> Backends with 'unconnected' in output:
Checking unconnected pins...
ERROR: Unconnected pin R1:2
ERROR: Unconnected pin R2:2
drc2
R1 0 unconnected_pin-1 1k
R2 0 unconnected_pin-2 1k
spice
R1 0 unconnected_pin-1 1k
R2 0 unconnected_pin-2 1k
spice-sdb
```

`> test_backends_pr254.sh issue172_with_nc_newsym.sch`

```
>> Backends with 'unnamed_net' in output:
systemc
vams
verilog
vhdl
>> Backends with 'unconnected' in output:
Checking unconnected pins...
drc2
```

- in both cases: `geda` backend is supposed to output *all* nets
- in 4 backends `NC` symbol is not recognized

vzh, 2018-10-12:
@graahnul-grom, I don't understand, what's the issue with `unconnected' nets? The output is the same as it was before, because the schematic *issue172.sch* has not changed. It has no "nc" symbols.
Please, make it clear what's wrong with it?

graahnul-grom, 2018-10-12:
No issues with "unconnected", it's just for debugging.
'unnamed_net'  in the output is another matter.

graahnul-grom, 2018-10-13:
Those four backends are using `get-pins-nets()`; hence `unnamed_net` in their output.

`> git grep -l get-pins-nets netlist/scheme/gnet-*`

```
netlist/scheme/gnet-drc2.scm
netlist/scheme/gnet-systemc.scm
netlist/scheme/gnet-vams.scm
netlist/scheme/gnet-verilog.scm
netlist/scheme/gnet-vhdl.scm

```

graahnul-grom, 2018-10-13:
Should we fix those backends?

graahnul-grom, 2018-10-13:
@vzh What do you think?

[gnet-systemc.scm.diff.txt](https://github.com/lepton-eda/lepton-eda/files/2475426/gnet-systemc.scm.diff.txt)
[gnet-vams.scm.diff.txt](https://github.com/lepton-eda/lepton-eda/files/2475427/gnet-vams.scm.diff.txt)
[gnet-verilog.scm.diff.txt](https://github.com/lepton-eda/lepton-eda/files/2475428/gnet-verilog.scm.diff.txt)
[gnet-vhdl.scm.diff.txt](https://github.com/lepton-eda/lepton-eda/files/2475429/gnet-vhdl.scm.diff.txt)

vzh, 2018-10-14:
@graahnul-grom, thank you for your investigation and fixes! The real issue here is that we have to fix each backend according to its specs. I don't know if we can get some feedback from our users, though. I know the people use verilog (and verilog ams - vams), not sure about vhdl, and systemc had never worked (it's my feeling based on my previous work). So each of the backends should be fixed separately. A few years ago I worked on vams, and learned that it has lots of rules to make a correct netlist. Yesterday, I tried to find how the verilog backend should output not connected pins, read some specs. Probably, it should not omit unconnected pins, but output anything unconnected as a whitespace, e.g. `a,b, ,d`, please note two commas with no letter between them. I'm not sure, though. The issue requires a bit (or much) more work to do things properly. I propose to open one or more issues to fix them separately for each noticed backend, and close this issue. WDYT?

graahnul-grom, 2018-10-14:
Since everything except aforementioned backends works, I think we can close this issue, PR #237 and merge #254.

PR #254 (2018-10-15): Filter nets with no-connect symbol
This patch-set based on and overriding the commits in #237 fixes the issue #172.

Closes #237.
Closes #172.
closed 2018-10-15 by vzh