lepton-eda/lepton-eda #224: Freestyle colors should not be disabled by default

graahnul-grom, 2018-06-30:
Disabling that colors effectively prohibit their usage:
if users want to share schematic files, they also have to
share their color schemes, because otherwise some text
will be invisible to those with default setup.

graahnul-grom, 2018-06-30:
Besides, I would add another 4-6 freestyle colors, since 4 IMHO
is not sufficient.  However, I doubt we can do it now (if it will be decided at all),
due to backward compatibility that we have to maintain in the upcoming release.

vzh, 2018-07-23:
Though we can do it afterwards :-)
I'd use the same color notation as in Scheme files (colormaps) to change colors of arbitrary objects.

closed 2018-07-23 by vzh

graahnul-grom, 2018-08-01:
@vzh 
> I'd use the same color notation as in Scheme files (colormaps) to change colors of arbitrary objects.

Are you talking about using that notation in *.sym and *.sch files instead of numbers (i.e. changing the file format)?

vzh, 2018-08-15:
> Are you talking about using that notation in *.sym and *.sch files instead of numbers (i.e. changing the file format)?

Yes, though we don't have to change the existing format. We could extend it transforming it into something new in time provided that we change its version (read, the header line).

PR #228 (2018-07-23): Enable freestyle colors
Closes #224