lepton-eda/lepton-eda #225: Reorganize the main menu

graahnul-grom, 2018-07-03:
A. `File` menu:
  1. delete duplicated `Revert` (already in `Page` menu)

B. `Edit` menu:
  1. combine to one item: `Object Properties...`:
    - `Color...`
    - `Line Width & Type...`
    - `Fill Type...`
    - `Pin Type...`
  2. move `Show/Hide Inv Text` to `Attrubute` menu

C. `View` menu:
  1. rename: `Sidebar` => `Right Dock`
  2. rename: `Status` => `Bottom Dock`
  3. add new item: `Find Text State`

D. `Page` menu:
  1. separate `Revert` from other items

E. `Hierarchy` menu:
  1. delete duplicated `Documentation...` (already in `Help` menu)

F. `Options` menu:
  1. delete obsolete `Text Size...` item
  2. rename/rearrange, for example:
    - `Snap Grid Spacing...` => `Options...`
    - [separator]
    - `Scale up Grid Spacing` => `Grid +`
    - `Scale down Grid Spacing` => `Grid -`
    - [separator]
	- `Cycle grid styles` => `Grid style: cycle Dots/Mesh/Off`
    - `Toggle Snap On/Off` => `Grid snap: cycle Grid/Resnap/Off`
    - [separator]
    - `Toggle Outline/Box` => `Feedback mode: Outline/Box`
    - `Toggle Net Rubberband` => `Net: Rubberband On/Off`
    - `Toggle Magnetic Net` => `Net: Magnetic On/Off`
    - [separator]
    - `Show Coords Window...` => `Coords Window`
    - `Show Log Window...` => `Log Window`

Comments:
- B.1: all that commands do exactly the same: show `Object` dialog
box (`x_widgets_show_object_properties()`).
- C.3: currently, there is no way to show `find text state` window
unless docking windows GUI is enabled. We need to add a new action,
e.g. `view-find-text-state`.
- F.2: `Options` menu:

[image]

[image]

Any suggestions are welcome.

graahnul-grom, 2018-08-31:
If anyone wants to try the menu modified as proposed, here's the diff attached.

[system-gschemrc.diff.txt](https://github.com/lepton-eda/lepton-eda/files/2341191/system-gschemrc.diff.txt)

vzh, 2018-09-04:
Several notes after your suggestions:

- Should we add `&object-properties` Scheme function and replace several `&edit-something` with it? WDYT?

- I'd like "Sidebar" better than "Right Dock" (and why could not it be left, for example?). However, I don't quite understand what could be used instead of "Status" then. Especially, when the user doesn't use embedded dialogs.

- Why have you thrown out &hierarchy-documentation?

- Why have you thrown out &options-text-size?

Otherwise, I'm pretty happy with your changes.

BTW, it's strange that `OL` is not displayed for the "Show log window..." menu entry.

graahnul-grom, 2018-09-05:
* Yes, I think we should replace `edit-color`, `edit-linetype`, `edit-filltype` and `edit-pin-type` with new `object-properties` action.
* Frankly, I do like "Sidebar" name, too. But `Right Dock` and `Bottom Dock` names precisely describes what we have now: rigid docking widgets that cannot be configured in any way. And these menu items makes no sense when `use-docks` is set to `false`. Some time ago I proposed (on gitter) to split up `system-gschemrc` file: that way it would be easier to dynamically generate the main menu in, say,  `system-gschemrc-menu.scm` file (taking into account user configuration).

* > Why have you thrown out &hierarchy-documentation?

I see no relation between hierarchy and component documentation. And that item is already present in `Help` menu.

* > Why have you thrown out &options-text-size?

When the users see such an item in `Options` menu they might expect that it should somehow affect the global settings (like `text-size` in `gschemrc`). But instead it just opens `Edit Text` dialog (the same as `Edit->Edit Text...`).

`O L` key sequence is rebound to another action in my configuration.

graahnul-grom, 2018-09-05:
Some further modifications:

* rearrange some items
* add/remove `...` where appropriate
* add more separators
* move `Edit->Invoke Macro` => `File->Invoke Macro...`
* move `Edit->Show/Hide Inv Text` => `Attributes->Show/Hide Hidden Text`
* rename `Component Documentation...` => `Find Component Documentation`

[system-gschemrc_2.diff.txt](https://github.com/lepton-eda/lepton-eda/files/2352326/system-gschemrc_2.diff.txt)

graahnul-grom, 2018-11-19:
@vzh I'll add a PR, though it would be hard to split it up.

Updated menu:
  - system-gschemrc: [system-gschemrc.txt](https://github.com/lepton-eda/lepton-eda/files/2595634/system-gschemrc.txt)
  - diff: [system-gschemrc_3.diff.txt](https://github.com/lepton-eda/lepton-eda/files/2595637/system-gschemrc_3.diff.txt)
  - screenshots (left - old, right - new):

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

vzh, 2018-11-29:
For some reason, #281 did not automatically close the issue, so I do it manually.

closed 2018-11-29 by vzh