lepton-eda/lepton-eda #231: utils/xyrs: 'make check' fails with perl >= 5.26

graahnul-grom, 2018-07-23:
After `perl` upgrade (to 5.26) `make check` starts to fail in `utils/tests/gxyrs`
(first two tests: `do_nothing_tabulated`, `do_nothing`).
This is due to removal of the current directory (`"."`) from `@INC` in `perl` 5.26
(see [perl 5.26 release notes](https://metacpan.org/changes/release/XSAWYERX/perl-5.26.0#Removal-of-the-current-directory-(%22.%22)-from-@INC)).
As a temporary fix we could `export PERL_USE_UNSAFE_INC=1` while running `xyrs` tests.

vzh, 2018-07-23:
Ideally, your fix description should consist of "what" (title) and "why" (description) parts. Please include such a description as the above inside commits to make them more comprehensible. Thanks!
BTW, perl 5.24.1 on debian stable also has this issue.

closed 2018-07-23 by vzh

graahnul-grom, 2018-07-30:
You are right, I should have included the above paragraph in the commit message.
(At the time, may be, I thought the comment in the file is sufficient.)
Strange, but for perl 5.24.4 (FreeBSD 11.1-STABLE) the issue does not manifest itself.

PR #232 (2018-07-23): xyrs: Fix 'make check' for perl >= 5.26
Closes #231