lepton-eda/lepton-eda #284: Improve documentation

graahnul-grom, 2018-12-02:
1) Most of the `README` file content should be moved to
a new `INSTALL` file. This file should also include a link
to the `Lepton EDA and Cygwin` wiki page.
2) `README` should be extended to provide more information
about the `Lepton EDA` suite:
  - it was forked by main `geda-gaf` developers
  - it is compatible with (and direct replacement for) `geda-gaf`
  - list of new features: link to `NEWS` file
  - what systems are supported
  - binary packages available
  - how to build it (link to `INSTALL` file)
  - link to (wiki) page with screenshots (maybe embed 1-2 screenshots
in `README` itself)
3) Some documents should be included in the installation
(`share/doc/lepton-eda/`):
  - `README`
  - `NEWS`
  - `Configuration settings`, `Attributes` wiki pages (converted to HTML (?))
4) A new `Configuration` (HTML) document should be added;
add a new `Help->Configuration` menu item for it.
This document should contain:
  - description of `component-library` and `source-library` options and
where to place them. This is the most misunderstood subject. Wiki page
(`wiki/geda-gschem_ug-config.html`) is not clear enough on that.
  - links to all files (local wiki, man pages, maybe web pages)
with information on how to configure various tools in the `Lepton EDA` suite.

graahnul-grom, 2018-12-18:
One more item to add to the `README` file:
- `lepton-eda` can easily be extended through the rich (and very well documented) `Scheme` API (`GNU Guile`).

I'd also consider adding `guile` to the list of "topics" (keywords shown at the top of the `github` repo page).

vzh, 2018-12-18:
I've added `guile`, as well as `c` and `geda` to the topic list.

graahnul-grom, 2019-05-19:
I think it's also worth installing the Scheme API reference manual HTML files.
We already installing HTML man page, so why not?

vzh, 2019-05-20:
Agreed. While I myself like plain info documentation, some users could prefer HTML.

graahnul-grom, 2019-11-12:
Having four help items in the `Help` menu is excessive.
They all point to essentially the same outdated wiki documentation
pages, one click away one from the other.
I think just one item (`gEDA Wiki`) would be enough.

vzh, 2019-11-12:
The menu should be reviewed, I believe. We have to somehow refactor the old docs from gEDA wiki and merge it with our up-to-date documentation (if any). I think we could move the docs first to our wiki, and then probably to another repository.

graahnul-grom, 2020-02-09:
@vzh So, should we remove those excessive items now?

[image]

graahnul-grom, 2020-02-10:
`github` wiki is pathetic. It can't be helped.
I seriously doubt that it's worth our time and
effort to make content for it. Most likely, it
will result in double work in near future.
Look how ugly those shrunk images are [1] [2];
and display size doesn't matter. As I said
elsewhere, we can forget about tables and pictures
(especially, screenshots: "Hey, folks, see how
scary our app looks like!").
When the user opens our wiki, he is greeted with
the half-screen github advertisement banner (in
the best traditions of malicious rogue sites),
which is absolutely irrelevant. Bad impressions
are guaranteed.

[1] https://github.com/lepton-eda/lepton-eda/wiki/Screenshots
[2] https://github.com/lepton-eda/lepton-eda/wiki/Verilog-example