lepton-eda/lepton-eda #299: Attribute parameter substitution patches

vzh, 2018-12-18:
(This is launchpad bug https://bugs.launchpad.net/geda/+bug/911440)

Stephan Böttcher writes:

> here's a patch to gnetlist that implements parameter substitution in
> hierarchical schematics. The idea is that some attribute values in the
> child page are defined in terms of parameters, that can be substituted
> in individual instances.
>
> A resistor in the sub-schematic may have the attribute
>
> value=$RFB
>
> and the instance symbol for that sub-scematic in tha parent schematic
> may have the attibute
>
> parameter=RFB:12k

He continues:

> a extension of the previous patch to allow parameter names to contain
> digits, and allow references to arbitrary parameter names in ${braces}.

felix-salfelder, 2025-05-12:
This is interesting, because it was motivated by the same issue that lead to #1081.

In gEDA/Lepton, the "value" attribute has been in use to make parameter assignments (aka. substitutions, aka. overrides).

It is a bad idea to have two different mechanisms to serve a single purpose. You quickly need heuristics to guess the users intent. Here: If the resistor instance has/is a sub-schematic, should we use "value" or "parameter" to make assignments?

The solution here is to just stick to "value"

    value=RFB=12k.

Note the '=' character, and how the attribute value will display. The assignment notation has been adopted in all areas of computing, its use relates to the intent here. It does abuse the equality notation, but there is little room for confusion. The proposed colon notation is also found in net attribute values where it follows its own and different logic.

There is no need to allow references to arbitrary parameter names. Declare a parameter using 'symbol=schematic-parameter', and a reference to it is made by naming its refdes, e.g.

    value=RFB=my_parameter_refdes.

> allow parameter names to contain digits, and allow references to arbitrary parameter names

Allowing parameter names to contain digits is asking for *identifiers containing arbitrary characters*. This needs a global solution if desired, not one that works for parameter identifiers only. Relevant questions: What are the rules for identifiers in Lepton? How do others do it? etc.

felix-salfelder, 2025-05-14:
On Tue, May 13, 2025 at 09:58:36AM -0700, Vladimir Zhbanov wrote:
> Well, for me it's an easy part.  Just select a preferred solution and
> mark another one not recommended or deprecated.  Having a good test
> suite and docs for both ones would be harder, though still solvable.

Not as easy if you are not in control of these recommendations.

> IMO, these different mechanisms of declaring net names using either
> symbols or nets  also make the development only harder.  This is also
> solvable by declaring a preferred way of doing things.

Yes, I can see the problem. Connectivity is a bit off topic here. If you
are interested in how we deal with it, see [1].

> Honestly, for me, there is no difference, if it is a symbol with the
> `symbol=schematic-parameter` attribute, or with a floating attribute
> `schematic-parameter=symbol-name` and others, if it does its job
> properly. Even more, I don't like drawing special boxes for parameters
> and prefer any notations to be plain text ones, as ISO/IEC standards
> show it.   However, I wouldn't mind if you implement any working
> approach.

I see the point about the additional boxes.  I do not know what you mean
by "floating attribute". Sounds interesting though. How would I use
this?

IEEE1364 aka Verilog requires interface parameters to be stated
explicitly. Guessing them from special syntax may be tricky and become
more tricky if more data is associated with a parameter. An instance of
the proposed parameter-1.sym was meant to also carry type and range and
such things...

> [identifiers] This is/was the biggest issue in Lepton/gEDA.

For good reasons, every programming or modelling language spec starts
with a chapter about valid identifiers. Maybe worth figuring that out
sooner than later. IEEE1364 may serve as inspiration. They have a simple
but effective "escape syntax" which allows to break away from basic
identifiers in a controlled way.

- pick a set of (recommended) identifiers
- pick your favourite escape syntax, once and for all
- get arbitrary strings in all identifiers

There are situations where people should put aside personal syntax
preferences and focus on content. With that in mind you could just
recommend IEEE1364 rules and call it a day.

[1] http://gnucap.org/dokuwiki/doku.php/gnucap:user:netlist_import_and_export:geda?&#connections_by_attribute