lepton-eda/lepton-eda #300: Trouble building developer docs

noqsi, 2018-12-18:
I can't get the developer documentation to build for lepton-eda-1.9.7 on Ubuntu Xenial. LaTeX emits a thousand errors and finally gives up.

[untitled.txt](https://github.com/lepton-eda/lepton-eda/files/2692091/untitled.txt)

graahnul-grom, 2018-12-19:
@noqsi Thank you for the report. I cannot generate the latex output on ubuntu (18.04, "bionic"), too. However, html output is fine. For a quick fix please try the attached patch.
What it does:
- disables generation of latex output files
- excludes `netlist` and `symcheck` subdirs: these tools were rewritten in `Scheme`; AFAIK doxygen doesn't support that language

[doxy_ubu1804.diff.txt](https://github.com/lepton-eda/lepton-eda/files/2694420/doxy_ubu1804.diff.txt)

To apply the patch:
- `cd` to `lepton-eda` source directory
- `git apply doxy_ubu1804.diff`

noqsi, 2018-12-21:
This seems to work. Thanks.

PR #309 (2019-01-03): Fix doxygen issues
Get rid of doxygen files related to symcheck and netlist.
Do not generate latex files when making doxygen documentation.

Closes #300
closed 2019-01-03 by vzh