lepton-eda/lepton-eda #306: Segfault from old gafrc

noqsi, 2018-12-21:
In Lepton 1.9.7 on Ubuntu Xenial, I get a segfault if gafrc has the line:

(always-promote-attributes "refdes")

I realize that this has been a deprecated construction for some time (wants a list, not a bare string), but it shouldn't segfault.

graahnul-grom, 2018-12-21:
```patch
diff --git a/liblepton/src/g_rc.c b/liblepton/src/g_rc.c
index 27dfa8c38..dbe3a62bf 100644
--- a/liblepton/src/g_rc.c
+++ b/liblepton/src/g_rc.c
@@ -863,7 +863,7 @@ SCM g_rc_always_promote_attributes(SCM attrlist)
     gchar **split_names = g_strsplit(attr_string," ", 0);
     free (attr_string);
 
-    for (gchar **iter = split_names; iter; ++iter) {
+    for (gchar **iter = split_names; *iter; ++iter) {
       if ((*iter)[0]) /* Only accept non-empty strings */
         g_ptr_array_add (promote, (gpointer) g_intern_string (*iter));
     }
```

Thank you, John!

vzh, 2018-12-21:
@noqsi, please check if the suggested patch fixes the error.

noqsi, 2018-12-21:
It fixes the error I saw, with a deprecation warning in the log. On the other hand, it's still possible to provoke a segfault. I tried some other cases:

(always-promote-attributes )
Displays error, continues.

(always-promote-attributes 0)
Displays error, segfaults.

(always-promote-attributes #f)
Displays error, segfaults.

(always-promote-attributes '())
No error.

(always-promote-attributes ())
Displays error, continues.

(always-promote-attributes "")
Deprecation warning in log, continues.

closed 2018-12-22 by noqsi

vzh, 2018-12-23:
@noqsi Thank you!
One note about our current commit policy:
Please use the *rebase and merge* option next time you merge something,  (or rebase, if needed, and fast-forward merge, if you won't use github's web-interface) to not pollute the git log history with unnecessary commits and branches.
(And I'm looking forward for committing your backends into lepton :-) I've not found time myself yet :-/)

PR #307 (2018-12-22): Fix segfault in liblepton
Segmentation fault happens when a string is passed to the
`always-promote-attributes` function (called from the `gafrc` configuration file)
instead of a string list.

Closes #306