lepton-eda/lepton-eda #358: Build fail with o_net.c:175:36: error: ‘current_rules’ may be used uninitialized

domichel, 2019-03-18:
On a gentoo system with gcc-8.2.0, the last git code for lepton-eda fail to build with
`make[4]: Entering directory '/var/tmp/portage/sci-electronics/lepton-eda-9999/work/lepton-eda-9999/schematic/src'

...snip...

x86_64-pc-linux-gnu-gcc -DLOCALEDIR=\"/usr/share/locale\" -DHAVE_CONFIG_H -I. -I../..  -I../../liblepton/include  -I./../include -I../.. -I/usr/include  -Wall -Werror=uninitialized -std=c99    -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -pthread -I/usr/include/gtk-2.0 -I/usr/lib64/gtk-2.0/include -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/libdrm -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/guile/2.0 -march=native -O2 -pipe -c -o lepton_schematic-o_net.o `test -f 'o_net.c' || echo './'`o_net.c
o_net.c: In function ‘o_net_guess_direction’:
o_net.c:175:36: error: ‘current_rules’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
    right = MAX(right, current_rules[1]);
                                    ^
cc1: some warnings being treated as errors
make[4]: *** [Makefile:2011: lepton_schematic-o_net.o] Error 1`

vzh, 2019-03-19:
Hi @domichel,
Strange, while on Debian, I cannot reproduce this neither with gcc 6.3, nor with gcc 8.3. I tried to use `-Werror=maybe-uninitialized`, as is seen in your report, either, with no effect.
Could you provide us with your gcc default options, please (`gcc -Q -v somefile.c`)?

vzh, 2019-03-19:
BTW, you could try the patch as follows:
```
diff --git a/schematic/src/o_net.c b/schematic/src/o_net.c
index c7f05f4764..95e90a3fc4 100644
--- a/schematic/src/o_net.c
+++ b/schematic/src/o_net.c
@@ -120,6 +120,7 @@ void o_net_guess_direction(GschemToplevel *w_current,
         current_rules = (int*) bus_rules;
         break;
       default:
+        current_rules = (int*) net_rules;
         g_assert_not_reached ();
       }
```
Though I'm not sure it works and even if so, that you won't have likewise issues somewhere else.

domichel, 2019-03-19:
Hi vzh
`# gcc -Q -v somefile.c
gcc: error: somefile.c: No such file or directory
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-pc-linux-gnu/8.2.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /var/tmp/notmpfs/portage/sys-devel/gcc-8.2.0-r6/work/gcc-8.2.0/configure --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/gcc-bin/8.2.0 --includedir=/usr/lib/gcc/x86_64-pc-linux-gnu/8.2.0/include --datadir=/usr/share/gcc-data/x86_64-pc-linux-gnu/8.2.0 --mandir=/usr/share/gcc-data/x86_64-pc-linux-gnu/8.2.0/man --infodir=/usr/share/gcc-data/x86_64-pc-linux-gnu/8.2.0/info --with-gxx-include-dir=/usr/lib/gcc/x86_64-pc-linux-gnu/8.2.0/include/g++-v8 --with-python-dir=/share/gcc-data/x86_64-pc-linux-gnu/8.2.0/python --enable-languages=c,c++,fortran --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo 8.2.0-r6 p1.7' --disable-esp --enable-libstdcxx-time --enable-shared --enable-threads=posix --enable-__cxa_atexit --enable-clocale=gnu --enable-multilib --with-multilib-list=m32,m64 --disable-altivec --disable-fixed-point --enable-targets=all --enable-libgomp --disable-libmudflap --disable-libssp --disable-libmpx --disable-systemtap --enable-vtable-verify --enable-libvtv --enable-lto --with-isl --disable-isl-version-check --enable-libsanitizer --enable-default-pie --enable-default-ssp
Thread model: posix
gcc version 8.2.0 (Gentoo 8.2.0-r6 p1.7) 
`

Your patch seam to work fine here. gcc succeeded to build lepton_schematic-o_net.o and the build succeeded.

domichel, 2019-03-19:
Thank you.

BTW, about the python dependency, does lepton need it only during the installation, or also at runtime?
I am making an ebuild (I like when portage is able to manage my installed software), it I need to know it in order to follow the gentoo policies into that ebuild.

domichel, 2019-03-19:
I am impressed, because gschem was unusable for me from the time they introduced the side dock, which imply I am still using gEDA-1.8.2 . With lepton, it is very slow the first time I click on a component, but after that, the speed of use is quite decent and I can shift the side dock on and off without problem. Very nice work, thank you!

vzh, 2019-03-19:
>  Your patch seam to work fine here. gcc succeeded to build lepton_schematic-o_net.o and the build succeeded.

OK, thanks for the report!

> BTW, about the python dependency...

For now there exist two utils written in Python in the repo, `lepton-archive` and `lepton-tragesym`.

graahnul-grom, 2019-03-19:
Can't reproduce it (gcc 8.3.0, FreeBSD).

@domichel Thank you. I had exactly the same problem: cannot use `gschem` after the docking widgets were introduced. That was solved in #135. As for the slow first startup, there's the solution (#339) to this problem, we are just discussing it. It would be interesting to hear you opinion on that subject, too.

PR #359 (2019-03-19): Fix build fail on Gentoo with gcc-8.2.0
Closes #358
commit 3da6705b (2019-03-19): schematic: Fix build failures on some systems.

Fixes the build error on a Gentoo system with gcc-8.2.0 as
follows:

o_net.c:175:36: error: ‘current_rules’ may be used uninitialized
in this function [-Werror=maybe-uninitialized]

Closes #358.
closed 2019-03-19 by vzh