lepton-eda/lepton-eda #360: No valid main category into the desktop files.

domichel, 2019-03-21:
lepton-eda provide 2 desktop files in /usr/share/applications and they don't provide any valid main category. As example:

`# cat  /usr/share/applications/lepton-schematic.desktop
[Desktop Entry]
Version=1.0
GenericName=lepton-schematic
Name=Lepton EDA Schematic Editor
Comment=Create and edit electrical schematics and symbols with lepton-schematic
Type=Application
Exec=lepton-schematic %F
Icon=lepton-schematic
MimeType=application/x-lepton-schematic;application/x-lepton-symbol;
Categories=Engineering;Electronics;
`

According to [Additional Categories
](https://specifications.freedesktop.org/menu-spec/latest/apas02.html), both Engineering and Electronics are additional categories.

The problem is that without a main category, it is not possible to know for sure if a freedesktop compliant menu will show these 2 applications. Some menu system incorporate a catch all "Other" category, which is not in the [spec](https://specifications.freedesktop.org/menu-spec/latest/apa.html) and the lepton applications will be shown here, but other like fvwm-crystal or xfce just follow the spec and does not provide that catch all category (fvwm-crystal) or provide them but are unable to catch the lepton desktop files (xfce). In such cases, the lepton apps will be ignored.

An easy fix would be to use
Categories=Science;Engineering;Electronics;
or
Categories=Education;Engineering;Electronics;

domichel, 2019-03-21:
I just constated that the Science category was an additional category in the past, and is now a main cat. In my system, I just modified fvwm-crystal to be aware of that, and "Categories=Science;Engineering;Electronics;" work fine now. But it still doesn't work with xfce, which imply xfce is not up to date with that situation. That imply it would be best to use 2 main cats, like:
"Categories=Science;Education;Engineering;Electronics;"
or
"Categories=Science;Development;Engineering;Electronics;"

domichel, 2019-03-21:
Last, I would prefer to have something like
"Categories=<MainCat>;Electronics;Engineering;"
because the first additional category will get priority, and I prefer to get on my electronics related apps into the same submenu, so that they don't get mixed with other engineering tasks. But this is a detail that is not very important, I can always change my application menu. 
In fact, I am not even sure than engineering is needed, because engineering is more about control and measure than about development or conception for me. But again, that's a detail.

vzh, 2019-03-21:
So, probably the best variant from the above is `Categories=Development;Electronics;Engineering;`, right?

domichel, 2019-03-21:
> So, probably the best variant from the above is `Categories=Development;Electronics;Engineering;`, right?

Yep, I agree fully!

PR #361 (2019-03-25): Add main category into desktop files.
According to
https://specifications.freedesktop.org/menu-spec/latest/apa.html,
previously there was no valid main category in the desktop files.

Issue description in #360:
"The problem is that without a main category, it is not possible to
know for sure if a freedesktop compliant menu will show these 2
applications."

This has been fixed by adding "Development" as main category for
Lepton. Two additional categories have been rearranged in order
"Electronics" to have priority over "Engineering".

Closes #360
closed 2019-03-25 by vzh