lepton-eda/lepton-eda #369: Reasonable defaults

graahnul-grom, 2019-04-06:
After more than two years of collective effort that was put
into making it better, `Lepton EDA` has become quite different
compared to what it was forked from.
However, it's hard to spot the difference at a glance:
almost all new features are disabled by default
(as if we so shy to show them) and, on the other hand,
old, obsolete settings are retained.
Before making a new release I propose to change some defaults.

1. Turn off cursor warping: `(warp-cursor "disabled")`.
In `conf/schematic/options.scm` it is referred to as
"annoying". It's not "annoying", it is outright irritating.
Those who like it knows very well how to turn it on, I'm sure.
But when the infuriated (new) user rushes to `Options` menu,
eager to immediately turn this devil's invention off, what
will he see? There are no configuration settings in `Options`
menu, just some volatile UI tweaks. What then? Try kicad?
Thank you very much. Their schematic editor is not an application,
but an instrument of torture. Could easily take its place in
[torturemuseum.com/instruments-of-torture](http://www.torturemuseum.com/instruments-of-torture/).
It's not worth to push users to that route.

2. Turn off fast mousepan: `(fast-mousepan "disabled")`.
20 years ago it was quite relevant, but with modern (and not
so modern) hardware, drawing boxes instead of text while panning
is not necessary. In exceptional cases (extremely old hardware,
insanely large schematics) it can be enabled, of course.

3. Do not show the log window on startup: `(log-window "later")`.
It may be not so important with docking windows GUI,
but otherwise it's quite annoying.

4. Increase the number of undo steps: `(undo-levels 20)`.
Now it's set to 10. Should be at least 20, I think.

5. Mouse panning with the right mouse button: `(third-button "mousepan")`.
Not only it's inconvenient to move mouse while holding
the middle button when it's actually a wheel (on some mice it may even
tilt), panning with the right mouse button is used in many pcb layout
tools. The advantage of having consistent mouse buttons mapping
becomes obvious when working on pcb project and switching frequently
between schematic and layout editors.
Context menu can be shown on middle-click: `(middle-button "popup")`.
Not a big deal: we have to admit that there is little use for our context
menu at present.

6. Enable tabbed GUI: `schematic.gui::use-tabs = true`.
Most modern GUIs use tabs: editors, web browsers, etc. It's convenient
and friendly interface. It was introduced more than a year ago and now,
I think, is tested enough (well, at least, by me).

7. Enable rubberband and magnetic net mode status bar indicators:
`schematic.status-bar::show-rubber-band = true`
`schematic.status-bar::show-magnetic-net = true`
Knowing the rubberband state is especially useful.

8. Use monospace font for the log window, e.g.:
`schematic.log-window::font = monospace 10`

9. Replace `Arial` font name with `Sans` in:
 a. configuration: `export::font` key
 b. `libleptonrenderer/edarenderer.c`
 c. `lepton-schdiff`
On UNIX systems one can hardly find `Arial` font. It means that
replacement will be chosen by launching `fc-match(1)` from the
`fontconfig` package, and it not always result in the best match.
It should not cause any problems on `Windows`: at least, I checked
it on `Windows 7`. `Sans` looks more appropriate.

10. Remove symbols with duplicated names from the library.
There are three sub-directories that contain symbols with
conflicting names: `gnetman/`, `verilog/` and `vhdl/`. Although
their inclusion is commented out in `symbols/geda-clib.scm`, they
will almost invariably cause problems: it's common to
`(reset-component-library)` in `gafrc` file to clear the environment
and then add project-specific libraries; if one tries then to load
lepton stock symbols by `component-library-search`, this pulls in
those directories and starts flooding the log with "duplicated symbol"
warnings, distracting and annoying the user.
I suggest either:
  a. delete them
  b. move those folders out of `sym/` (to e.g. `sym-misc/`)
  c. rename symbols. For example, by adding a prefix: `gnetman-`, `verilog-`, `vhdl-`
I understand that we should better redesign the library loading
mechanism, but this requires a lot of thought and then a lot of work.
Until it is done, the problem should be solved, any way.

11. Add `value` and `footprint` attributes to basic symbols.
There are basic symbols in `sym/analog/`; new user most likely start
their work with lepton by using them. The only attribute they have
(most of them) is `refdes`. What's the use for symbol which has
neither `value` nor `footprint` attribute? I propose to add these
attributes without bumping `symversion`.

12. Although this is highly subjective, I would disable
the docking GUI by default. The idea itself is very good, no doubt,
but currently this feature is in its early stages of development:
no keyboard focus control, no configuration available, cannot
rearrange docks, etc. etc. etc. Personally, I would not put things
in such state into production, but rather mark them as
"experimental", invite users to test them, providing a way
to turn them on.

@all Please vote, comment and provide your suggestions.

graahnul-grom, 2019-04-16:
@bdale Does `pcb-rnd` do panning with the middle mouse button? And what are you struggling with when working with `lepton-schematic`/`pcb-rnd` most?
About symbols: I do not use stock symbol library, too. The above changes targeted mostly at new users of `Lepton EDA`. The point here is that it's easier to change value of the provided attribute, than to add a new one (in addition, new attributes are placed at some odd positions - another problem we have to solve). Having some default attributes also helps to quickly create some test schematics.

bdale, 2019-04-18:
Yes, pcb-rnd does panning with middle mouse button.  I'm sure it's configurable, but to regain my sanity, at some point I guess I'll have to decide what I like best and configure some set of command keyboard accelerators that both tools understand...

graahnul-grom, 2019-04-18:
@bdale Addressing the same problem wrt sanity, long ago I decided to not use
standard keyboard shortcuts in both schematic and layout editors, but to develop
a set of my own keybindings for the most frequently used commands.
Though it took quite some time and effort (and I'm still changing things from time
to time), I'm rather comfortable with the result.
There are some intersections (e.g. "show library window", "toggle grid"), but the
majority of commands are specific to a particular tool.

graahnul-grom, 2019-05-19:
@vzh Could you please express your opinion of the proposed changes?
I'm going to start changing the defaults, if you don't mind.

vzh, 2019-05-20:
Not sure about 5. I'm used to default mouse settings.

Wrt 10, I would not remove the symbols, several people reported
they work with them.  I would prefer moving them to some other
place for now.  If we rename the symbols, the names should be
short, otherwise they would look ugly.

Wrt 11, I disagree with changing symbols without bumping
symversion=. At least the minor version should change.  I believe
we have to carefully choose default value= and footprint= for any
symbol. That's a lot of work though we could start it.

I'm fine with all other proposed changes.

graahnul-grom, 2019-05-21:
> Wrt 10, I would not remove the symbols, several people reported
they work with them. I would prefer moving them to some other
place for now.

Another thought: could we modify `component-library-search` to make it skip those directories?
If we about to move them, what name of directory do you think should be (I proposed `sym-misc/` above)?

vzh, 2019-05-21:
> Another thought: could we modify `component-library-search` to make it skip those directories?

I don't think it's a good idea to skip arbitrary directories. I'd suggest another approach to resolve this issue:
- add a set of attributes which could uniquely (well, in most cases) determine what library they belong to, and process them as `symversion=` is processed now. E.g., `author=vzh`, `library-name=verilog`, `library-version=1.0`. As a first-shot approach, this `library-name=` might be the basename of the directory the symbol in question resides in. Later, we could use some file/hash/file-with-hash within the directory to uniquely define it.

> If we about to move them, what name of directory do you think should be (I proposed `sym-misc/` above)?

I'm not a fan of short names, you know :-) I would prefer `symbols-verilog`, etc. in the toplevel directory (to always remind me this should be fixed some time).

graahnul-grom, 2019-05-23:
I forgot about the controversial `undo-panzoom` `gschemrc` option:
It should be disabled by default, too, I believe. I wouldn't surprise new users with such exotic features.

There are related discussions on gitter (April, 2017), in #114, #117

vzh, 2019-05-27:
Agreed.

graahnul-grom, 2019-06-26:
13.  Mouse buttons assignment indicators should not be shown by default.
Such assignments usually done only once (using
appropriate `gschemrc` options), and there
is no point in displaying these static strings,
cluttering the status bar.

graahnul-grom, 2021-03-10:
I'm closing this, since most of the points listed above have
been addressed. The remaining no. 11 (about symbols)
should be moved to a new issue, and assignment of the
mouse buttons (no. 5) is not a big deal: GUI settings
have been migrated to new (well documented, BTW)
[configuration system](https://github.com/lepton-eda/lepton-eda/wiki/Configuration-Settings). It's quick and easy to change such
parameters with [lepton-cli](https://graahnul-grom.github.io/ref-man/lepton-cli.html) or [lepton-conf](https://graahnul-grom.github.io/lepton-conf/):
- [schematic.gui::third-button](https://github.com/lepton-eda/lepton-eda/wiki/Configuration-Settings#user-content-third-button)
- [schematic.gui::middle-button](https://github.com/lepton-eda/lepton-eda/wiki/Configuration-Settings#user-content-middle-button)

Thanks!

closed 2021-03-10 by graahnul-grom