lepton-eda/lepton-eda #419: Remove deprecated export functions and options

graahnul-grom, 2019-05-31:
Exporting with `scripts/image.scm` and `scripts/print.scm`
`lepton-schematic` scripts is deprecated since 2012 (git: 626a205).
We have enough tools to perform schematic export:
`File->Write Image...` dialog in `lepton-schematic`
and `lepton-cli export`.

The following should be deleted:
- `image-color` `gschemrc` option
- `image-size` `gschemrc` option
- `gschem-pdf` guile function (`g_funcs.c`)
- `gschem-image` guile function (`g_funcs.c`)
- `gschem-use-rc-values` guile function (`g_funcs.c`)
- `lepton-schematic`'s `--output` command line switch

Define `image-color` and `image-size` using `define-rc-dead-config`.
Define functions from `g_funcs.c` in the `(gschem deprecated)` module.
Make them do nothing, just print deprecation warning.

vzh, 2019-06-04:
In my view, it'd be very nice if we still had guile functions for exporting pdf and images available for using in Scheme scripts.

graahnul-grom, 2019-06-05:
Agree, it would be useful.
We could add such functions (later), e.g. something like `(schematic export)` module.
But current implementation is not satisfactory, and if you don't mind, I will remove it.

PR #432 (2019-06-09): Remove deprecated export scripts
Remove deprecated `image.scm` and `print.scm` export scripts,
delete related code from `liblepton` and `lepton-schematic`.
Deprecate `gschemrc` options: `image-size` and `image-color`.
Deprecate Scheme functions that used to be implemented in `lepton-schematic`:
`gschem-image`, `gschem-pdf`, `gschem-use-rc-values`.

Closes #419
closed 2019-06-09 by vzh