lepton-eda/lepton-eda #420: net-selection-mode does not distinguish between "enabled_net" and "enabled_all"

graahnul-grom, 2019-05-31:
`lepton-schematic` does not behave differently when `net-selection-mode`
`gschemrc` option is set to`"enabled_net"` or `"enabled_all"`.
Checks preformed in `lepton-schematic` code assumes that it's a boolean.
So, it's either enabled or disabled.
I have no idea about what `"enabled_all"` is supposed to do. When converting
settings to new configuration system, I propose to make this parameter a boolean,
thus reflecting the current state of affairs:
- `true`: mouse click on net selects it; the next click selects all connected segments
- `false`: subsequent clicks do not select connected segments

vzh, 2019-06-09:
`o_select_connected_nets()` can shed some light on this. As I understand the idea, `enabled_all` allowed selecting all nets with a netname of the net selected first: first click selects a net, second click selects all nets directly connected to it, and the third click selects not directly connected nets having the same netname as for the first group. Dunno, if it ever worked so, though I would like to have something like this implemented. It would be easier to do in Scheme, I think.

graahnul-grom, 2019-06-20:
Indeed, the third click selects all nets having equal "netname" attributes.
There is `w_current->net_selection_state` that is examined in `o_select_connected_nets()`.
@vzh Please check if it works on your system, then the issue could be closed.

vzh, 2019-06-20:
Ha! It's amazing why I couldn't reproduce it at home :-) I suggest to fix the description of the option in `options.scm` before closing the issue. OK?

graahnul-grom, 2019-06-20:
Yes, the description given there is absolutely inadequate.

PR #438 (2019-06-20): Better description for the net-selection-mode gschemrc option
More detailed description of the **net-selection-mode** `gschemrc` option.

Closes #420
closed 2019-06-20 by vzh