lepton-eda/lepton-eda #423: Migrate gschemrc settings to new configuration system

graahnul-grom, 2019-06-01:
Options currently defined in `conf/schematic/options.scm`
should be migrated to new configuration system.

- Define them using `define-rc-deprecated-config`
- Add new-style configuration settings:
  - Name keys after the corresponding `RC` function names
  - Place keys in the following groups:
    - `[schematic.gui]`
    - `[schematic.undo]`
    - new group `[schematic]`: all other keys (i.e. not sorted yet)
- Implement reading of new config in `i_vars.c` (`i_vars_set()`)
and refactor it later

In the process we can change names of groups and keys, default
values, types, decide where to move implementation, etc.
I think it's important to just start the migration now.
Converting keyboard shortcuts and color schemes is another issue.

Below I list option names and groups they should be put to,
along with the data types and default values (in parenthesis),
with few exceptions:
- `image-color` and `image-size` are not listed, they should be
removed as obsolete (#419)
- window-size should be removed, too (#422)
- `bus-ripper-symname` is defined in `liblepton`, but used
only in `lepton-schematic`. Either all related code should
be moved to the latter, or `bus-ripper-symname` option
transferred from `options.scm` to `system-gafrc`. Any
advice is welcome.

`[schematic.gui]`:
1. toolbars: bool (true)
2. scrollbars: bool (true)
3. handleboxes: bool (true)
4. draw-grips: bool (true)
5. zoom-with-pan: bool (true)
6. zoom-gain: int (20)
7. fast-mousepan: bool (false)
8. mousepan-gain: int (1)
9. keyboardpan-gain: int (20)
10. select-slack-pixels: int (10)
11. action-feedback-mode: string ("outline"), other values: "boundingbox"
12. continue-component-place: bool (true)
13. text-size: int (10)
14. snap-size: int (100)
15. text-caps-style: string ("both"), other values: "lower", "upper"
16. file-preview: bool (true)
17. enforce-hierarchy: bool (true)
18. middle-button: string ("mousepan"), other values: "popup", "action", "stroke", "repeat"
19. third-button: string ("popup"), other values: "mousepan"
20. third-button-cancel: bool (true)
21. scroll-wheel: string ("classic"), other values: "gtk"
22. scrollpan-steps: int (8)
23. warp-cursor: bool (false)
24. grid-mode: string ("mesh"), other values: "dots", "none"
25. dots-grid-mode: string ("variable"), other values: "fixed"
26. dots-grid-dot-size: int (1)
27. dots-grid-fixed-threshold: int (10)
28. mesh-grid-display-threshold: int (3)
29. force-boundingbox: bool (false)
30. net-direction-mode: bool (true)
31. net-selection-mode: bool (true) [see #420]
32. netconn-rubberband: bool (true)
33. magnetic-net-mode: bool (true)
34. embed-components: bool (false)

`[schematic.undo]`:
35. undo-control: bool (true)
36. undo-levels: int (20)
37. undo-type: string ("disk"), other values: "memory"
38. undo-panzoom: bool (false)

`[schematic]`:
39. auto-save-interval: int (120)
40. net-consolidate: bool (true)
41. logging: bool (true)
42. log-window: bool (false) (What about moving it to `[schematic.log-window]::show-on-startup`?)
43. bus-ripper-size: int (200)
44. bus-ripper-type: string ("component"), other values: "net"
45. bus-ripper-rotation "non-symmetric"

For `string` keys with only two possible values, I would
consider converting them to `bool` (if the parameter in
question is not going to be extended with new values).

Some implementation details were compiled on [this page](https://graahnul-grom.github.io/lepton-schematic-and-system-gschemrc.html), that may be of help.

graahnul-grom, 2019-06-10:
> 4. draw-grips: bool (true)

What this option is supposed to do? The description says that it
`Controls if the editing grips are drawn when selecting objects.`

However, setting it to "disabled" does not change how anything is drawn.
The only "effect" is that it disables ALL resize operations, so you cannot
change size of nets, buses, boxes, lines, etc.

vzh, 2019-06-12:
It should be other way around and seems to be a bug. It should prevent rendering of rectangular grips without disallowing any resize actions.

graahnul-grom, 2019-06-13:
The same with `geda-gaf` `1.8.2` and `1.9.2`, except that in `1.8.2` grips are not drawn.

graahnul-grom, 2019-06-15:
If it doesn't work, `draw-grips` option should be deleted.
It's safe, since nobody has ever used it (no bug reports, nor
questions on the mailing lists).
Later, when we'll fix the behavior (as I can see, it's far from trivial),
appropriate configuration parameter could be easily added.

graahnul-grom, 2019-06-18:
@vzh What about removing the `draw-grips` option (`define-rc-dead-config`)?

vzh, 2019-06-18:
I'm OK with this (never used it, too).

closed 2019-12-19 by vzh

graahnul-grom, 2019-12-21:
The [Configuration-Settings](https://github.com/graahnul-grom/lepton-eda/wiki/Configuration-Settings) document in my wiki has been updated with new options.
Please check it.

vzh, 2019-12-24:
@graahnul-grom,
I've read your document thoroughly, please see my notes below:
- The link to  [conf/schematic/options.scm](https://github.com/lepton-eda/lepton-eda/blob/master/schematic/scheme/conf/schematic/options.scm)  is no longer functional.
- PATH configuration context: instead of "contains settings for a directory", I would say  "contains per-project (per-directory) settings".
- Expression "(or directory specified by PATH)" is ambiguous since  the reader can think it is about the same-named env variable.
- `fast-mousepan` description is ambiguous a little, too. You  could say "text string bounding-box" there.
- There are some other minor issues, though they are inessential.
- I would split up the document into main page and sub-pages by sections and  groups combining them with the tables of deprecated or obsolete  settings. Otherwise it's hard to read it and find necessary  info.
- Please add the document to the main lepton wiki.

graahnul-grom, 2019-12-30:
@vzh Thank you! I've updated the document with your notes and uploaded it
to the main wiki. However, I have no idea how do I split it up in several pages.
IMHO, wiki (and especially the github one) is a nightmare. Even writing HTML
in text editor is much better.
To make it easier to navigate through the "Configuration Settings" page,
I've extended the table of contents, though.

vzh, 2020-01-05:
@graahnul-grom,
Thank YOU for your work! (And as I see, it's a lot of work.)
Github's wiki interface is **ugly**.  Especially when your links written in its web interface work differently compared to those done in the wiki git repository. (Probably, when Peter B. suggested github vs gitlab we did a wrong decision...)
Still I have managed to separate the last part of your document--Obsolete Settings.  I'm not sure, though it seems github considers all documents as living in one virtually plain (not hierarchical) directory, so links to them should have no directory prefix (or you have to include an absolute web path to make 'em work). I had to do `git push --force-with-lease` three or four times and reopen the last TOC links before things worked as expected.  Please check if the changes done are working for you, too (I'm still afraid they might not work :-))

graahnul-grom, 2020-01-05:
Yes, the *Obsolete Settings* page works, all links are OK, thank you.
What is really bad about gh wiki: it's useless side bar. It cannot be removed.
It just shows the list of links that is impossible to rearrange with stupid
"show more 1 page..." at the bottom. It takes almost 1/4 of the page width,
so we can forget about more or less wide tables and full size images.
And did you notice that gh/ms geniuses broke local links recently?
Now `#`-links do not work without the `user-content-` prefix in all browsers,
but firefox (maybe chromium and other "heavyweights", I don't know).

vzh, 2020-01-05:
It always looks like it is their success plan (not only ms's one): buy popular services, change them so others always spend more time on their new changes, make others just run after them instead of doing something usable.
The page width depends on your monitor :-)  While I have a pretty large monitor at home, my boss at work has a bigger one, I think. Netbook and smartphone users are suffering, yeah ;-) 
The issue with links smells horrible, anyway.

graahnul-grom, 2020-01-05:
I have no particular opinion on those sneaky "success plans", but for me it looks exactly the same as the road to hell should look like. Making assumption
that everyone in the world is rich and has a couple of 40'' monitors on the desk
is the last thing a least self-respecting developer would do. Set that aside, what
about, for example, tablet computers? They're everywhere.
BTW, that freaking sidebar is there for a long  time, long before ms bought them.

vzh, 2020-01-06:
I've just improved the wiki sidebar, I think, along with other changes.

graahnul-grom, 2020-01-07:
Thank you, it looks better now.

graahnul-grom, 2020-01-09:
@vzh Could you please remove the "To-Do list" page from our wiki?
It's embarrassing.
I wanted to discuss it earlier, but now I'm convinced that there's
nothing to discuss. Almost every phrase in it is a crying shame.
No technical person in his senses could write **and publish** such
a "document". A drunk housewife could have done it, that's for sure.
If you desperately need a formal TODO list, let's discuss and
write it together, no problem, we can open an issue for that.

vzh, 2020-01-10:
Done.  And please calm down.  The document was created by @peter-b in his **private** repository **before** it became `lepton-eda`.  I added nothing significant to it since then.  And no, I don't need a formal TODO list, it's enough for me to have the issue list.  (BTW, in the repository `lepton-eda` itself there are lots of to-do files lurking around).

vzh, 2020-01-10:
@graahnul-grom,
I've deleted two previous comments (yours and mine), because your message violated our [Code of Conduct](https://github.com/lepton-eda/lepton-eda/blob/master/CODE_OF_CONDUCT.md), was completely unacceptable to be posted in any lepton-eda communication channel, deeply insulting to the developers worked on gEDA and Lepton EDA, especially those who worked on the mentioned document, disrespectful, and rude.  Please read the section *Moderation* of the code of conduct for more information.  You've been warned.

I myself am sorry and apologize to all insulted people for being so inattentive in this particular case.

Dmitry, please do the same here, for devs and many users could read or are reading your message.

graahnul-grom, 2020-01-10:
@vzh Thank you very much!
Of course, I'm sorry if someone got offended.
In no way the message was intended to insult anyone, but to deliver
a trivial assertion: we shall not make a laughing stock of us by
publishing that kind of stuff as an official document (as it turned
out, that was some private scribbled notes, didn't know about that.
How on earth did they get to our wiki!? Unbelievable!).

vzh, 2020-01-10:
It's our (developer's) task to make things better (without offending anyone).  If you feel like something like you've posted here, please put this privately (to me, if you want), we'll discuss this then.

graahnul-grom, 2020-01-10:
OK!

Do you remember that crazy autotools library versioning pitfall (this formula:
`AX_LIBFOO[X:Y:Z] => libfoo.so.X-Z.Z.Y`)? Shouldn't we add this to the
"Release Engineering" page?

vzh, 2020-01-11:
Yes, I thought of this, too.

PR #527 (2019-12-19): Complete gschemrc settings migration
- move `bus-ripper-symname` to `system-gafrc` (it's `liblepton` option)
- remove options.scm

Together with #511, #518, #524 it closes #423.