lepton-eda/lepton-eda #440: utils/scripts/lepton-archive.py is incompatible with Python 3.x

danfe, 2019-07-01:
Its shebang line specifies `/usr/bin/env python` (versionless), which these days defaults to Python 3.x in most GNU and BSD distributions.  Consequently, the script does not work as is:

 - Because of `print Usage` (should be a function call)
 - Using deprecated `commands` module (removed in 3.x)
 - `raw_input() was renamed in `input()`

There might be more, but fixing these allows the script at least to start with Python 3.6 (which is now default Python version on FreeBSD).

vzh, 2019-07-01:
Well, the script is pretty obsolete and I wanted to rewrite it in Scheme to take the opportunity of using our new functions instead of parsing sch files as the users/devs did for many years.
However (since I'm not sure if the work I started would be ever finished ;-) taking into account how many other things we have to do), if you know how to fix that, any fixes are welcome and will do into the next release.

danfe, 2019-07-01:
I'm looping @graahnul-grom in since he's the maintainer of the FreeBSD port, had also reported some problems with Guile 2.2, and seems to be quite active here.  (This does not mean I'm trying to walk away from the issue, and would gladly provide any help, tests, and patches.)

graahnul-grom, 2019-07-02:
@danfe I've [fixed](https://github.com/graahnul-grom/freebsd-lepton-eda/compare/guile22) the FreeBSD port: now it uses `lang/python27` and works with guile 2.2.
However, today I have no access to [bugzilla](https://bugs.freebsd.org/bugzilla/) site,
it reports "Error 503 Backend fetch failed" error, so I'm attaching the patch here.

[lepton-eda.diff.txt](https://github.com/lepton-eda/lepton-eda/files/3349355/lepton-eda.diff.txt)

danfe, 2019-07-02:
Yes, some FreeBSD internal services are currently broken (not just Bugzilla, svnweb as well).  It's OK, I'll handle the patch from here.

danfe, 2019-07-02:
Committed as [r505685](https://svnweb.freebsd.org/ports?view=revision&revision=505685), thank you!

graahnul-grom, 2019-07-02:
@danfe Thank you!

graahnul-grom, 2019-07-09:
Could we move `lepton-archive` and `lepton-tragesym` to `contrib/`,
eliminating dependency on python?

vzh, 2019-07-14:
We could, however I would like to have the like functionality anyway amongst our tools.  Besides, the dependency on python is not strong, the same as on, say, perl. Maintainers of distribution packages could drop it or make it something like "recommended" or "suggested" (in terms of Debian).

danfe, 2019-07-14:
Or, perhaps, if you wanted to rewrite it in Scheme to take the opportunity of the new functions instead of parsing `*.sch` files, it's the best time to do so. :-)

vzh, 2019-09-26:
@danfe, Is the original issue solved?  I want to close it.

graahnul-grom, 2019-09-26:
The issue with python pseudo-programming language still persists.
Though I made the port depends on version 2.7, there are another questions.
py 2.7 announced eol. distro makers follows.
there are [bug reports](https://bugs.debian.org/cgi-bin/pkgreport.cgi?src=lepton-eda) about that.
Comrade Ahmed comes to the rescue again - here: https://bugs.launchpad.net/geda/+bug/1842509.  It means switching to p3.
I have to repeat that I think that all this stuff should be in a separated project (at least, for now).

danfe, 2019-09-30:
I agree.  Even though _technically_ it was "fixed" for the moment by hardwiring to Python 2.x, it should be either made Python 3.x friendly (because 2.7 is being phased out), or rewritten in Scheme as mentioned earlier.

graahnul-grom, 2019-10-04:
`lepton-archive` doesn't work, anyway, since 2007 (git: 9dde023) when
`system-gafrc` was refactored by removing `component-library` calls
from it. `lepton-archive` searches that file for "component-library" strings,
so it cannot find and save any stock symbols.
No bug reports were filed about that (AFAICT), it seems that nobody uses it.
@vzh I agree that we need a tools with similar functionality in the suite, but
`lepton-archive` in it's current state should be moved to `contrib`.
What do you think?

vzh, 2019-10-08:
Hi friends! Please test and comment on #468. It's probably not feature-complete just now. Just tell me if it could be a decent replacement for (probably non-working) lepton-archive.py. Thanks.

closed 2019-10-11 by vzh