lepton-eda/lepton-eda #446: Homebrew on OSX: error: GIO 2.38.0 or later is required

craignied, 2019-07-15:
Compiling with Homebrew on OSX:

checking for gio-2.0 >= 2.38.0... no
configure: error: GIO 2.38.0 or later is required.

GIO isn't a module that I can find on Homebrew. Is it part of something else?

Thanks in advance

graahnul-grom, 2019-07-15:
@craignied Thank you for reporting this. `GIO` is a part of `glib`. We recently
bumped minimum required version of `glib` to 2.38.0. Do you have the `glib`
package installed?

craignied, 2019-07-15:
I have glib 2.60.5 installed:

brew info glib
glib: stable 2.60.5 (bottled)
Core application library for C
https://developer.gnome.org/glib/
/usr/local/Cellar/glib/2.60.5 (429 files, 15.2MB) *
  Poured from bottle on 2019-07-13 at 20:19:37
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/glib.rb
==> Dependencies
Build: meson ✘, ninja ✘, pkg-config ✔
Required: gettext ✔, libffi ✔, pcre ✔, python ✔
==> Analytics
install: 280,229 (30 days), 837,113 (90 days), 2,289,367 (365 days)
install_on_request: 15,309 (30 days), 42,956 (90 days), 105,085 (365 days)
build_error: 0 (30 days)

graahnul-grom, 2019-07-15:
Could you please try the following command:
```
 $ pkg-config --libs gio-2.0
```

craignied, 2019-07-15:
Still getting

checking for gio-2.0 >= 2.38.0... no
configure: error: GIO 2.38.0 or later is required.

graahnul-grom, 2019-07-15:
What is exactly the `./configure` command line you are using?
It would be helpful if you attach the full output of `./configure`
and the generated `config.log` file.
You could also try to disable the `GIO` check (just comment out two lines in `configure.ac`):
```diff
diff --git a/configure.ac b/configure.ac
index 9923799ea..40fd51ed1 100644
--- a/configure.ac
+++ b/configure.ac
@@ -79,8 +79,8 @@ PKG_CHECK_MODULES(GLIB, [glib-2.0 >= 2.38.0], ,
   AC_MSG_ERROR([GLib 2.38.0 or later is required.]))
 AC_DEFINE([G_DISABLE_DEPRECATED], [1], [Disable deprecated GLib features])
 
-PKG_CHECK_MODULES(GIO, [gio-2.0 >= 2.38.0], ,
-  AC_MSG_ERROR([GIO 2.38.0 or later is required.]))
+#PKG_CHECK_MODULES(GIO, [gio-2.0 >= 2.38.0], ,
+#  AC_MSG_ERROR([GIO 2.38.0 or later is required.]))
 
 PKG_CHECK_MODULES(GTK, [gtk+-2.0 >= 2.24.0], ,
   AC_MSG_ERROR([GTK+ 2.24.0 or later is required.]))
```

[gio_off.diff.txt](https://github.com/lepton-eda/lepton-eda/files/3393830/gio_off.diff.txt)

craignied, 2019-07-15:
`./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... build-tools/install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for git... git
checking if the source directory is a git repository... no
checking build system type... x86_64-apple-darwin18.6.0
checking host system type... x86_64-apple-darwin18.6.0
checking for Linux host... no
checking for Windows host... no
checking for Cygwin host... no
checking for Mac OS X Carbon host... yes
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking how to run the C preprocessor... gcc -E
checking for ar... ar
checking the archiver (ar) interface... ar
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
checking if the linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 196608
[... 66 lines cut ...]
checking for msgmerge... /usr/local/bin/msgmerge
checking for ld used by GCC... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
checking if the linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) is GNU ld... no
checking for shared library run path origin... done
checking for CFPreferencesCopyAppValue... yes
checking for CFLocaleCopyCurrent... yes
checking for GNU gettext in libc... no
checking for iconv... yes
checking for working iconv... yes
checking how to link with libiconv... -liconv
checking for GNU gettext in libintl... yes
checking whether to use NLS... yes
checking where the gettext function comes from... external libintl
checking how to link with libintl... -L/usr/local/lib -lintl -Wl,-framework -Wl,CoreFoundation
checking for gettext... /usr/local/bin/gettext
checking for pkg-config... /usr/local/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for guile-2.2 >= 2.0.0... yes
checking for guile... guile
checking for guile-snarf... guile-snarf
checking whether scm_display_error accepts a stack... yes
checking whether scm_display_error accepts a frame... yes
checking for glib-2.0 >= 2.38.0... yes
checking for gio-2.0 >= 2.38.0... no
configure: error: GIO 2.38.0 or later is required.`

craignied, 2019-07-15:
./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... build-tools/install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for git... git
checking if the source directory is a git repository... no
checking build system type... x86_64-apple-darwin18.6.0
checking host system type... x86_64-apple-darwin18.6.0
checking for Linux host... no
checking for Windows host... no
checking for Cygwin host... no
checking for Mac OS X Carbon host... yes
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking how to run the C preprocessor... gcc -E
checking for ar... ar
checking the archiver (ar) interface... ar
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
checking if the linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 196608
[... 66 lines cut ...]
checking for msgmerge... /usr/local/bin/msgmerge
checking for ld used by GCC... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
checking if the linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) is GNU ld... no
checking for shared library run path origin... done
checking for CFPreferencesCopyAppValue... yes
checking for CFLocaleCopyCurrent... yes
checking for GNU gettext in libc... no
checking for iconv... yes
checking for working iconv... yes
checking how to link with libiconv... -liconv
checking for GNU gettext in libintl... yes
checking whether to use NLS... yes
checking where the gettext function comes from... external libintl
checking how to link with libintl... -L/usr/local/lib -lintl -Wl,-framework -Wl,CoreFoundation
checking for gettext... /usr/local/bin/gettext
checking for pkg-config... /usr/local/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for guile-2.2 >= 2.0.0... yes
checking for guile... guile
checking for guile-snarf... guile-snarf
checking whether scm_display_error accepts a stack... yes
checking whether scm_display_error accepts a frame... yes
checking for glib-2.0 >= 2.38.0... yes
checking for gio-2.0 >= 2.38.0... no
configure: error: GIO 2.38.0 or later is required.

craignied, 2019-07-15:
Commented out gio, and now it's not finding gtk:
...
checking for glib-2.0 >= 2.38.0... yes
checking for gtk+-2.0 >= 2.24.0... no
configure: error: GTK+ 2.24.0 or later is required.

brew info gtk
gtk+: stable 2.24.32 (bottled), HEAD
GUI toolkit
https://gtk.org/
/usr/local/Cellar/gtk+/2.24.32_3 (1,194 files, 59.9MB) *
  Poured from bottle on 2019-07-13 at 20:21:43
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/gtk+.rb
==> Dependencies
Build: gobject-introspection ✘, pkg-config ✔
Required: atk ✔, gdk-pixbuf ✔, hicolor-icon-theme ✔, pango ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 18,049 (30 days), 31,289 (90 days), 119,651 (365 days)
install_on_request: 3,799 (30 days), 7,029 (90 days), 24,049 (365 days)
build_error: 0 (30 days)

graahnul-grom, 2019-07-15:
Let's start from scratch:
1) `cd` to `lepton-eda` source directory
2) `git clean -x -f -d`
3) comment out `GIO` check in `configure.ac`
4) `./autogen.sh`
5) `./configure`

If it fails, please attach the `config.log` file.

craignied, 2019-07-15:
Scratch is the same as above 😊

[config.log](https://github.com/lepton-eda/lepton-eda/files/3393973/config.log)

graahnul-grom, 2019-07-15:
:) Please install `libffi` and try again.

graahnul-grom, 2019-07-15:
@craignied I found [this thread](https://github.com/otrv4/pidgin-otrng/issues/104#issuecomment-511144582) where a similar problem with `GTK` (actually, with`libffi`) is discussed.
Please check the `brew info libffi` output. You may need to set `$PKG_CONFIG_PATH` environment variable.

graahnul-grom, 2019-07-15:
In the issue mentioned above they got exactly the same message I see in the `config.log`:
```
Package 'libffi', required by 'gobject-2.0', not found
```

craignied, 2019-07-15:
We've moved forward! I had libffi installed, but I set:

export LDFLAGS="-L/usr/local/opt/libffi/lib"
export PKG_CONFIG_PATH="/usr/local/opt/libffi/lib/pkgconfig"

./configure
Threw this error:

configure: error: The update-desktop-database tool could not be found. Ensure it is
installed and in your path, or configure with
--disable-update-xdg-database.

I poked around and couldn't figure out how to install that, so I tried
./configure --disable-update-xdg-database

And it made a Makefile! When I issued a make, it did a bunch of things but finally:
...
Making all in scheme-api
Updating ./version.texi
  MAKEINFO lepton-scheme.info
lepton-scheme.texi:5: warning: unrecognized encoding name `utf-8'.
  MAKEINFO lepton-scheme.html
makeinfo: unrecognized option `--css-ref=lepton-scheme.css'
Try `makeinfo --help' for more information.
make[3]: *** [lepton-scheme.html] Error 1
make[2]: *** [all-recursive] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

Ideas?

[config.log](https://github.com/lepton-eda/lepton-eda/files/3394456/config.log)

vzh, 2019-07-16:
I've just googled for **makeinfo unrecognized encoding name**. The very first link is [this](https://github.com/xiaohanyu/oh-my-emacs/issues/85). Could it be of any help? Especially the advice to use `brew link texinfo --force`?

graahnul-grom, 2019-07-16:
> configure: error: The update-desktop-database tool could not be found. Ensure it is
installed and in your path, or configure with --disable-update-xdg-database

`brew install desktop-file-utils`

craignied, 2019-07-16:
@vzh thanks, I had done that previously.
@graahnul-grom thanks super, it now compiles completely. I'm still getting errors with make, though. I've attached the output of make. Could the commented out GIO be the problem?
[makelog.txt](https://github.com/lepton-eda/lepton-eda/files/3396996/makelog.txt)

craignied, 2019-07-16:
I figured out how to get rid of all the Warning: Failed to set locale category LC_...s with
export LC_ALL=en_US.UTF-8

So this is a little cleaner make log
[make.log](https://github.com/lepton-eda/lepton-eda/files/3397362/make.log)

Essentially it's failing with:
```
Making all in scheme-api
  MAKEINFO lepton-scheme.html
makeinfo: unrecognized option `--css-ref=lepton-scheme.css'
```
Thoughts?

vzh, 2019-07-16:
Googling it a bit, I found that the option `--css-ref` is available in `makeinfo` at least since version 4.13. First you have to make sure your `makeinfo` supports the option.  If it's outdated, you need to update it.

graahnul-grom, 2019-07-16:
@craignied You're almost there. Don't let the issue with `makeinfo` get in the way:
please revert [this commit](https://github.com/lepton-eda/lepton-eda/commit/381c17be949077ce8301427eaa5079bab646e003).
```diff
diff --git a/docs/scheme-api/Makefile.am b/docs/scheme-api/Makefile.am
index 39a1f679a..51e504d08 100644
--- a/docs/scheme-api/Makefile.am
+++ b/docs/scheme-api/Makefile.am
@@ -1,13 +1,2 @@
 info_TEXINFOS = lepton-scheme.texi
 
-AM_MAKEINFOHTMLFLAGS = --css-ref=lepton-scheme.css
-
-EXTRA_DIST = lepton-scheme.css
-
-html-local:
-   $(MKDIR_P) $(builddir)/lepton-scheme.html/
-   cp -fv $(srcdir)/lepton-scheme.css $(builddir)/lepton-scheme.html/
-
-all-local: html
-
-install-data-local: install-html
```

craignied, 2019-07-17:
Thanks @graahnul-grom!

Do you know why this might be happening:

> git revert 381c17b
> error: could not revert 381c17be9... docs/scheme-api: Add CSS file for HTML output.
> hint: after resolving the conflicts, mark the corrected paths
> hint: with 'git add <paths>' or 'git rm <paths>'
> hint: and commit the result with 'git commit'

graahnul-grom, 2019-07-17:
Just delete all, but the first line ("info_TEXINFOS = lepton-scheme.texi") from
`docs/scheme-api/Makefile.am`, then do `git commit`. The commit message
will look like `Revert "docs/scheme-api: Add CSS file for HTML output."`

craignied, 2019-07-17:
It did make without error,  @graahnul-grom! Woot! OK, big question: coming from geda-gaf, is there a place to learn about functionality? Like, are the names of the commands different, and are the commands themselves different?

Many thanks for your help!

craignied, 2019-07-17:
I have my first bug report! I was trying to open an old schematic, when it crashed with:
Assertion failed: (s->stack->next != NULL), function CGGStackRestore, file /BuildRoot/Library/Caches/com.apple.xbs/Sources/CoreGraphics/CoreGraphics-1260.2/CoreGraphics/Context/CGGStack.c, line 77.
[1]    24347 abort      lepton-schematic

graahnul-grom, 2019-07-20:
@craignied Thank you, congratulations on getting through the configure and make.
Could this bug be related to already reported crash in #76?
What version of mac os you are using?

vzh, 2019-07-20:
@craignied, could you please answer a couple of questions:
- are you able to work on new schematics with `lepton-schematic` at all?
- could you provide us with the schematic that triggers the crash, even with some restricted variant, to let the members of the project who uses macs to check what's wrong.

And, since that seems to be another issue (is it? do the tools work on your system at all?), please open another issue to let the people interested in it to concentrate on the topic. Then we could close this issue.

BTW, if you found a good recipe of how to make lepton on mac using brew, please post it here. We could add it to our wiki then to help other users.

Thanks in advance.

graahnul-grom, 2019-07-21:
Yes, since this issue is about *building* lepton on OSX, it's better to move the run-time
problem to a new issue. However, we didn't address the `GIO` and `makeinfo`
issues yet.

graahnul-grom, 2019-07-21:
`GIO` check was added to `configure.ac` in commit 1b78564 in response to
launchpad.net bug report [1258217: gschem >1.7.2 won't build in Cygwin](https://bugs.launchpad.net/geda/+bug/1258217).
In the comment #5 there it was suggested that simply adding `$(GIO_LIBS)` to
`liblepton/src/Makefile.am` is enough.
I'll try to check it with `cygwin`.

graahnul-grom, 2019-07-21:
I've managed to build `lepton-eda` on `cygwin` (`windows 7` x64 in `virtualbox`,
`guile 2.0`) both with and without `GIO` check.

graahnul-grom, 2019-07-21:
According to various web search results, the run-time bug is related to `cairo`
rendering code.
@craignied Could you please try to export some schematics to `png` and `svg`
formats using `lepton-cli export`? Will it crash, to? It would be helpful if you can
obtain a back-trace (there should be crash log files in
`~/Library/Logs/DiagnosticReports/` on `OSX`, IIUC).