lepton-eda/lepton-eda #454: lepton-netlist -h fails

tomdean1939, 2019-09-25:
Ubuntu 18.04
up-to-date sources
configure, compile and install with no errors
Anything I try with lepton-netlist

> lepton-netlist -h
> lepton-netlist -h
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/rc.scm
;;; WARNING: compilation of /usr/local/share/lepton-eda/scheme/lepton/rc.scm failed:
;;; Wrong type to apply: #<unknown-type (0x24f . 0x7f6a8782e348) @ 0x7f6a87aed7f0>
;;; compiling /usr/local/share/lepton-eda/scheme/netlist.scm
;;; WARNING: compilation of /usr/local/share/lepton-eda/scheme/netlist.scm failed:
;;; no binding `user-config-dir' in module (geda os)
Backtrace:
In ice-9/boot-9.scm:
   222:29 19 (map1 (((netlist option)) ((netlist #)) ((netlist #)) ?))
   222:29 18 (map1 (((netlist schematic-component)) ((netlist #)) # ?))
   222:29 17 (map1 (((netlist sort)) ((netlist attrib compare)) (#) ?))
   222:29 16 (map1 (((netlist attrib compare)) ((geda page)) ((?)) ?))
   222:29 15 (map1 (((geda page)) ((geda deprecated)) ((geda log)) ?))
   222:29 14 (map1 (((geda deprecated)) ((geda log)) ((geda repl)) ?))
   222:29 13 (map1 (((geda log)) ((geda repl)) ((lepton library)) # ?))
   222:29 12 (map1 (((geda repl)) ((lepton library)) ((lepton #)) # ?))
   222:29 11 (map1 (((lepton library)) ((lepton version)) ((# # ?)) ?))
   222:29 10 (map1 (((lepton version)) ((netlist core gettext)) (#) ?))
   222:29  9 (map1 (((netlist core gettext)) ((netlist config)) (#) ?))
   222:17  8 (map1 (((netlist config)) ((netlist deprecated)) ((?)) ?))
  2788:17  7 (resolve-interface (netlist config) #:select _ #:hide _ ?)
  2714:10  6 (_ (netlist config) _ _ #:ensure _)
  2982:16  5 (try-module-autoload _ _)
   2312:4  4 (save-module-excursion #<procedure 558c95e58f60 at ice-?>)
  3002:22  3 (_)
In unknown file:
           2 (primitive-load-path "netlist/config" #<procedure 558c9?>)
In netlist/config.scm:
    32:11  1 (_)
In unknown file:
           0 (_)

ERROR: Wrong type to apply: #<unknown-type (0x24f . 0x7f6a8782e348) @ 0x7f6a87aee4b0>

vzh, 2019-09-25:
Please show us the output of the commands as follows:
```
guile --version
apt-cache policy guile-2.0
apt-cache policy guile-2.0-dev
apt-cache policy guile-2.2
apt-cache policy guile-2.2-dev
```

As a rule of thumb, you have to have the same versions of guile binary and its development version installed.  After it is setup correctly, please remove your guile cached precompiled *go* files, that is, do `rm -rf ~/.cache/guile/ccache/`.

tomdean1939, 2019-09-25:
p9x79> rm -Rf ~/.cache/guile/ccache/
p9x79> guile --version
guile (GNU Guile) 2.2.3
Packaged by Debian (2.2.3-deb+1-3ubuntu0.1)
Copyright (C) 2017 Free Software Foundation, Inc.

License LGPLv3+: GNU LGPL 3 or later <http://gnu.org/licenses/lgpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
p9x79> apt-cache policy guile-2.0
guile-2.0:
  Installed: 2.0.13+1-5ubuntu0.1
  Candidate: 2.0.13+1-5ubuntu0.1
  Version table:
 *** 2.0.13+1-5ubuntu0.1 500
        500 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     2.0.13+1-5build2 500
        500 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 Packages
p9x79> apt-cache policy guile-2.0-dev
guile-2.0-dev:
  Installed: 2.0.13+1-5ubuntu0.1
  Candidate: 2.0.13+1-5ubuntu0.1
  Version table:
 *** 2.0.13+1-5ubuntu0.1 500
        500 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     2.0.13+1-5build2 500
        500 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 Packages
p9x79> apt-cache policy guile-2.2
guile-2.2:
  Installed: 2.2.3+1-3ubuntu0.1
  Candidate: 2.2.3+1-3ubuntu0.1
  Version table:
 *** 2.2.3+1-3ubuntu0.1 500
        500 http://us.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages
        100 /var/lib/dpkg/status
     2.2.3+1-3build1 500
        500 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
p9x79> apt-cache policy guile-2.2-dev
guile-2.2-dev:
  Installed: (none)
  Candidate: 2.2.3+1-3ubuntu0.1
  Version table:
     2.2.3+1-3ubuntu0.1 500
        500 http://us.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages
     2.2.3+1-3build1 500
        500 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
p9x79> lepton-netlist *.sch -o xx
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/local/bin/lepton-netlist
;;; /usr/local/bin/lepton-netlist:60:0: warning: possibly unbound variable `init-netlist-options!'
;;; /usr/local/bin/lepton-netlist:65:10: warning: possibly unbound variable `netlist-option-ref/toplevel'
;;; /usr/local/bin/lepton-netlist:80:0: warning: possibly unbound variable `%with-toplevel'
;;; /usr/local/bin/lepton-netlist:80:16: warning: possibly unbound variable `%make-toplevel'
;;; /usr/local/bin/lepton-netlist:84:4: warning: possibly unbound variable `init-log'
;;; /usr/local/bin/lepton-netlist:85:4: warning: possibly unbound variable `lepton-netlist-version'
;;; /usr/local/bin/lepton-netlist:86:4: warning: possibly unbound variable `main'
;;; compiled /home/tomdean/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/bin/lepton-netlist.go
;;; compiling /usr/local/share/lepton-eda/scheme/netlist/option.scm
;;; compiled /home/tomdean/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/netlist/option.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/geda/deprecated.scm
;;; compiling /usr/local/share/lepton-eda/scheme/geda/core/gettext.scm
;;; compiled /home/tomdean/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/geda/core/gettext.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/geda/page.scm
;;; compiled /home/tomdean/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/geda/page.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/geda/object.scm
;;; compiled /home/tomdean/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/geda/object.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/geda/attrib.scm
[... 36 lines cut ...]
   222:29 19 (map1 _)
   222:29 18 (map1 _)
   222:29 17 (map1 _)
   222:29 16 (map1 _)
   222:29 15 (map1 _)
   222:29 14 (map1 _)
   222:29 13 (map1 _)
   222:29 12 (map1 _)
   222:29 11 (map1 _)
   222:29 10 (map1 _)
   222:29  9 (map1 _)
   222:17  8 (map1 (((netlist config)) ((netlist deprecated)) ((?)) ?))
  2788:17  7 (resolve-interface (netlist config) #:select _ #:hide _ ?)
  2714:10  6 (_ (netlist config) _ _ #:ensure _)
  2982:16  5 (try-module-autoload _ _)
   2312:4  4 (save-module-excursion _)
  3002:22  3 (_)
In unknown file:
           2 (primitive-load-path "netlist/config" #<procedure 5588e?>)
In netlist/config.scm:
    32:11  1 (_)
In unknown file:
           0 (_)

ERROR: Wrong type to apply: #<unknown-type (0x24f . 0x7f8f82bc6348) @ 0x7f8f82e874b0>

vzh, 2019-09-26:
So, as you can see, your working `guile` version is 2.2. However, you've installed `guile-2.0-dev` and that devel package is used to build the underlying C code. guile 2.0 and guile 2.2 are different enough and not compatible.  Please do as follows:
- uninstall the `guile-2.0-dev` package
- install `guile-2.2-dev` instead
- remove the guile cache as you did before
- recompile and install lepton
- run it and please report if it works :-)

(Another way would be to make guile-2.0 your main guile, though this is a different story...)

tomdean1939, 2019-09-26:
This solved the problem.

I think guile-2.0-dev was installed for some other reason???

Is it possible for configure to catch this?

Thanks,
Tom Dean

vzh, 2019-09-26:
>  I think guile-2.0-dev was installed for some other reason??? 
Is it possible for configure to catch this?

`-dev` packages are usually installed by the users who want to build something themselves.   They contain C headers (*.h* files) and such.  Probably, you've installed it to build some program dependent on guile (`geda-gaf`?). Anything would work if your guile binary had the same version (Debian derived distributions have `update-alternatives` to opt a more appropriate one).  The user may want to have two guile versions installed, so probably the installation and deinstallation is the most sane way to deal with the `-dev` packages when compilation is needed.  In theory, `configure` could check the headers installed, though we need a volunteer who could investigate it :-)

closed 2019-09-26 by vzh