lepton-eda/lepton-eda #463: Configuration Can Not Check -dev or -devel Packages

tomdean1939, 2019-09-30:
Configuration can not determine if the corresponding guile-2.0 or guile-2.2 -dev[el...]  is installed.
Configuration fails on debian-based systems if the guile-2.2 and the guile-2.0-dev packages are installed.  The packages must be coordinated, -2.0 and -2.0-dev, -2.2 and -2.2-dev, etc.

The attached patch will work with debian-based systems, but is not portable to others such as redhat based systems where the associated development package seems to be -devel (?).  What about FreeBSD, NetBSD or windows systems?

[geta-guile.txt](https://github.com/lepton-eda/lepton-eda/files/3672812/geta-guile.txt)

graahnul-grom, 2019-09-30:
On FreeBSD we have no `-devel` packages. Guile package, for example,
is named just `guile2` and contains all the files needed for development.
IMHO, package separation is largely thing of the past and it leads to user
confusion, additional amount of maintainers' work and explosive growth
in number of packages.

bdale, 2019-10-01:
tomdean1939 <notifications@github.com> writes:

> Configuration fails on debian-based systems if the guile-2.2 and the
> guile-2.0-dev packages are installed.  The packages must be
> coordinated, -2.0 and -2.0-dev, -2.2 and -2.2-dev, etc.

FWIW, I handle this in the official package builds for Debian by a
build-depend on guile-2.2-dev, and delivering a run-time depend on
guile-2.2. 

Bdale

vzh, 2020-01-08:
Friends, I propose to close this issue.  It can be reopened if/when a portable solution will be found.

tomdean1939, 2020-01-08:
I think my install is OK.  This is still an issue, but, a solution is 
not evicent.

Please close it.

Tom Dean

vzh, 2020-01-09:
Thank you, Tom!

vzh, 2020-01-10:
We could ease life of our users by just improving the following line in `geda-guile.m4`:
`AC_MSG_ERROR([you need at least version ${GUILE_MIN_VER} of guile])`
to point them to the direction where the issue might be.  E.g., we could add something like: *Please check if appropriate development package (-dev or -devel), if any, with headers for this version is installed on your system.*
WDYT?

graahnul-grom, 2020-01-11:
I think it would be helpful, while there's no bulletproof solution.
wrt `guile-config`: I've checked most of the packages at https://pkgs.org/download/guile and indeed, it's present only in development packages, AFAICS.

vzh, 2020-01-13:
@graahnul-grom,
So you suggest to check for `guile-config --version` in the m4 file, somehow parse its output and compare with found guile's effective version?
Probably a better solution exists. [Here](https://www.gnu.org/software/guile/manual/html_node/Autoconf-Macros.html) I found that `GUILE_PKG` can be used to search for development packages. And [here](https://github.com/aconchillo/guile-json/blob/master/m4/guile.m4) there is a real example of how it works.  So we just need somebody who could accommodate it to our code :-)

graahnul-grom, 2020-01-13:
Thank you for the link, haven't seen that before.
The file in the second link is not an example, it's a copy of stock `guile.m4`.
I've added the following lines to `m4/geda-guile.m4`:
```m4
GUILE_FLAGS
GUILE_PROGS
GUILE_PKG([2.2 2.0])
```
Then (on ubuntu 19.04 "disco") all guile packages were removed. After that:
```sh
$ sudo apt-get install guile-2.2 guile-2.0-dev
```
And `configure` correctly reported `-dev` package mismatch.

@vzh @tomdean1939 Could you please test this on your systems?

[GUILE_PKG.diff.txt](https://github.com/lepton-eda/lepton-eda/files/4054812/GUILE_PKG.diff.txt)

tomdean1939, 2020-01-14:
On 2020-01-13 09:27, dmn wrote:
> sudo apt-get install guile-2.2 guile-2.0-dev

On Ubuntu 18.04, apt-get wants to match guile-2.0 with guile-2.0-dev.

 > sudo apt-get install guile-2.2 guile-2.0-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
guile-2.2 is already the newest version (2.2.3+1-3ubuntu0.1).
Suggested packages:
   guile-2.0-doc
The following packages will be REMOVED:
   guile-2.2-dev
The following NEW packages will be installed:
   guile-2.0 guile-2.0-dev
0 upgraded, 2 newly installed, 1 to remove and 4 not upgraded.
Need to get 693 kB of archives.
After this operation, 1905 kB of additional disk space will be used.
Do you want to continue? [Y/n] n

Tom Dean

graahnul-grom, 2020-01-14:
Yes. After that you will have guile 2.2 and the development
package for 2.0 installed. We want to check whether
`configure` will complain about versions mismatch.

BTW: unfortunately, I'm unable to apply your patch,
`git apply` says "error: unrecognized input".

vzh, 2020-01-17:
@graahnul-grom,
Just tried it, your patch works OK, thank you! Please make a PR.

graahnul-grom, 2020-01-19:
Now I know why people include a copy of `guile.m4` in their projects. :-)
If guile (one of the dev packages, checked it on an old `altlinux` distro) is
not installed, `GUILE_PKG` and other macros won't work.
Copying `share/aclocal/guile.m4` (from the 2.2.6 FreeBSD package) to our
`m4/` folder solves it.
Is it OK to distribute that file with lepton sources?

vzh, 2020-01-20:
Why not? I think, OK.

PR #549 (2020-01-21): Improve detection of Guile packages on configure stage
- add `guile.m4` file from Guile 2.2.6 distribution
- use `guile.m4` macros `GUILE_FLAGS`, `GUILE_PROGS`, `GUILE_PKG`
to detect missing and mismatched Guile packages

Closes #463
closed 2020-01-21 by vzh