lepton-eda/lepton-eda #476: system-gschemrc problem with 1.99

noqsi, 2019-10-13:
Installed 1.99 on an Ubuntu 16 VM on my Mac. Launching lepton-schematic, I get a panel with:

> Failed to load RC file [/usr/local/share/lepton-eda/system-gschemrc]: ERROR: Configuration does not have group 'schematic.gui'

The schematic window has only *File* and *Edit* in the menu bar.

The RC file exists, so I don't understand the reason it failed to load.

graahnul-grom, 2019-10-13:
Hi John!
Please make sure that `share/lepton-eda/geda-system.conf` is installed:
```ini
[schematic.gui]
use-docks=false
use-tabs=true
<etc>
```
The error message comes from this file:
`share/lepton-eda/scheme/conf/schematic/menu.scm`.
It checks for the `[schematic.gui]::use-docks` and `[schematic.gui]::use-tabs` configuration keys while constructing the main menu.

noqsi, 2019-10-13:
Good hint.

The problem was an empty /etc/xdg/lepton-eda/geda-system.conf

How to avoid tripping over this?

graahnul-grom, 2019-10-13:
I would not put `geda-system.conf` into `/etc/xdg/`, but leave it in its default location, then override settings in `$HOME/.config/lepton-eda/geda-user.conf`, if necessary.
BTW, we did not have the `geda-system.conf` file before `1.9.8`.
Probably, I should also amend `menu.scm` to make it work even without config file.

noqsi, 2019-10-13:
The essential problem is that I don't know how it got into /etc/xdg/ in the first place. I don't even know what /etc/xdg/ is for. Maybe installation needs a "clean up configurations" step.

graahnul-grom, 2019-10-13:
lepton-eda doesn't install anything to `/etc`, and never did.
Dealing with old/new configuration files on upgrades is a package
manager's task on many systems.
Did you install 1.9.9 from source this time?

noqsi, 2019-10-13:
I installed from the release source.

graahnul-grom, 2019-10-13:
Maybe that file in /etc/xdg is a leftover from old times.
Could we close this issue?
And what do you think about including your `gnet-spice-noqsi` backend
into lepton-eda? Does it work with 1.9.9 release?

noqsi, 2019-10-14:
I'll close this.

There's a lepton branch in the [gnet-spice-noqsi](https://github.com/noqsi/gnet-spice-noqsi) repo that seems to be working with 1.9.9 after some tinkering today. Could use some cleaning up..

closed 2019-10-14 by noqsi

graahnul-grom, 2019-10-14:
@noqsi It works if the following expression in `spice-noqsi()` is commented out:
```scheme
(set-current-output-port(open-output-file filename))
```

noqsi, 2019-10-14:
It's been working for me with that line uncommented. I suppose, however, that it's useless.