lepton-eda/lepton-eda #477: SPICE cleverness

noqsi, 2019-10-14:
It appears that if a netlist back end has "spice" in its name, the behavior of `(schematic-package-names)` changes. Slotted components get split into multiple packages. This causes *gnet-spice-noqsi* trouble, since it works by package, not slot.

graahnul-grom, 2019-10-14:
Short answer:
```scheme
( set! names
    ( map
        ( lambda( pkg ) (package-attribute pkg 'refdes) )
        ( schematic-packages (toplevel-schematic) )
    )
)

( set! unique-names (delete-duplicates names string-ci=?) )
```

noqsi, 2019-10-14:
I don't think this will work. The cleverness afflicts things like `(get-nets)`: referring to a package by refdes yields 

> Invalid refdes ('U8') and pin ('unknown') passed to get-nets

noqsi, 2019-10-14:
```
$ lepton-netlist -i -g spice *.sch
Welcome to Lepton REPL!
GNU Guile 2.0.11
Copyright (C) 1995-2014 Free Software Foundation, Inc.

Guile comes with ABSOLUTELY NO WARRANTY; for details type `,show w'.
This program is free software, and you are welcome to redistribute it
under certain conditions; type `,show c' for details.

Enter `,help' for help.
scheme@(guile-user)> (get-nets "U8" "1")
$1 = ("ERROR_INVALID_PIN")
```

But **U8** has a pin **1**. It is, however, a slotted dual comparator with a separate power symbol.

graahnul-grom, 2019-10-14:
There is `set-netlist-mode!()` in the `(netlist mode)` module, which sets
either `'guile` or `'spice` mode. The mode affects how schematics are parsed,
not the behavior of various functions (such as `schematic-package-names()`).
The problem is that it's called in the `(netlist)` module before parsing and before
loading of backend. And the `'spice` mode is set if the backend name contains
"spice".
It's possible to redefine that mode at invocation time:
```
lepton-netlist  -c "(use-modules (netlist mode)) (set-netlist-mode! 'geda)" ...
```
but it's not an option, of course, in your particular case.

noqsi, 2019-10-14:
Could this be repaired so that only the "spice" and "spice-sdb" back ends trigger it? It's a kludgy work-around for the design error of overloading *pinseq*, but I would hope that future back ends do better. SPICE netlisting should be fundamentally the same as layout netlisting.

noqsi, 2019-10-14:
I've pushed a new version to the lepton branch at [github](https://github.com/noqsi/gnet-spice-noqsi). I've renamed it gnet-**space**-noqsi for now ツ 

I've been using it for real work, it seems good.

graahnul-grom, 2019-10-14:
> Could this be repaired so that only the "spice" and "spice-sdb" back ends trigger it?

It's possible (and quite easy).
But how do you process slotted components?

graahnul-grom, 2019-10-14:
Selecting which mode to use by parsing the backend's name is not a good solution, IMHO.
I think that backends should be converted to guile modules, first.
Then, situations like this could easily be handled by, say, calling a function like
`what-mode-i-want-to-use()`, exported from that module. So, we can define
the interface for backend modules.

noqsi, 2019-10-14:
I normally do slotted components by pinnumber. Here's the default prototype for a dual opamp with conventional pinout and model pin order:

```
X1? #3 #2 #8 #4 #1 model-name@
X2? #5 #6 #8 #4 #7 model-name@
```

graahnul-grom, 2019-10-14:
"~A uses slotting. You must list its connections by pinnumber, not pinseq."
Could you please explain the meaning of this error message?

graahnul-grom, 2019-10-14:
Does it mean that I can not use several components with the same `refdes`
and different `slot` attributes?

noqsi, 2019-10-14:
spice-noqsi has no standard convention for mapping slots to models through *pinseq*. I don't even understand how that might work for heterogeneous devices. What do you do when the two slot device has three symbol instances (two devices and power)?

graahnul-grom, 2019-10-14:
Didn't think about that; I never use separate power symbols.

noqsi, 2019-10-14:
You can use several symbol instances with the same `refdes`. They need not have slots, if they have different pin numbers. Or some may have slots. The usual machinery figures out the pin numbers, and then the prototypes for such devices use those, in arbitrary order, to generate an arbitrary number of SPICE "cards".

noqsi, 2019-10-14:
I think it's much easier to have a pinout and a model in front of you, and compose a prototype from them rather than wrapping your mind around the `pinseq` definition. Let `pinseq` do its main job of managing pin numbering of slots.

noqsi, 2019-10-14:
One problem with library symbols is that the author may not have had a SPICE model in mind when assigning pinseq. It's easy to fix this with a prototype that uses pin numbers.

graahnul-grom, 2019-10-14:
Agree, all that `pinseq` stuff is not so obvious.

graahnul-grom, 2019-10-14:
If I want to simulate a dual opamp, then I have to create two symbols (with different pin numbers), right?

noqsi, 2019-10-15:
You can use slots. I usually use Kai-Martin Knaak's [opamp_dual](http://www.gedasymbols.org/user/kaimartin_knaak/essential/symbols/analog/opamp_dual.sym) and [opamp_dual_pwr](http://www.gedasymbols.org/user/kaimartin_knaak/essential/symbols/analog/opamp_dual_pwr.sym) or some variant.

noqsi, 2019-10-15:
Use `pinseq` and `slotdef` to provide the pin numbers for your slots, as you must for layout, and then have `spice-noqsi` use those pin numbers, too. The default prototype for `device=DUAL_OPAMP` works for the common pin convention and model convention (but not every supplier uses them, so check).

graahnul-grom, 2019-10-15:
Aha, got it!
1) add instances of slotted component to the schematic
2) set the `slot` attributes for them to make lepton-schematic renumber pins
3) set the `slot` attributes to `0`

noqsi, 2019-10-15:
What? I never set slot attributes to 0. I just draw the circuit for layout, using slots the usual way, and then add a few `spice-prototype` and other attributes (`model-name`, etc.) to symbols for which the default prototype isn't right or more data is needed for simulation. The idea is that the SPICE netlister should always work with a schematic drawn for a PC board. It should never ask of you that you use attributes in ways that are incompatible with your design for your board.

graahnul-grom, 2019-10-15:
Sorry, I meant `numslots` in item (3) above.
The code checks if `numslots > 0` and reports error in that case:
"R1 uses slotting. You must list its connections by pinnumber, not pinseq."

graahnul-grom, 2019-11-22:
@noqsi @vzh I have removed the kludge from lepton-netlist that decides what
mode should be used (`'geda` or `'spice`) based on the backend's name.
Backend can request the desired mode by defining `request-netlist-mode()`
function in its code. Only `spice` and `spice-sdb` backends have to be fixed.
John, could you please rename your file back to `gnet-spice-noqsi`, update
lepton sources and test the attached patch? If it works, I'll make a PR from
the [tb_226_netlist_mode](https://github.com/graahnul-grom/lepton-eda/tree/tb_226_netlist_mode) branch.

[tb226.diff.txt](https://github.com/lepton-eda/lepton-eda/files/3880063/tb226.diff.txt)

graahnul-grom, 2019-11-25:
I've tested [`gnet-spice-noqsi`](https://github.com/noqsi/gnet-spice-noqsi/blob/lepton/gnet-space-noqsi.scm) with [tb_226_netlist_mode](https://github.com/graahnul-grom/lepton-eda/tree/tb_226_netlist_mode) changes and it works.

graahnul-grom, 2019-11-26:
#513

vzh, 2019-12-08:
@noqsi,
Could you please review #513 if it fixes the issue without the need to rename your backend?

PR #513 (2019-12-21): Allow lepton-netlist backends to set the netlist mode ('spice or 'geda)
- Netlister backends can now request the desired mode by defining
`request-netlist-mode()` function, which (if present) should return
either `'geda` or `'spice`. By default, the `'geda` mode is used.
- Remove the old kludge from lepton-netlist code, that decides what
mode should be used based on the backend's file name (it sets the `'spice`
mode if backend filename contains `"spice"`, e.g. `gnet-spice.scm`,
`gnet-spice-sdb.scm`, `gnet-spice-noqsi.scm`).
- Add `request-netlist-mode()` function to the `gnet-spice.scm` and
`g [... cut]
closed 2019-12-21 by graahnul-grom