lepton-eda/lepton-eda #480: Core dump on print

noqsi, 2019-10-15:
In 1.9.9, File->Print causes the following:

> (lepton-schematic:23671): Gtk-WARNING **: Unknown paper size 

** (process:23671): WARNING (recursed) **: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Avahi was not provided by any .service filesAborted (core dumped)

graahnul-grom, 2019-10-15:
Could be solved by installing and running `avahi-daemon`.
Though it's strange: other apps print the same error message, but do not crash.
Paper size can be set in configuration: `[gschem.printing]::paper` key (e.g. "iso_a4"), but it's not related to crash.

vzh, 2019-10-17:
@noqsi, I can confirm the crash on Debian (mix of stable/oldstable/testing).
If i comment out those two lines in `schematic/src/gschem_log_widget.c:348`, it works without crash:
```
  while (gtk_events_pending())
    gtk_main_iteration();
```

Could you try this?

@graahnul-grom, it looks like there are some racing conditions when using `gtk_main_iteration` in such a way and many lines are added to the log (by `g_log`).  Probably it affects only some systems, though we have to fix this. Please look into it.

noqsi, 2019-10-17:
This prevents the crash on my Xenial VM.

> @noqsi <https://github.com/noqsi>, I can confirm the crash on Debian (mix of stable/oldstable/testing).
> If i comment out those two lines in schematic/src/gschem_log_widget.c:348, it works without [...]

John Doty              Noqsi Aerospace, Ltd.
jpd@noqsi.com

vzh, 2019-10-18:
Thanks, John.
I'll close the issue.

closed 2019-10-18 by vzh

graahnul-grom, 2019-10-18:
Cannot reproduce it, on FreeBSD File->Print works fine.

vzh, 2019-10-18:
@graahnul-grom, it could be an issue on Debian and its derivations.  It crashed on my PC, too, as I already said, and John has confirmed that my fix worked.  I decided it's better to not deprive our users of the printing functionality until we fix it otherwise, so I committed my work-around patch, sorry.

graahnul-grom, 2019-10-18:
You were right committing this, there's nothing to be sorry about.
Debugging shows that there is indeed a race condition when GTK tries to log
a message.

graahnul-grom, 2019-10-18:
Though it breaks scrolling in the log window (reported in #238, fixed in #239).
@vzh @noqsi Could you please try the attached patch? It should fix the
scrolling issue (using a different trick). Will it crash on Debian?

[log_scroll.diff.txt](https://github.com/lepton-eda/lepton-eda/files/3745600/log_scroll.diff.txt)

noqsi, 2019-10-18:
‘patch’ doesn’t like the patch versus 1.9.9

Hunk #3 FAILED at 314.
1 out of 4 hunks FAILED

> Though it breaks scrolling in the log window (reported in #238 <https://github.com/lepton-eda/lepton-eda/issues/238>, fixed in #239 <https://github.com/lepton-eda/lepton-eda/pull/239>).
> @vzh [...]

John Doty              Noqsi Aerospace, Ltd.
jpd@noqsi.com

graahnul-grom, 2019-10-18:
@noqsi This patch should be applied to the latest git source tree.

graahnul-grom, 2019-10-18:
@noqsi Here's the patch against 1.9.9 release.

[log_scroll_199.diff.txt](https://github.com/lepton-eda/lepton-eda/files/3745800/log_scroll_199.diff.txt)

noqsi, 2019-10-18:
Just rebuilt from patched git, all is well. No crash on print.

> 
> @noqsi <https://github.com/noqsi> Here's the patch against 1.9.9 release.
> 
> log_scroll_199.diff.txt <https://github.com/lepton-eda/lepton-eda/files/3745800/log_scroll_199.diff.txt>

John Doty              Noqsi Aerospace, Ltd.
jpd@noqsi.com

graahnul-grom, 2019-10-18:
Thank you, John. And what about scrolling? Does the window scroll to the bottom
when something is written to the log?

noqsi, 2019-10-18:
> 
> Thank you, John. And what about scrolling? Does the window scroll to the bottom
> when something is written to the log?

Yes.

John Doty              Noqsi Aerospace, Ltd.
jpd@noqsi.com

graahnul-grom, 2019-10-18:
Отлично, сейчас запилим PR. Ваше здоровье, камрад Джон!

noqsi, 2019-10-18:
> 
> Отлично, сейчас запилим PR. Ваше здоровье, камрад Джон!

何ですか？

John Doty              Noqsi Aerospace, Ltd.
jpd@noqsi.com

vzh, 2019-10-19:
@noqsi, he jokes :-) Literally, it's something like "Greate, we'll make a PR just now. (A cup of wine, vodka, or some) to you health, comrade John!"

@graahnul-grom, your patch works great, anything is scrolled correctly. Waiting for the promised PR.

PR #482 (2019-10-18): schematic: Fix crash on yielding 'File -> Print'.
This patch fixes issue #480 on my home PC.
Closes #480.