lepton-eda/lepton-eda #483: Shorted net complaints

noqsi, 2019-10-19:
Using lepton-netlist 1.9.9.20191003 (git: ad4ada5), I get numerous:

> CRITICAL **: Rename shorted nets (net= has priority): GND -> GND

It doesn't appear to cause errors in the netlist.

graahnul-grom, 2019-10-24:
I see this, too. It seems that it happens with all kinds of schematics (hierarchical or not) and with all backends.

graahnul-grom, 2019-11-23:
It's `check-shorted-nets()` function in the `(netlist hierarchy)` module.
```scheme
(define (check-shorted-nets a b priority)
    (log! 'critical
          (_ "Rename shorted nets (~A= has priority): ~A -> ~A")
          priority
          a
          b)
    (add-net-rename a b))
```

graahnul-grom, 2020-01-02:
In case somebody got tired with those self shorted nets warnings,
here's a temporary solution.

```diff
diff --git a/netlist/scheme/netlist/hierarchy.scm b/netlist/scheme/netlist/hierarchy.scm
index 9d4ed4f1d..6d4b80dfa 100644
--- a/netlist/scheme/netlist/hierarchy.scm
+++ b/netlist/scheme/netlist/hierarchy.scm
@@ -160,11 +160,13 @@
   (define tag (schematic-component-tag component))
 
   (define (check-shorted-nets a b priority)
+    (unless (string=? a b)
     (log! 'critical
           (_ "Rename shorted nets (~A= has priority): ~A -> ~A")
           priority
           a
           b)
+    )
     (add-net-rename a b))
 
   (define (unnamed-net-or-unconnected-pin? name)
```

[self_shorted_nets.diff.txt](https://github.com/lepton-eda/lepton-eda/files/4017032/self_shorted_nets.diff.txt)

PR #540 (2020-01-05): Suppress lepton-netlist warnings about nets shorted to themselves 
Closes #483.
closed 2020-01-05 by vzh