lepton-eda/lepton-eda #528: gsch2pcb work <--> lepton-sch2pcb does not, different netlists

nik570, 2019-12-18:
Then working with a schematic with hierarchy I have discovered gsch2pcb work but then using leptong-sch2pcb net name is not propagated down in hierarchy and there is no warning.

vzh, 2019-12-18:
@nik570,
A little example schematic would be very handy for us in order to solve your issue.
Thanks in advance.

nik570, 2019-12-18:
Searched and did not find anywhere to put example schematic, found button with funny looking faces. Sent attached files to this mailing list a few hours ago, example was not so small and are not certain it work without symbols/packages.

Regards Nicklas Karsson

graahnul-grom, 2019-12-19:
@nik570 At the bottom of the box where you write a comment, there is text
saying "Attach files by dragging & dropping, selecting or pasting them".
Click on the "selecting" word - and the "File Upload" dialog will appear.

vzh, 2019-12-19:
Nicklas, I haven't seen any attachments in your posts on geda-user, sorry.

nik570, 2019-12-19:
Example files attached, missed the text on bottom and got distracted by the funny looking figures.

graahnul-grom, 2019-12-20:
@nik570 Supported file extensions are listed in the drop-down box of the
"File Upload" dialog (at its bottom-right corner). `zip` is among them, so you
can create a zip archive with your files and upload it here.

nik570, 2019-12-21:
[test_sch2pcb.zip](https://github.com/lepton-eda/lepton-eda/files/3991580/test_sch2pcb.zip)
Uploaded files with test case.

nik570, 2019-12-28:
Think I nailed the problem to several layers of hierarchy. gsch2pcb work with several layers of hierarchy while lepton-sch2pcb work only with single layer of hierarchy. Same schematic but layers further down is not connected.

vzh, 2019-12-29:
@nik570,
Could you please add missing symbols to the archive (or just embed them to your schematics)?
Symbols I cannot see are as follows:
LM1117-3.3.sym (steppersBoard.sch)
Power_1in_1out.sym (steppersBoard.sch)
STM32F103xx_LQFP48.sym (steppersBoard.sch)
Fiducial.sym (steppersBoard.sch)
dual_diode_series-1.sym (StepperL6474.sch)
L6474_36PIN.sym (StepperL6474.sch)
connector14-1.sym (SPI.sch)
DS8295A.sym (Power 4.5-23V in 5V 2A out.sch, Power 4.5-23V in 3.3V 2A out.sch)
ISO7421.sym (CAN_UART.sch)

Thank you for your report!

nik570, 2019-12-29:
Embedded symbols and added symbols in separate file in case I missed some. If I know about it embedding had been a better option from start then updated could be done then decided  and no symbols will be missing.

[stepperSym.zip](https://github.com/lepton-eda/lepton-eda/files/4008306/stepperSym.zip)
[Uploading steppersBoard_embedded_components.zip…]()

Karl Aspo Data also provided me a working example with multiple levels of hierarchy there net attribute had been use. Though I still consider it a problem gsch2pcb work with several layers of attribute but lepton-sch2pcb only work with a single layer then refdes is used and guess it is not supposed to be like this.

vzh, 2019-12-30:
Nicklas, I cannot promise I'll look into the issue just now, I'll have holidays and am going to go to my mum, so I'm going to look into it after my holidays, sorry. Happy New Year!

nik570, 2019-12-30:
> Nicklas, I cannot promise I'll look into the issue just now, I'll have holidays and am going to go to my mum, so I'm going to look into it after my holidays, sorry. Happy New Year!

Of course not, you have no obligations to do anything. I have to be grateful for anything that get done.

graahnul-grom, 2020-01-03:
@nik570
Several points:

1. Happy New Year, Nicklas!
2. `lepton-sch2pcb` has nothing to do with errors in generated netlists.
As can be seen in its [man page](https://graahnul-grom.github.io/ref-man/lepton-sch2pcb.html),
it only invokes the netlister program three times to produce appropriate
files that can be read by the `pcb` layout editor.
3. Wherever the actual problem is, in situation where we want to discover
whether a particular tool do perform correctly a certain task, it's much
better to design a test case (as simple as possible) and do not try to find
where and why it misbehaves in large, complex projects.
4. Having said that, I propose to check whether `lepton-netlist` supports
more than two levels of hierarchy (IIUC, that is your problem, isn't it?).
I have constructed a set of hierarchical schematics (3 levels) for that purpose.
Let's begin with the following simple current source circuit (the first
level sub-circuit we want to make is enclosed in the green dotted box,
the second level one is in the dashed blue box):

[image]

The hierarchy will look like this:

[image]

Main schematic:

[image]

First sub-schematic:

[image]

Second sub-schematic:

[image]

`geda.conf` configuration file:

```ini
[gnetlist.hierarchy]
mangle-net-attribute=false
mangle-netname-attribute=false
mangle-refdes-attribute=false
```

Then we run `lepton-netlist` with the `PCB` backend:

```sh
$ lepton-netlist -o - -g PCB main.sch

GND       R3-2  V1-2  V2-2
base      Q1-2  R1-2
csin      R1-1  V2-1
emitter   Q1-3  R2-1
mid       R2-2  R3-1
vcc       Q1-1  V1-1
```

It looks like a correct netlist.
I did not use name mangling in this case for clarity, but have
had the same results when it is on.

Now try `lepton-sch2pcb` (the `project.sch2pcb` project file
is included in the archive attached):

```sh
$ lepton-sch2pcb project.sch2pcb
```

And, sure enough, if we can produce a correct netlist, there will be
no problems with `lepton-sch2pcb`. Here is our advanced board done:

[image]

So, `lepton-netlist` works for me with multi-level hierarchical
schematics. Please check if it works for you.
Of course, this is a very simple example, but it's a "proof of concept",
nevertheless.
All required files are in the attached `tar.gz` archive along with
the `README` describing the archive's contents.

[hier2nick.tar.gz](https://github.com/lepton-eda/lepton-eda/files/4021032/hier2nick.tar.gz)

nik570, 2020-01-04:
> 2. `lepton-sch2pcb` has nothing to do with errors in generated netlists.
> As can be seen in its [man page](https://graahnul-grom.github.io/ref-man/lepton-sch2pcb.html),
> it only invokes the netlister program three times to produce appropriate
> files that can be read by the `pcb` layout editor.

Well lepton-sch2pcb is what I use and I know it call lepton-netlist to get work done.

> 3. Wherever the actual problem is, in situation where we want to discover
> whether a particular tool do perform correctly a certain task, it's much
> better to design a test case (as simple as possible) and do not try to find
> where and why it misbehaves in large, complex projects.

Inded but as someone asked and time was limited I used what I had at hand.

> 4. Having said that, I propose to check whether `lepton-netlist` supports
> more than two levels of hierarchy (IIUC, that is your problem, isn't it?).
> ...
> [hier2nick.tar.gz](https://github.com/lepton-eda/lepton-eda/files/4021032/hier2nick.tar.gz)

Yes this one work. I have also concluded for my schematic there is a difference between gsch2pcb and lepton-sch2pcb and lepton-sch2pcb work with one layer of hierachy, will try to nail it further with a more simple example.

vzh, 2020-01-05:
@graahnul-grom, @nik570,
I am sorry, though I should state our Lepton netlister has some issues with hierarchy.  While it correctly processes all tests in the suite, there are reports (e.g., a report from Karl Hammar) that it does the job wrongly in some cases (I have not opened an issue about this yet, sorry).  There can be some inherited and probably some new issues (after its full rewriting in Scheme).  And the issues with the netlister are first in my list of to-do's.  Unfortunately, other things (like renaming after forking) have slowed down our work significantly (and there are lots of work to do)...
@graahnul-grom,
your examples like this would look much more impressive in our wiki ;-) Hint, hint...

vzh, 2020-01-17:
@nik570,
I seem to have fixed this in #546. `lepton-sch2pcb`'s output differs from `gsch2pcb`'s one since in Lepton all elements are sorted.  However, I have carefully checked all output files and they now contain the same data when building with either tool (names of unnamed nets differ, though).  Please try to compile that branch and check if it works for you.  Your feedback is much appreciated. Thanks.

closed 2020-01-18 by vzh