lepton-eda/lepton-eda #529: Segfault when locale not supported by system

noqsi, 2019-12-21:
On an Ubuntu 16.04.6 VM without Japanese support:

> LANG=ja_JP.UTF-8 lepton-schematic
> 
> (process:12486): Gtk-WARNING **: Locale not supported by C library.
> 	Using the fallback 'C' locale.
> Backtrace:
> In ice-9/boot-9.scm:
>  157: 3 [catch #t #<catch-closure bd5a40> ...]
> In unknown file:
>    ?: 2 [apply-smob/1 #<catch-closure bd5a40>]
> In ice-9/boot-9.scm:
>  157: 1 [catch #t #<catch-closure bd5220> ...]
> In unknown file:
>    ?: 0 [apply-smob/1 #<catch-closure bd5220>]
> 
> ERROR: In procedure apply-smob/1:
> ERROR: In procedure setlocale: Invalid argument
> 
> (lepton-schematic:12486): GLib-CRITICAL **: g_hash_table_lookup: assertion 'hash_table != NULL' failed
> 
> (lepton-schematic:12486): GLib-CRITICAL **: g_hash_table_lookup: assertion 'hash_table != NULL' failed
> 
> (lepton-schematic:12486): GLib-CRITICAL **: g_hash_table_insert_internal: assertion 'hash_table != NULL' failed
> Segmentation fault (core dumped)
> $ lepton-schematic --version
> Lepton EDA 1.9.9 (git: 0d60339)

With language-pack-ja installed, all is well. Of course, I don't expect it to work without support for the language, but it shouldn't segfault.

graahnul-grom, 2019-12-22:
```
$ locale -a | grep JP
ja_JP.SJIS
ja_JP.UTF-8
ja_JP.eucJP

$ LANG=ja_JP.UTF-8 lepton-schematic
```

[image]

vzh, 2019-12-22:
@graahnul-grom,
@noqsi says that the program segfaults when the appropriate locale has NOT been installed.
I can confirm this.  Have seen this about a month ago when tried to demonstrate lepton-schematic working in other locale to my colleagues at work.
Your picture just demonstrates that you don't have necessary Japan fonts.

noqsi, 2019-12-22:
@graahnul-grom 
Looks good if you have the locale and the fonts installed.

[image]

vzh, 2019-12-22:
John, please check if #532 fixes the issue for you.

PR #532 (2019-12-22): Fix segfaults when appropriate locale has not been installed.
Closes #529
closed 2019-12-22 by vzh