lepton-eda/lepton-eda #545: Reorganize the context menu

graahnul-grom, 2020-01-16:
Although I do not use context menu in lepton-schematic and still believe
that the right mouse button should be used for mouse panning (#369, point 5),
it's time to update it with more relevant menu items.
Here's the screenshots of current context menu (on the left)
and possible replacement (on the right):

[image]

Any suggestions?

vzh, 2020-01-18:
You don't like buses, as I can see :-)  I wouldn't throw them out.
Getting rid of zooming entries is OK with me (why are they there?).  I would remove `Select` as well.
Moving `Add component` to the top seems to be a good idea.

graahnul-grom, 2020-01-19:
[image]

vzh, 2020-01-19:
Yeah, this last version suits me well :-)

PR #548 (2020-01-21): Reorganize the context menu
[image]

Closes #545
closed 2020-01-21 by vzh