lepton-eda/lepton-eda #557: netlist not creating expected results

drforbin, 2020-01-31:
expected results netllister

Example diode:

    refdes = D5
    model-name = 1N1004
    model = IS=0.5UA RS=6 BV=5.20

SPICE lines generated: D5 2 6 1N1004 MODEL 1N1004 D (IS=0.5UA RS=6 BV=5.20)

supporting files attached.

Also bug related to including ONE LINE spice models via spice include symbol.

When a one line spice model is included netlist fails.
netlist works as expected if a multi-line model used. 

[files.zip](https://github.com/lepton-eda/lepton-eda/files/4140716/files.zip)

vzh, 2020-02-01:
@drforbin,
IIUC, you use the `spice-sdb` backend?
What it outputs for me when processing your schematic is:
```
* lepton-netlist -o - -g spice-sdb doubler.sch
*********************************************************
* Spice file generated by lepton-netlist                *
* spice-sdb by SDB                                      *
* provides advanced spice netlisting capability.        *
* Documentation at wiki.geda-project.org/geda:csygas    *
*********************************************************
*==============  Begin SPICE netlist of main design ============
C1 2 1 1000p
C2 0 1 1000p
D1 0 1 D1N4148
.MODEL D1N4148 D (D1N4148)
V1 2 0 sin(0 5 1k)
.end
```
As i can see, both [gEDA SPICE HOWTO by SDB](http://wiki.geda-project.org/geda:csygas) and netlister do show wrong results. IIUC, the result from your schematic should be as follows:
```
...
D1 0 1 D1N4148
.MODEL D1N4148 D (IS=0.1PA, RS=16 CJO=2PF TT=12N BV=100 IBV=0.1PA)
...
```
Right?

graahnul-grom, 2020-02-01:
Please try the attached patch.

```diff
diff --git a/netlist/scheme/backend/gnet-spice-sdb.scm b/netlist/scheme/backend/gnet-spice-sdb.scm
index 1f475a48f..fb5ea0880 100644
--- a/netlist/scheme/backend/gnet-spice-sdb.scm
+++ b/netlist/scheme/backend/gnet-spice-sdb.scm
@@ -351,7 +351,7 @@
 
     ;; Now write out any model which is pointed to by the part.
     (when (not (unknown? model))
-      (format #t ".MODEL ~A ~A (~A)\n" model-name type package-model))))
+      (format #t ".MODEL ~A ~A (~A)\n" model-name type model))))
 
 
 ;;  Writes diode SPICE card for PACKAGE.
```

[gnet-spice-sdb_fix_model.diff.txt](https://github.com/lepton-eda/lepton-eda/files/4142401/gnet-spice-sdb_fix_model.diff.txt)

vzh, 2020-02-01:
Emm, the better solution would be:
```
diff --git a/netlist/scheme/backend/gnet-spice-sdb.scm b/netlist/scheme/backend/gnet-spice-sdb.scm
index 1f475a48f..6bc0472af 100644
--- a/netlist/scheme/backend/gnet-spice-sdb.scm
+++ b/netlist/scheme/backend/gnet-spice-sdb.scm
@@ -331,7 +331,7 @@
          (area (gnetlist:get-package-attribute package "area"))
          (off (gnetlist:get-package-attribute package "off"))
          (model-file (gnetlist:get-package-attribute package "file"))
-         (package-model (if (unknown? model-name) value model-name)))
+         (package-model (if (unknown? model-name) value model)))
 
     ;; Write out the refdes prefix, if specified and necessary.
     (when prefix (spice-sdb:write-prefix package prefix))
```

Otherwise, you'll lose models specified by the `value=` attribute.

vzh, 2020-02-01:
And let's keep the issue open until not only a fix, but a test schematic goes to the repository (I suggest the schematic from the archive by @drforbin).

graahnul-grom, 2020-02-01:
I'm not sure if SPICE model (IS=0.1PA, RS=16 etc. line) could be
specified via the `value` attribute.

graahnul-grom, 2020-02-01:
With you patch I get this:
```
D1 0 mid IS=0.1PA, RS=16 CJO=2PF TT=12N BV=100 IBV=0.1PA
.MODEL D1N4148 D (IS=0.1PA, RS=16 CJO=2PF TT=12N BV=100 IBV=0.1PA)
```

vzh, 2020-02-01:
Ah, yes, didn't check it.  Your solution seems better.
@drforbin, please check if @graahnul-grom's fix works for you.

drforbin, 2020-02-01:
Vlad. Not suppose to specify models with value attribute.

noqsi, 2020-02-01:
The difficulty of dealing with models in spice-sdb style is that there is a lot of this hard-wired code for common devices, along with heuristics that attempt to figure out how it applies. It doesn't always work the way you want, especially if your devices are uncommon. This is one reason why I wrote the [spice-noqsi](https://github.com/noqsi/gnet-spice-noqsi/tree/lepton) back end. It does most of the things spice-sdb does by default, but allows the user to supply a spice-prototype attribute if different behavior is required. Its "lepton" branch was working a few months ago, but I haven't tried it since: lepton-netlist seems a bit of a moving target.

drforbin, 2020-02-01:
I will apply the patch and get back with you all.
There is another issue with include files as well.
I have already told Vlad about this.

Should I post it here or open another issue?

vzh, 2020-02-02:
Merlyn, please open a new issue for another bug.

drforbin, 2020-02-02:
[files.zip](https://github.com/lepton-eda/lepton-eda/files/4144166/files.zip)

I tried the patch, it did no work.
Here are the files I used.

graahnul-grom, 2020-02-02:
> [files.zip](https://github.com/lepton-eda/lepton-eda/files/4144166/files.zip)
> 
> I tried the patch, it did no work.
> Here are the files I used.

Please provide us with the following information:
1. The command you run
2. The output you expect
3. The output you actually got

drforbin, 2020-02-02:
The command line is lepton-net 0g spice-sdb -o out.spice doubler.sch

[tryer.txt](https://github.com/lepton-eda/lepton-eda/files/4145141/tryer.txt)
attached file is obtained output.

expected outcome 

[frg_band_pass.txt](https://github.com/lepton-eda/lepton-eda/files/4145144/frg_band_pass.txt)

I apologize. I would fix this myself but I donot know guile.
Any suggestions for me to learn it?

drforbin, 2020-02-02:
Please double check. I think I was wrong. I believe your patch fixes the problem.
Please double check for me with above files and give me some references for guile.

thank you

drforbin, 2020-02-02:
FYI..
it is producing a multi-line model.
It reall should be something like this.

D5 2 6 1N1004 MODEL 1N1004 D (IS=0.5UA RS=6 BV=5.20)

noqsi, 2020-02-02:
What SPICE dialect is 

> D5 2 6 1N1004 MODEL 1N1004 D (IS=0.5UA RS=6 BV=5.20)

?

In most SPICE dialects, that kind of model requires a separate *.MODEL* "card". No general-purpose SPICE netlister should produce the above by default.

drforbin, 2020-02-02:
I did alittle research, it seems your correct.
The reference for gEDA and lepton shows under the wiki it should produce
the above line (D5 2 6 1N1004 MODEL 1N1004 D (IS=0.5UA RS=6 BV=5.20)) .
This wiki was produced by the writer of spice-sdb backend.

I always use a separate model line. 

Thank you.

noqsi, 2020-02-02:
Stuart's original document was not part of a wiki. It may have lost a newline in translation.

drforbin, 2020-02-02:
Ok thank you.

Well it was part of the howto. 

I looked through spice2/spice3 and ngspice doc and as I said it seems you are correct.
These kinda of things should be clarified. This is valuable.

vzh, 2020-02-03:
@drforbin, @graahnul-grom, @noqsi, thank you all!
@graahnul-grom, please open a new PR with your patch.
Thanks.

vzh, 2020-02-03:
@drforbin,
as for guile Scheme, [schemers.org](https://schemers.org/), especially [this page](https://schemers.org/Documents/#all-texts), contains lots of info. Guile specific info can be found in [Guile Manual](https://www.gnu.org/software/guile/manual/).

drforbin, 2020-02-03:
Thank you...patch works confirmed

PR #564 (2020-02-09): Fix .MODEL line generation in the spice-sdb lepton-netlist backend
For components with the `model-name` and `model`
attributes, incorrect output was generated, for example:
```
model-name = D1N4148
model = IS=0.1P RS=16 CJO=2P TT=12N BV=100 IBV=0.1P
```

Incorrect output:
```
D1 node1 node2 D1N4148
.MODEL D1N4148 D (D1N4148)
```

The correct output should be:
```
D1 node1 node2 D1N4148
.MODEL D1N4148 D (IS=0.1P RS=16 CJO=2P TT=12N BV=100 IBV=0.1P)
```

Closes #557.
closed 2020-02-09 by vzh