lepton-eda/lepton-eda #558: problem with spice  inclued models

drforbin, 2020-02-02:
When a model is included with symbol spice include directive and the referenced file
is NOT multi-line it fails.

Please find included files proving failure.
[first.zip](https://github.com/lepton-eda/lepton-eda/files/4145304/first.zip)
[second.zip](https://github.com/lepton-eda/lepton-eda/files/4145305/second.zip)

drforbin, 2020-02-03:
Please confirm before close.
Upon review my distro Gentoo is running all output directed to 'less' to |pipeless first before display. Apparently it does not like spice files and thinks they are something else.
Therefore It processes them and screws the output up.
Please confirm that the spice generation is correct on you all ends.
If so it's just my distro pre processing threw less and this issue  can be closed.  

Thanks you all.

drforbin, 2020-02-03:
If so I can get onto productive work finishing up more example schematics!

vzh, 2020-02-03:
The files look OK for me.

vzh, 2020-02-03:
I close this issue. Please re-open it if something is really wrong.

closed 2020-02-03 by vzh

graahnul-grom, 2020-02-06:
> ...my distro Gentoo...

Merlyn, are you using the Dominique's Gentoo overlay with lepton-eda `ebuild` [1]?
I think we should add the link to it to our Gentoo wiki page [2].

[1] https://github.com/domichel/GenCool/tree/master/sci-electronics/lepton-eda
[2] https://github.com/lepton-eda/lepton-eda/wiki/Gentoo