lepton-eda/lepton-eda #601: Add a way to use add-ons without using any program options or RC files.

vzh, 2020-03-17:
It would be nice to have a mechanism to support add-ons/packages.
The user of an add-on should be able to install it without
figuring out where to put the components. The user doesn't have to
figure out load paths and such.  The user may want to install a
back end.  It can import some modules and use some data files.

One lispy example of how it can be implemented is Emacs. It just
uses a set of predefined file and directory names to store
packages (modules), and each of them can use its current
installation directory or specify another one to store required
files. We already have something like this on the system level,
the directory `autoload` contains some files required for the
tools to function.  We could extend this by adding some user
autoload directory.

So in this case, the user configuration directory could, say, have
the subdirectory `autoload` which would be automatically added to
Guile `%load-path`. Storing and searching for data files could be
done by package code in any subdir it lives itself or in the
`autoload` directory. The directory name is not significant, IMO,
another one, e.g. `scheme`, could be used instead.

Please see this [discussion](https://github.com/orgs/lepton-eda/teams/lepton-developers/discussions/2) for more.

graahnul-grom, 2020-03-17:
FWIW, `$HOME/.local/share/lepton-eda/scheme/` is automatically added
to the `%load-path`.

noqsi, 2020-03-17:
> $HOME/.local/share/lepton-eda/scheme/ is automatically added
to the %load-path

And then there's `user-data-dir` (`$HOME/.gEDA`) for data files.

graahnul-grom, 2020-03-17:
That is, `$XDG_DATA_HOME/lepton-eda/scheme/`.
`eda_get_user_data_dir()`

graahnul-grom, 2020-03-17:
`user-data-dir()` no longer returns `$HOME/.gEDA`. It was changed
a long time ago. It returns the directory just mentioned above.

noqsi, 2020-03-17:
Oops, wrong version. For production, I was using an old version for stability. Time to retire that. I have (git: 602f004) installed for new development.

graahnul-grom, 2020-03-17:
I would recommend to update to the latest git: there are significant
changes after that commit, namely, we do not use `geda*.conf` files
anymore, and some configuration groups were renamed
(e.g. `gnetlist*` => `netlist*`), please see #598.

noqsi, 2020-03-17:
I updated last week for development purposes. Latest git is a moving target. And I have multiple environments on multiple computers, so I can become confused occasionally ツ

vzh, 2020-03-17:
> FWIW, `$HOME/.local/share/lepton-eda/scheme/` is automatically added
> to the `%load-path`.

Ha, I forgot about this. It's time to do three things:
- eventually move my scheme add-ons/plugins to new locations;
- make a new release;
- and close this issue.

WDYT, friends? ;-)

vzh, 2020-03-18:
After thinking a bit more about the issue, there is one thing I would change. For now, the user can load and set up her/his Scheme functions in `gafrc`.  It would be convenient to specify the name and probably place of this file using the new config system. I suggest adding a setting defining scheme RC file name, relative or absolute, which should be auto-loaded by the lepton tools and contain any necessary set up/loading functions. For example:
```
[scheme]
load="leptonrc"
```
in the `lepton-user.conf` to load the file `~/.config/lepton-eda/leptonrc` automatically every time the user launches `lepton-schematic`, `lepton-netlist`, etc., or
```
[scheme]
load="/tmp/my-new-test-extension.scm"
```
in project's `lepton.conf` to play with that new extension locally in one project.