lepton-eda/lepton-eda #602: lepton-upchg has no man page

bdale, 2020-03-19:
When building lepton-eda 1.9.10 for upload to Debian, our lintian tool pointed out that lepton-upchg has no man page.

vzh, 2020-03-20:
@graahnul-grom, could you tackle this?

graahnul-grom, 2020-03-20:
Dear friends! Let me send here a document that can shed some light
on recent developments (or lack theroff).

Доклад.
Ст. сержанта Пупкина В. В. майору Бузякину А. А.
В процессе патрулирование южной окраины военного городка Н
в составе гарнизонного караула, мною был замечен лейтенант Д,
вызывающее поведение которого и привлекло внимание военнослужащих.
Находясь в окружении красивых женщин, чью социальную значимость
следует поставить под большое сомнение, лейтенант Д позволял
себе делать чудовищные высказывания про веру, царя и отечество.
На законное требование начальника патруля прекратить безобразие
и сдать табельное оружие, лейтенант Д заявил следующее:
We have to write the man page for the lepton-upcfg utility,
and ideally, eliminate all the stupid articles. I just can't
stand all those freakin' "a"'s and "the"'s!, после чего мною была
немедленно вызвана бригада скорой помощи.
В дальнейшем, на допросах задержанным было показано, что
большая часть антисоциального поведения последнего была
вызвана так называемым "днём рождения" и сопутствующими
мероприятиями. Вовлекая в свою удручающую деятельность
всё новых и новых товарищей, лейтенант Д, употребляющий
четвёртый день спиртуозные напитки, безусловно, подрывает
обороноспособность страны...

graahnul-grom, 2020-03-20:
`man` страница будет написана. не вопрос.

vzh, 2020-03-20:
@graahnul-grom
(Please, please, we're not alone here :-)) I'm afraid to suspect who of presented here is the lieutenant D.  The stanza of the lack of articles and one of lack of developments transparently hints it's me }:-| OTOH, the name D looks strange then }}:-|
If seriously, there is no need to write an extensive document: name, usage, output of `--help`, and copyright would be sufficient ;-)

graahnul-grom, 2020-03-22:
My friends, tomorrow we will resume the development process.

[image]

PR #620 (2020-04-03): lepton-upcfg man page
html preview version is [here](https://graahnul-grom.github.io/ref-man/lepton-upcfg.html).

Can you imagine, my friends! Someone takes word "quarantine"
too seriously. They cut off my home phone and internet lines.
It's only yesterday's night I've managed to borrow an old tablet PC,
and, thanks to whoever you put here, it was detected by FreeBSD
and usb tethering was successfully set up. Prosit!

Closes #602
closed 2020-04-03 by graahnul-grom