lepton-eda/lepton-eda #622: Printing Issue : Text Translates In Y-Axis

suboptimallife, 2020-04-06:
I am having a problem when printing schematics from Lepton and Google searches didn't bring up anything relevant. I was using a Lepton build downloaded from source back in January but downloaded 1.9.10.20200319 and the problem carries over between versions. Text which is positioned correctly in the schematic is translated in the y-axis in both print preview and print-to-file output (see attachment [print_test.pdf](https://github.com/lepton-eda/lepton-eda/files/4436818/print_test.pdf)). The issue is repeatable with different schematics, I have created an example schematic (file extension changed for GitHub compatibility [print_test.txt](https://github.com/lepton-eda/lepton-eda/files/4436837/print_test.txt)).

When using lepton-cli export the resulting output file is entirely blank (see attachment [print_test01.pdf](https://github.com/lepton-eda/lepton-eda/files/4436842/print_test01.pdf))

vzh, 2020-04-06:
Could you please embed your title symbol into the schematic file or use `lepton-archive` to get all files necessary to correctly render it?
And  could you please show us the exact `lepton-cli` command you used for export?

suboptimallife, 2020-04-06:
lepton-archive output attached. Commands used were:

lepton-cli export -f pdf -o print_test01.pdf -p iso_a3 -l landscape --no-color print_test.sch
lepton-cli export -f pdf -o print_test01.pdf -p iso_a3 -l landscape print_test.sch
lepton-cli export -o print_test01.pdf print_test.sch

suboptimallife, 2020-04-06:
[project-archive.tar.gz](https://github.com/lepton-eda/lepton-eda/files/4438218/project-archive.tar.gz)

vzh, 2020-04-06:
OK, thanks.  Could you try the following commands and report their output here:
> lepton-cli --version
> lepton-schematic --version

Please try also to `rm ~/.cache/guile -rf` and re-check if `lepton-cli` works or not.

The issue with printing from `lepton-schematic` may be solved in #623.  I would very appreciate if you rebuild Lepton using the branch there and try if it works better.

suboptimallife, 2020-04-07:
lepton-cli --version Lepton EDA 1.9.10.20200319 (git: )
lepton-schematic --version Lepton EDA 1.9.10.20200319 (git: )

rm ~/.cache/guile -rf still results in a blank pdf output from lepton-cli export

suboptimallife, 2020-04-07:
The effect of applying the patch from #623 and rebuilding has been to make the printed font different from that of the rendered font but the displacement still exists.

vzh, 2020-04-07:
The most interesting info is missing, I mean git commit :-)
Have you tried to use another font in GUI using 'Options->Font...'. Please try first to choose another font and save your configuration, then restart `lepton-schematic` and compare rendered and printed fonts.
Another thing to check is what fonts are used on your system. You could find what font is used in your pdf output (say, in `evince` in the Settings dialog), and what your system chooses for Sans (the default font name).  Sometimes, [you may have troubles with some fonts](https://wiki.archlinux.org/index.php/Font_configuration#Calibri,_Cambria,_Monaco,_etc._not_rendering_properly). [Here](http://wiki.geda-project.org/geda:gaf_utility) is some useful info on fonts in gEDA (and hence, in Lepton).  First we have to check those external possibilities of the issue.
For `lepton-cli`, you could also try to setup another font first. For example, on my system I have
> [export]
> font=OpenGost Type B TT Italic

in ~/.config/lepton-eda/lepton-user.conf.

suboptimallife, 2020-04-07:
Played around with various fonts but to no effect. Sans is substituted with Roboto on my system (OpenSuSE Tumbleweed). I shall uninstall Lepton EDA and grab it from git so at least I can provide commit information.

vzh, 2020-04-07:
BTW, what's the alignment of your text? Please try to insert the same multiline block of text with different alignment along Y and show us the result of printing.

suboptimallife, 2020-04-07:
Problem persists with lepton cloned in git:
lepton-cli --version Lepton EDA 1.9.10.20200319 (git: 41225ef)
lepton-schematic --version Lepton EDA 1.9.10.20200319 (git: 41225ef)

I installed evince to see if that had any effect. But sadly no.

I have applied the patch from #623 to this git commit and still the problem persists. If I interpret your comments correctly this seems to be a problem specific to my environment?

vzh, 2020-04-07:
> Problem persists with lepton cloned in git:
> lepton-cli --version Lepton EDA 1.9.10.20200319 (git: [41225ef](https://github.com/lepton-eda/lepton-eda/commit/41225efbc7268a09619dfae9626503fe48a8f503)) [...]

OK, thank you.

> I installed evince to see if that had any effect. But sadly no.

You've probably misunderstood me.  I mentioned `evince` as a tool where you could easily check what font is used in your pdf document.  You could use some TUI utils as well, like, e.g., `pdfinfo` or such, to learn what fonts are really used/embedded into your document.

> I have applied the patch from #623 to this git commit and still the problem persists. If I interpret your comments correctly this seems to be a problem specific to my environment?

It may be superposition of the issues in lepton and your system environment.  About half a year ago Rob Gilton reported on gitter an issue with font overbars after upgrade of his system to a new version of Fedora.  Wrt the issue you reported, I seem to have read something very resembling reported for `geda-gaf` not long ago, that is, the issues can have same roots.  E.g., using of some new cairo/pango versions of libraries or other/new fonts by default.

suboptimallife, 2020-04-07:
> BTW, what's the alignment of your text? Please try to insert the same multiline block of text with different alignment along Y and show us the result of printing.

Sorry, I did not see this message earlier. This is absolutely the source of my issue. Attached is a test file [print_test-alignment.pdf](https://github.com/lepton-eda/lepton-eda/files/4447109/print_test-alignment.pdf) which shows that every alignment option which is LOWER aligned along Y is subject to the unwanted translation. The other six possible options are fine.

For now that means I can work around the issue. Thanks very much for your help.

vzh, 2020-04-08:
Thank you very much!  Now I see that this is not a font issue, at least the part related to `lepton-schematic`.  Some wrong calculation of lower aligned text for landscape orientation.  Please upload here the schematic you've made the last pdf from (print_test-alignment.pdf).

suboptimallife, 2020-04-08:
As requested (file extension changed):
[print_test.txt](https://github.com/lepton-eda/lepton-eda/files/4449002/print_test.txt)

vzh, 2020-04-08:
I'm confused even more.  Tried the same font as in your PDF, different page orientation and sizes, printed via image export and 'print to file',  and no, I could **not** reproduce your issue.  OK, let's try another way.  Could you please post here what your `configure` says during compilation?

suboptimallife, 2020-04-08:
Attached.
[config.log](https://github.com/lepton-eda/lepton-eda/files/4451060/config.log)

graahnul-grom, 2020-04-08:
I'm unable to reproduce it, too.
@suboptimallife IIUC, your system is openSUSE. What version are you using?

suboptimallife, 2020-04-08:
NAME="openSUSE Tumbleweed"
# VERSION="20200305"

vzh, 2020-04-08:
Looks like an issue with [harfbuzz](https://github.com/harfbuzz/harfbuzz/). @suboptimallife, what version of that library do you have?

suboptimallife, 2020-04-08:
2.6.4-1.2

suboptimallife, 2020-04-08:
OK. So, when checking the versioning of harfbuzz I noticed there was an update available to move from 2.6.4-1.2 to 2.6.4-1.3. I executed that update and now things are really goosed.
I am presented with the following dialogue:
[image]
Then this dialogue:
[image]
Then this dialogue:
[image]
The following assertion failures are listed in the terminal:
** (lepton-schematic:3944): CRITICAL **: 22:21:16.677: geda_color_map_get_color: assertion 'map[index].enabled' failed

** (lepton-schematic:3944): CRITICAL **: 22:21:16.983: geda_color_map_get_color: assertion 'map[index].enabled' failed

** (lepton-schematic:3944): CRITICAL **: 22:21:22.957: geda_color_map_get_color: assertion 'map[index].enabled' failed

** (lepton-schematic:3944): CRITICAL **: 22:22:37.521: geda_color_map_get_color: assertion 'map[index].enabled' failed

** (lepton-schematic:3944): CRITICAL **: 22:22:42.339: geda_color_map_get_color: assertion 'map[index].enabled' failed

** (lepton-schematic:3944): CRITICAL **: 22:22:42.603: geda_color_map_get_color: assertion 'map[index].enabled' failed

** (lepton-schematic:3944): CRITICAL **: 22:22:43.433: geda_color_map_get_color: assertion 'map[index].enabled' failed

** (lepton-schematic:3944): CRITICAL **: 22:22:44.379: geda_color_map_get_color: assertion 'map[index].enabled' failed

** (lepton-schematic:3944): CRITICAL **: 22:22:44.535: geda_color_map_get_color: assertion 'map[index].enabled' failed

(lepton-schematic:3944): GLib-CRITICAL **: 22:22:55.610: g_ptr_array_unref: assertion 'array' failed

The lepton-schematic menu is also gone:
[image]

suboptimallife, 2020-04-08:
Rebooted and those recent errors are cleared. Still have Y offset on LOWER aligned text items.

vzh, 2020-04-09:
OK, thanks. Let's go further.
[This script](https://github.com/lepton-eda/lepton-eda/files/4454256/text-add-bounds.scm.txt) adds boxes showing bounds of all toplevel text objects in schematic.  Could you please do as follows:
- Download the script and rename it, say, to **/tmp/text-add-bounds.scm**.
- Open your schematic **print_test-alignment.sch**, hit `:` and enter `(primitive-load "/tmp/text-add-bounds.scm")` in the macro widget entry, then `Enter`.
- Print the schematic to PDF and upload the result here.

suboptimallife, 2020-04-09:
Bounding box script was unsuccessful:

Lepton EDA/lepton-schematic version 1.9.10.20200319 git: 41225ef
Loaded RC file [/usr/local/lepton/share/lepton-eda/system-gafrc]
Loaded RC file [/home/sandy/GDrive/tel-project/BEIS/gafrc]
Loaded RC file [/usr/local/lepton/share/lepton-eda/system-gschemrc]

Loading schematic [/home/sandy/GDrive/tel-project/BEIS/print_test.sch]

Backtrace:
          13 (apply-smob/1 #<catch-closure 7f4d13009560>)
          12 (apply-smob/1 #<catch-closure 7f4d1036e360>)
          11 (eval-string "(use-modules (lepton log)) (log! 'messag…" …)
In ice-9/boot-9.scm:
   2312:4 10 (save-module-excursion #<procedure 7f4d103919e0 at ice-…>)
In ice-9/eval-string.scm:
     38:6  9 (read-and-eval #<input: string 7f4d10389a80> #:lang _)
In ice-9/eval.scm:
   196:43  8 (_ #f)
   202:51  7 (_ #f)
In unknown file:
           6 (primitive-load "/home/sandy/GDrive/tel-project/BEIS/te…")
In ice-9/eval.scm:
   202:51  5 (_ #f)
In srfi/srfi-1.scm:
   592:17  4 (map1 (#<geda-object 0x1bf5f40> #<geda-object 0x1bf6…> …))
In ice-9/eval.scm:
    159:9  3 (_ #(#(#(#<directory (guile-user) 7f4d130b8140>) #) #f))
    155:9  2 (_ _)
In ice-9/boot-9.scm:
   751:25  1 (dispatch-exception 0 wrong-type-arg ("car" "Wrong t…" …))
In unknown file:
           0 (apply-smob/1 #<catch-closure 7f4d1036e320> # "car" "W…" …)

ERROR: In procedure apply-smob/1:
In procedure car: Wrong type argument in position 1 (expecting pair): #f

graahnul-grom, 2020-04-09:
The script works, though it could be slightly modified to
check for text visibility using `text-visible?()` function.
`object-bounds()` returns `#f` for invisible objects.
Please select `Attributes->Show/Hide Hidden Text` in menu
and try again.

graahnul-grom, 2020-04-09:
```diff
--- text-add-bounds.scm.orig	2020-04-09 14:02:01.929834000 +0300
+++ text-add-bounds.scm	2020-04-09 14:05:27.746468000 +0300
@@ -5,7 +5,11 @@
   (let ((bounds (object-bounds object)))
     (make-box (car bounds) (cdr bounds))))
 
+(define (text-ok? object)
+  (and (text? object)
+       (text-visible? object)))
+
 (apply page-append!
        (active-page)
        (map make-box*
-            (filter text? (page-contents (active-page)))))
+            (filter text-ok? (page-contents (active-page)))))
```

[text-add-bounds.scm.diff.txt](https://github.com/lepton-eda/lepton-eda/files/4455781/text-add-bounds.scm.diff.txt)

vzh, 2020-04-09:
Don't know, what version is now compiled on my system, though it works for hidden text objects without modification.

graahnul-grom, 2020-04-09:
It works with 1.9.10 release and doesn't work with the latest git version.

suboptimallife, 2020-04-09:
Applied the patch and bounding boxes are now included in printed output attached 
[print_test-alignment-bounds.pdf](https://github.com/lepton-eda/lepton-eda/files/4456662/print_test-alignment-bounds.pdf)

vzh, 2020-04-09:
@graahnul-grom, @suboptimallife, thank you!  It is obvious the position of lower aligned text depends on its size. I've found that *font descent* is used for such (lower aligned) text objects to calculate their size and position.  I also found an [interesting program](https://gist.github.com/ynkdir/849071) to check the descent of the font used for printing.  As I see the font "Roboto-Regular" was used for export in the last attached PDF by @suboptimallife.  Could you please try to compile it and show its output for the font (you'll have to change it in the code) and your current locale (it should be changed, too, that is, replace "ja-jp" with your current locale settings, as well as in command line as suggested in the comments).

graahnul-grom, 2020-04-09:
The results for "Roboto Regular 10" and "Roboto 10" on my system:
```
lang=default ascent=14.000000 descent=4.000000
lang=C ascent=14.000000 descent=4.000000
lang=ru_RU.UTF-8 ascent=14.000000 descent=4.000000
lang=en_US.UTF-8 ascent=14.000000 descent=4.000000
```

suboptimallife, 2020-04-10:
I hope I've done this right:

Roboto Regular 10> LC_ALL=en_GB ./a.out
lang=default ascent=12.369141 descent=3.254883
lang=C ascent=12.369141 descent=3.254883
lang=en-GB ascent=12.369141 descent=3.254883

Roboto 10>LC_ALL=en_GB ./a.out
lang=default ascent=12.369141 descent=3.254883
lang=C ascent=12.369141 descent=3.254883
lang=en-GB ascent=12.369141 descent=3.254883

graahnul-grom, 2020-04-10:
I've just downloaded and installed openSUSE Tumbleweed 20200407
in VirtualBox, compiled lepton-eda git master and 1.9.10 release.
Here's my observations:

- `lepton-cli export` (master) produces empty `pdf` and `ps` files
- `lepton-cli export` (1.9.10) works, and there's no text shifted
- `export::font` config value doesn't matter
- `File->Print` (and preview) shifts text down, not just in Y
direction (rotated text shifts in X) if two conditions are met:
  1) text size > 8
  2) text alignment is set to either:
    - `lower left`
    - `lower middle`
    - `lower right`

graahnul-grom, 2020-04-10:
@suboptimallife Could you confirm the above behavior, please?
Or is it different on your system?

suboptimallife, 2020-04-11:
Lepton EDA 1.9.10.20200319 (git: c81b378)
`lepton-cli export -o print_test-003.pdf print_test.sch` produces an empty pdf
lepton-schematic print attached which shows different behaviour from @graahnul-grom in that it doesn't seem to matter what the text size is, just the alignment.
[print_test-005.pdf](https://github.com/lepton-eda/lepton-eda/files/4464522/print_test-005.pdf)

graahnul-grom, 2020-04-11:
Could you please answer the following questions:

1) What's the text size of the string "SMALLER_TEXT"?
2) Could you confirm that not only `lower-left`-aligned text is shifted,
but also `lower-middle` and `lower-right` aligned text is affected?
3) Does `lepton-cli` from `1.9.10` *release* (git: 56bb6e5) produce
correct pdf output without any text shifts?

Thank you.

graahnul-grom, 2020-04-11:
Yet another discovery:
In the schematic below, there are 10 text strings sized from 1 to 10,
lower-aligned (left, right, middle - same results).
'File->Print' in lepton-schematic built from the master branch
produces pdf screenshot'ed (and attached) here:

[image]

**All** text is shifted down, except one that has the smallest size.
If I delete "L-L 1" text (which has size == 1), save schematic, and
do 'File->Print' again, "L-L 2" becomes one that does not get shifted.
And so on.

[issue-622-test-dmn.sch.pdf](https://github.com/lepton-eda/lepton-eda/files/4464975/issue-622-test-dmn.sch.pdf)
[issue-622-test-dmn.sch.txt](https://github.com/lepton-eda/lepton-eda/files/4464976/issue-622-test-dmn.sch.txt)

graahnul-grom, 2020-04-11:
Confirmed.
1) In one of the comments [above](https://github.com/lepton-eda/lepton-eda/issues/622#issuecomment-612125330) I said that text size should be equal to, or less than 8 for proper text pdf rendering. That is because it was the smallest text size in my test schematic.
2) If all texts in a schematic have equal size - no problems.

suboptimallife, 2020-04-11:
The SMALLER TEXT is size 6.
lower-left, lower-middle, lower-right are all subject to shifting.
I'll uninstall Lepton EDA 1.9.10.20200319 (git: c81b378) and install release 56bb6e5 and get back to you on the export behaviour.

suboptimallife, 2020-04-11:
Lepton EDA 1.9.10.20200319 (git: 56bb6e5) `lepton-cli export` produces a pdf with no shifts. 
Lepton EDA 1.9.10.20200319 (git: 56bb6e5) `lepton-schematic' does still shift lower aligned text.

graahnul-grom, 2020-04-11:
@suboptimallife Thank you very much. And I see the same on my
VirtualBox Tumbleweed.

vzh, 2020-04-21:
Hi again!
I found the culprit, it's Pango 1.44, and more bugs related to it.  It manifests itself the same way on Debian after I updated the lib. Please check files in the directory **schematic/tests** in the repo, especially **alignment.sch** and **multiline2.sch**.  Try to zoom in and out and you'll find what was reported about a half year ago and more.  Currently, I see three issue with the new pango version:
- incorrect rendering of lower aligned text (reported here)
- incorrect overbar rendering (reported by Rob Gilton on gitter)
- incorrect rendering of multiline text (new).

I've work-arounded the first two in my `fix-font-rendering issues` branch.  Don't know yet what to do with the last one.  @suboptimallife, @graahnul-grom, and anybody interested, please check out my [branch](https://github.com/vzh/lepton-eda/tree/fix-font-rendering-issues) and let me know if it's OK for you if I merge it to the repo.

Thanks in advance.

graahnul-grom, 2020-04-24:
This is what I get on my system (FreeBSD) with pango 1.42.4.
`lepton-cli export` to `pdf` and `ps` now works.
Schematic, exported image (File->Write Image),
exported pdf (`lepton-cli export`), printed pdf (File->Print)
and print preview all look the same. All strings are
shifted up a little.
I didn't try it on openSUSE yet.

[image]

[issue-622-test-dmn.sch.txt](https://github.com/lepton-eda/lepton-eda/files/4531224/issue-622-test-dmn.sch.txt)

[issue-622-test-dmn.sch_export.pdf](https://github.com/lepton-eda/lepton-eda/files/4531185/issue-622-test-dmn.sch_export.pdf)

[issue-622-test-dmn.sch_print.pdf](https://github.com/lepton-eda/lepton-eda/files/4531187/issue-622-test-dmn.sch_print.pdf)

vzh, 2020-04-24:
> All strings are shifted up a little.

It's OK, I believe. Now *logical* text extents are used for rendering instead of *inked* ones.  So text blocks, say, "xxx" and "yyy" will be aligned on same baseline when lower alignment is used. Previously, the "feet" of the former were aligned with the "tail" of the latter.

> I didn't try it on openSUSE yet.

I did :-)  And tonight I'm going to try it on debian with new pango, too.

vzh, 2020-04-24:
@suboptimallife, please check if the branch in #629 fixes your issue.

graahnul-grom, 2020-04-24:
Yes, that baseline issue was quite annoying. I've found a bug
report on launchpad about that ([795130](https://bugs.launchpad.net/geda/+bug/795130)).

suboptimallife, 2020-04-24:
> @suboptimallife, please check if the branch in #629 fixes your issue.

Lepton EDA 1.9.10.20200319 (git: 9d3dcec) : text translation is still an issue and lepton-cli export still produces a blank pdf. Is this the correct build to test against?

vzh, 2020-04-24:
@suboptimallife, the branch I've told you about is not merged yet (waiting for you approval :-)) The PR #629 contains my own branch which you could fetch and try: https://github.com/vzh/lepton-eda/tree/fix-font-rendering-issues. Git commit 0be19e3 is the top of it.  In order to check it out, you need to add my "fork" (really, my development repo) https://github.com/vzh/lepton-eda to your git remotes.  Well, it's the way github works, sorry.  Looking forward for your review.

graahnul-grom, 2020-04-25:
Tested #629 on:
- FreeBSD 11.3-STABLE, pango-1.42.4
- antiX 17 (based on Debian 9 "stretch" - oldstable), pango-1.40.5
- antiX 19 (based on Debian 10 "buster" - stable), pango-1.42.4
- openSUSE Tumbleweed 20200407 (in VirtualBox), pango-1.44.7

On all those systems I get consistent results.

vzh, 2020-04-26:
@graahnul-grom, thank you! I'll apply the branch then.

closed 2020-04-26 by vzh

saierd, 2021-10-06:
For anyone who stumbles on this ticket, because Ubuntu 20.04 ships lepton-eda in version 1.9.10 and still contains this bug. Here is how to fix it:

* Download the deb package for Ubuntu 20.10 (lepton-eda version 1.9.11) from [here](https://packages.ubuntu.com/groovy/lepton-eda) and install it using `dpkg`.
* Remove the directory `~/.cache/guile`.

Without the second step, you might get the error message `Cannot load lepton-schematic configuration. Failed to load RC file [/usr/local/lepton/share/lepton-eda/system-gafrc]: no code for module (lepton core gettext)`, which was already shown in [this comment](https://github.com/lepton-eda/lepton-eda/issues/622#issuecomment-611202737).

PR #629 (2020-04-26): Fix font rendering issues
The branch fixes rendering issues that manifest themselves when new Pango library (1.44) is used:
- issue with different sizes of overbars on zooming (first reported by @rgilton on gitter)
- issue with wrong alignment of lower aligned text in printed output (#622)
- issue with changing vertical spacing between paragraphs of text separated with empty lines on zooming.

Besides, functionality of text bounds calculation no longer depends on `TOPLEVEL` and it has been moved to `libleptonrenderer` from `liblepton`.

Closes #622.